WEC-Sim/WEC-Sim #970: Setting up WEC-Sim and Simulink for Pivoting Sphere [Theory or Implementation]

hfpim, 2022-12-07:
**Personal project context**
_I am trying to simulate a WEC that consists of a sphere in heave attached to a pivoting arm. That arm is connected to and rotates about a rigid structure, and both the arm and the structure are out of the water. So I set up three bodies on Simulink: body(1) - floating sphere; body(2) - pivoting arm; body(3): fixed structure._

**Describe the theory or implementation approach that you have a question**
__My question relates to the wecSimInputFile as I don't know how to setup bodies that are out of the water. On the tutorials page both the RM3 and the OSWEC have underwater components and the way they are set up in the files reflect that._

**Describe the type of conclusion or resolution you are looking for**
_What considerations and configurations should I use for bodies that are out of the water? In my case, the bodies would be the arm and the fixed structure. How can I properly setup the wecsimInputFile and Simulink to properly run the simulation?_

**Additional context**
_Here is a schematic of what I'm trying to do. Any help would be greatly appreciated, thank you._
[image]

_Here is my Simulink setup so far:_
[image]

**Is your question request related to a problem? Please describe.**
_N/A_

nathanmtom, 2022-12-07:
@hfpim Thank you for your question and I will do my best to answer your question. 

Quickly I will mention that the example you are proposing above is very similar to our WECCCOMP Applications Case which you can find here https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/WECCCOMP/WECCCOMP. I highly suggest reviewing to help you understand how the `wecSimInputFile.m` was written. 

We suggest that for any bodies outside of and not interacting with the waves to use the non-hydro body option. More information can be found here https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#non-hydrodynamic-bodies.

Please review these sections and see if they help you to construct a model that represents your system shown above. If you still have questions, I am happy to continue this discussion.

Cheers, 
Nathan
National Renewable Energy Laboratory

hfpim, 2022-12-08:
@nathanmtom thank you for sending those links. Let me spend some time looking at them, I will try and setup my model similar to what's being shown. I will definitely let you know if questions come up. Thanks!

hfpim, 2022-12-08:
So I went through the example and I have a question (I've had this question for quite some time now): how can I connect the bodies or make sure the bodies are connected through constraints at the right locations?
Ex: how can I make sure my float is connected to the end of my arm?
I saw some <constraint(i).location> code lines but I can't seem to find how those interact with the geometries. I have been browsing through WEC-Sim's page but nothing comes up. 
Thanks in advance

hfpim, 2022-12-08:
Here is an update: I set up my simulation based on the files you sent and I am getting this error:
`Error using bodyClass/checkInputs (line 137)`
`Could not locate and open geometry file NONE`

`Error in initializeWecSim (line 140)`
`body(ii).checkInputs(simu.explorer);`

`Error in run (line 91)`
`evalin('caller', strcat(script, ';'));`

`Error in wecSim (line 43)`

I checked my geometry file and it seems fine. I ran wecSimInputFile and it did not point out any errors.
Thank you for your help

hfpim, 2022-12-08:
Hello, I was able to get all the geometry files and make the simulation run, but my animation is not moving. I tried changing masses and inertia of bodies and damping/stiffness coefficients of PTO but nothing happens. I ran the OSWEC example and it works fine, so the problem is in my setup. I can't seem to find what I'm doing wrong or if there is any configuration I'm missing.
- the animation shows the bodies properly connected, although the sphere (made using SolidWorks) looks much larger than what it should be.
- There are no error messages.
- Here is a copy of my wecSimInputFile:
````
%%Simulation Data
simu = simulationClass();
simu.simMechanicsFile = 'ODriveTrial.slx'; %HUGO: updated
simu.mode = 'normal';
simu.explorer = 'on';
simu.startTime = 0;
simu.rampTime = 100;
simu.endTime = 400;
simu.dt = 0.1;
simu.solver = 'ode4';
simu.cicEndTime = 30;

%%Wave Parameters
%Regular waves
waves = waveClass('regular');
waves.height = 2.5;
waves.period = 6.8540; %freq > freqmin and <freqmax set in Nemoh.cal

%%Body Data
%Float Sphere HUGO: updated
body(1) = bodyClass('hydroData/SphereMeshNemoh.h5'); %HUGO: updated
body(1).geometryFile = 'geometry/sphere.stl'; %HUGO: updated
body(1).mass = 80;  % mass[kg] HUGO: updated
body(1).inertia = [0 213 0]; %HUGO: updated; try changing Ixx and Izz to 0
body(1).linearDamping = zeros(6); %HUGO: try adding damping coeff

%Arm HUGO: created (12m length, 0.4m height, [0 0 0] CoM)
body(2) = bodyClass('');
body(2).geometryFile = 'geometry/arm.stl';
body(2).nonHydro = 1; %HUGO: non-hydrodynamic body
body(2).name = 'arm';
body(2).mass = 25;
body(2).inertia = [0 3100 0]; %HUGO inertia is actually 31,000
body(2).volume = 0;
body(2).centerGravity = [0 0 0];
body(2).centerBuoyancy = [0 0 0];

%Base
body(3) = bodyClass('');
body(3).geometryFile = 'geometry/fixedblock.stl';
body(3).nonHydro = 1;
body(3).name = 'Base';
body(3).mass = 999; %HUGO: fixed mass - copying value from WaveStar example
body(3).inertia = [999 999 999]; %HUGO: fixed inertia - copying value from WaveStar example
body(3).volume = 0;
body(3).centerGravity = [0 0 0];
body(3).centerBuoyancy = [0 0 0];

%PTO and Constraint Parameters
%Arm-Float - Fixed
constraint(1) = constraintClass('arm-float');
constraint(1).location = [0 0 0];

%Oscillo-Drive - Rotational PTO
pto(1) = ptoClass('O-Drive');
pto(1).location = [8.4853 0 8.4853];
pto(1).stiffness = 1;
pto(1).damping = 2;

%Base - Fixed
constraint(2) = constraintClass('base');
constraint(2).location = [8.4853 0 8.0];
````

Would it have to do with my hydrodynamic data? I obtained that through Nemoh and used bemio to convert those to `h5`.

nathanmtom, 2022-12-09:
@hfpim Thank you for the follow-up with your model. 

To your question about the visualization, please check that the units of the .stl you are using is in meters. Often times the object seems much larger because the export units were in say centi- or milli-meters or even inches/feet. 

Is it possible to also share your latest Simulink model? Based on the initial Simulink model image in your first post there are issues as you only have constraints and not a PTO. If your model is not moving you could quite easily have a fixed constraint in the wrong connection that is locking your device. 

I do not think your hydrodynamic data is the issue unless you are not scaling your data, but I am assuming that the files used for the visualization were the same as those used in your NEMOH run. 

If you are okay with .zipping and sharing your current model folder, this would help us try to run the model ourselves to troubleshoot your issue.

Cheers, 
Nathan

hfpim, 2022-12-09:
@nathanmtom Here are some updates I got during the day:
- I was in fact using a larger float for the .stl file. I got my hydro coefficients using a 4-m diam sphere and I used a 4-m radius .stl file, which has been corrected. That explains the size issue in the animation.
- I decreased the step size to 0.05 and changed the solver to ode45 hoping that the adaptive step size would help.
- I also used nonlinearHydro = 1 to get a more accurate response.
- I changed my Simulink model to reflect the PTO constraint between the arm and the fixed base.

I get motion for the nonlinearHydro option, although it seems like it is unstable as it keeps increasing its motion amplitude up to a point where it seems like it'll flip to the other size. I tweaked damping, masses, inertia, but that doesn't seem to stabilize the simulation.
For the linear scenario I get no motion at all. I actually get (sometimes) an initial displacement which gets dampened and becomes stationary.

I'm sending attached the zipped file. Here is how you can go straight to the important files:
- SphereMeshNemoh contains the WEC-Sim files alongside the files used with Nemoh to get the coefficients.
- WECSim Trial has the necessary files to run the simulation.

**Some information regarding the geometry:**
- Float (4m diam) has originally 80kg and a calculated mass moment of inertia of 213 kg.m²
- Arm has a mass of 25920Kg (12m length x 0.4m height) and moment of inertia of 31140kg.m²
- PTO damping is also arbitrary (at this point I'm just trying to get the simulation up and running)

My overall impression is that there is something destabilizing the simulation for the non-linear case. I just ran it again with the numbers above and the float and arm started ok and then begun swinging left and right like a pendulum.

Thank you very much for taking a look into this, much appreciated. Looking forward to your findings

Regards,
Hugo

[SphereMeshNemoh.zip](https://github.com/WEC-Sim/WEC-Sim/files/10191108/SphereMeshNemoh.zip)

nathanmtom, 2022-12-12:
@hfpim Thank you for providing information on additional tests that you have completed. 

After downloading your .zip file, I ran the `noWaveCIC` condition to see what was occurring in your simulation in the absence of waves. Without any waves upon the start of the simulation the pendulum swings down and oscillates about a new mean when the `nonlinearHydro` option is off. This is because your displaced volume from the sphere is less than the mass of the sphere and arm. My assumption would have been that you wanted hydrostatic equilibrium so if left alone in calm water the sphere would sit as you have initially placed it. 

Therefore, given the imbalance in mass and buoyancy the impact of the waves could be minimal. It would be best to try and correct the mass imbalance first before trying to determine if there are any issues with the wave dynamics. 

I would also hesitate to jump right to using the `nonlinearHydro` as the mesh resolution to accurately capture the time-varying forces could be much higher than what you currently have in your .stl file. This might be why you ended up with a pendulum type response when that was turned on rather than oscillation about a mean when it was turned off. 

Once you have had a chance to address the issues above, please let us know if your simulation is still not performing as expected.

Cheers, 
Nathan
National Renewable Energy Laboratory

hfpim, 2022-12-17:
@nathanmtom thanks very much for the invaluable insights. I will try and correct that imbalance you mentioned. I will post an update here as soon as I have meaningful results (or no results at all). My apologies for the late reply, I was getting the qualifying exams done.

Thank you!
Hugo

hfpim, 2022-12-17:
@nathanmtom I corrected the float mass and basically used a mass-less arm, with 0.01kg of mass and 0.01kg.m² of inertia. According to my calculations, using half a 2m radius sphere submerged (condition I used to get coefficients through Nemoh) and 1000kg/m³ water density:
- displaced water volume: 16.755m³ 
- equivalent sphere mass: 16700kg
- sphere inertia: 16720kg.m²

So I used those numbers under `noWave` condition and it remains stationary. If I increase or decrease the mass and inertia the float oscillates. However, when running `regular` wave condition nothing happened and the body remained stationary as if `noWave` was being used. 
I tried `equilibrium` and the mass shown was 16673.4kg, close to my number. I plotted the wave elevation and it's showing a normal sinusoidal wave. 
I don't know what to do next or how to proceed.

Thank you
Hugo

nathanmtom, 2022-12-19:
@hfpim Thank you for the update on your model progress and glad we were able to resolve the buoyancy and mass mismatch. 

To clarify, what wave conditions were you running for `regular` such as the wave amplitude and wave height. 

Second, when you say the body remained stationary is that only from the visualization or if you plot the position time history of the float the pitch motion is zero? Potentially your mass is so large that the wave amplitude and wave frequency you simulated did not generate a large response so the visualization would look like it is not moving. 

Cheers, 
Nathan
National Renewable Energy Laboratory

hfpim, 2022-12-19:
@nathanmtom The wave height and period were 2m and 7.25s.
I plotted the float's heave and heave forces and they were in fact tiny. However, I reran the simulation considering waves 5 and 10m tall and nothing happened. I tried to plot the float's heave but I got an empty plot for both cases. That keeps happening for other cases also.
[image]

Now when I plot the arm pitch I get a nice plot, although numbers are tiny. This was simulated with a 10m high wave:
[image]

Here are all the tests I ran and the results I got. I'm using `ode45` for a more precise solution, and initial step size of 0.01.

* Free decay test:
`waves = waveClass('noWave')`
`body(1).quadDrag.drag = 8` Note that this is extreme as I was trying to get a gradually decreasing amplitude response
I gave the float a negative initial displacement: `body(1).initial.displacement = [0 0 -0.5]`
Minimum PTO stiffness and damping coefficients: `pto(1).stiffness = 1` and `pto(1).damping = 1` (I didn't want to leave 0)
I left the mass of the float as user input and gave it its equilibrium value of 16750kg.
Float inertia: 26800
**Results:** the float kept oscillating about its mean equilibrium position nonstop. I even increase the drag coefficient to 10 but I got no change in its response.  I was also not able to plot the float's heave, only the arm pitch as shown:
[image]

If I increase the PTO damping coefficient to 120,000, then I get a somewhat realistic result:
[image]

* Float mass set to `equilibrium` and `body(1).nonlinearHydro = 0`
`waves = waveClass('regular')`
Wave height and period: 5m and 7.25s
No drag
No initial displacement
Minimum PTO damping and stiffness as before
Same float inertia as previous test
**Results**: no motion visualization, but the arm pitch plot shows tiny values. The float heave plot also worked:
[image]
[image]

* Float mass set to `equilibrium` and `body(1).nonlinearHydro = 1`
Same parameters as the previous test
Same `.h5` file as before
I created a finer mesh for the sphere to be used when the non-linear option is on, shown below:
[image]
**Results**: motion gradually shows up in the animation. The arm pitch and float heave responses increased with time, but still within small values as shown below. It seems like its an unstable behavior, so I increased the simulation duration to 400s and the float and arm behaved like a swing or pendulum.
_80s of simulation_
[image]
[image]

* Free decay test with non-linear hydro on and float mass set to `equilibrium`
Same parameters for the first free decay test but with a finer mesh, mass set to `equilibrium` and non-linear hydro as 1
**Results**: the same unstable result was observed (float heave plot was empty)
[image]

nathanmtom, 2022-12-20:
@hfpim Thank you for the update to your issue. Based on your results I now expect that the lack of motion is a result of you hydrodynamic excitation force coefficients being too small. When I run the BEMIO script included in your original .zip file, the coefficients are extremely small, for a 1 meter amplitude wave you are only getting ~0.01 N value in heave which is too small to excite the sphere of diameter 4 m. 

[image]

Therefore, I would suggest you double check your NEMOH run and ensure you are using the correct units in the mesh. You may also want to run over a wider frequency range. The lowest period you have is about 8 s and you may need to have a wave period higher than that to generate wave excitation force coefficients large enough to drive the float. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2023-01-03:
@hfpim I wanted to inquire if you had an update on your model development? Please let us know if this issue is still open by the end of the week or else we will close the issue and assume your original answer in this thread has been answered.

Cheers, 
Nathan
National Renewable Energy Laboratory

hfpim, 2023-01-05:
@nathanmtom I am resuming my work on NEMOH and I will make sure to check the points you mentioned. But yes, the issue is still open. I will get back with some updates as soon as I have them.

Thank you
Hugo

nathanmtom, 2023-01-25:
@hfpim Thank you for getting back to us about the status of your WECSim model build. I wanted to inquire about the status of your work as it is not 3 weeks since your last update. Please notify me if you have an update on this issue by the end of the week, 1/27. If there is no new update to this issue, we would like to close the issue and then work with you once you have additional updates to share. 

Cheers, 
Nathan
National Renewable Energy Laboratory

hfpim, 2023-01-27:
@nathanmtom please go ahead and close the issue. The project is currently paused. Thank you for the support. I will come back with more updates as soon as I can.

closed 2023-01-27 by nathanmtom