WEC-Sim/WEC-Sim #972: [WEC-Sim Applications]

wpoulter, 2022-12-11:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
N/A

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
Waves are not showing in the mechanics explorer and the view defaults to a very far away position. 
Rotational PTO class fixes second bodies movement to main body.
Mechanics explorer closes itself the first time running a simulation after initialising.

**To Reproduce**
Files included
[VAPWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/10203514/VAPWEC.zip)

**Expected behavior**
_A clear and concise description of what you expected to happen._
View to default to closer more useful position like in oswec or rm3 examples.
Rotational PTO class should allow second body to rotate in yaw about first body.
Mechanics explorer should open every time without having to stop and rerun a simulation to get it to appear.

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] R2022b
 - [WEC-Sim Release] v 5.0.1

**Additional context**
_Add any other context about the problem here._
I am trying to model a vertical axis pendulum's response to regular waves but I have got movement that seems strange (maybe it is expected but without seeming the waves the buoy appears to hover). The pendulum remains fixed relative to the buoy despite being subject to a rotational PTO.

nathanmtom, 2022-12-12:
@wpoulter Thank you for submitting your issue and following our template. 

1. By default the waves to do not show up in the mechanics explorer. There are options to place wave markers to show the wave train http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#wec-sim-visualization.
2. The rotational PTO appears to have too large of a rotational damping coefficient relative to the inertia of the rotating mass. When I set the pto damping to 0 the internal reaction mass appeared to spin relative to the larger sphere. 
3. We are aware that the mechanics explorer does require you to close before executing a new WEC-Sim run if you want to see the visualizations. We believe during debugging this is not a huge obstacle to overcome and once you are more confident in your simulation the visualization will be turned off to improve computational speed. 

Please let me know if these responses have answered your questions and issues. But please let us know if you still have questions.

Cheers, 
Nathan
National Renewable Energy Laboratory

wpoulter, 2022-12-12:
Thanks for your suggestions @nathanmtom, much appreciated. 

I just ran the OSWEC and RM3 examples and realised the mean free surface did not show any waves there. I had looked at it before and I thought they did move but I was clearly imagining it lol. 
I should have tried to removing the damping, I had tried reducing it but not removing it. I now see that it rotates, I will have to look at using a more realistic pendulum mass and damping coefficient.  
You have answered all my issues. 
Thanks again,

William Poulter

closed 2022-12-12 by wpoulter