WEC-Sim/WEC-Sim #973: [Theory or Implementation] Clarification of OSWEC hydrodynamic parameters

brittlyd, 2022-12-16:
**Is your question request related to a problem? Please describe.**
This is related to the OSWEC example in WEC-Sim and the given WAMIT results.

**Describe the theory or implementation approach that you have a question**
My question centers around the BEM data from WAMIT for the OSWEC example. First, I want to make sure I understand the indexing correctly. I am trying to extract the hydrodynamic parameters from the "hydro" structure that comes out of the bemio.m file in examples>BEMIO>WAMIT>OSWEC. The dimensions of the added mass matrix A, is [6*N, 6*N, M], where N is the number of bodies, and M is the number of frequencies, correct? So if I wanted to look at the added mass as a function of frequency for the flap in the pitch direction, that would be A(5,5,:), correct? Also, for excitation force, the dimensions are [6*N, P, M], where P is the number of wave directions, correct? That should mean that added mass is not dependent on wave direction, because there is no dependence on P, is that right? 

Assuming my indexing is correct, I am confused by the BEM results for this example. Based on the OSWEC geometry and properties, the WAMIT results should be very similar to that of previous works including "Hydrodynamics of the oscillating wave surge converter in the open ocean" by Renzi and Dias in 2013, which use an OSWEC of almost identical geometry and mass properties, from what I can tell. However, they look very different, even when accounting for the normalization that WEC-Sim uses. I assume that the OSWEC example was modeled after the geometry in this paper (which has also been used in other work, since it's a scale model of the Aquamarine Oyster). That being said, I attached images of the hydro parameters from the paper as well as the output from the hydro structure in WEC-Sim. They are very different, and I'm wondering if it's because of user error, or maybe if there is something fundamentally different with the OSWEC in the WEC-Sim example? Any guidance would be appreciated. Please let me know if you need further clarification.

**Personal project context**
I am trying to compare the hydrodynamic parameters of the OSWEC example in WEC-Sim to the hydrodynamic parameters that comes out of the semi-analytical model from the paper "Hydrodynamics of the oscillating wave surge converter in the open ocean" by Renzi and Dias in 2013. Similarly, I am trying to compare the WAMIT results from WEC-Sim to that of the 2013 thesis from van't Hoff titled "Hydrodynamic modeling of the oscillating wave surge converter", as the geometry of the OSWEC in the WEC-Sim example seems very similar. 

**Describe the type of conclusion or resolution you are looking for**
I am hoping to gain clarity on whether I am looking at the correct indices in the hydrodynamic parameter matrices, and why the WAMIT results for this OSWEC look so different than the results from previous papers with similar OSWEC geometry. 

**Additional context**
I've attached images of the hydrodynamic parameters from WEC-Sim (in the pitch direction) and the plots of the hydrodynamic parameters of the paper with similar OSWEC geometry. For the paper results, the WAMIT results that seem directly comparable are the solid circles, corresponding to a=0.9m, or 1.8m flap thickness. |F| is the magnitude of excitation force, and mu is the added mass coefficient. 

Parameters from Renzi and Dias (2013):
[image]

Parameters from WAMIT and WEC-Sim:
[image]

nathanmtom, 2022-12-16:
@brittlyd Thank you for your question and I'll do my best to answer. 

I have briefly reviewed the Renzi and Dias 2013 paper and from my understanding they are calculating the hydrodynamics about the hinge point in order to generate a one-degree of freedom equation of motion. If this is true, that is different that WEC-Sim where we calculate the hydrodynamic coefficients at the center of gravity and then use our constraints to enforce the rotational motion about the hinge. 

In this case often the rotational pitch added mass is smaller when calculated at the center of gravity rather than the hinge point. For example, to account for the moment generated at the hinge when using coefficients calculated at the center of gravity you would use the (5,5) values but would also need to use the (1,1) values times the distance from the hinge to the center of gravity. When you add these two together you will likely get closer to the solution provided by Renzi and Dias. 

Please let me know if this helps answer your question, but happy to provide additional clarity if needed.

Cheers, 
Nathan
National Renewable Energy Laboratory

brittlyd, 2022-12-16:
Hi Nathan,

Thanks so much, this is super helpful. The results are matching much better now. I had another question about what exactly the "restoring force" is in the output structure. In the Advanced Features section of the User Manual, it says that the restoring for is a "combination of buoyancy, gravity and hydrostatic stiffness", but the way that I think of hydrostatic stiffness it already is the balance between buoyancy and gravity (the difference in moments they provide about the base), so what exactly does this force consider? I would think it would just be F_restoring = C * x, where C is the hydrostatic stiffness and x is your translation variable, but it sounds like it may be different?

Thanks!
Brittany

nathanmtom, 2022-12-16:
@brittlyd Glad my response above has helped. To answer your second question, the best is to show you inside our hydrostatic restoring block, see below. If I have read your question, if the body is in hydrostatic equilibrium (displaced water mass = material mass) and the center of gravity and center of buoyancy are at the same point in the X-Y plane then the restoring force will be equal to C*x as you have listed above, where C is obtained from your BEM hydrodynamic run. However, if you have a pre-tensioned system or the center of gravity and the center of buoyancy are offset in the X-Y plane there will be net buoyancy (mass & buoyancy) forces and moments that arise. These end up being constant forces and torques as linear hydrodynamic theory has the underlying assumption of small amplitude displacements. These constant forces and torques need to be counteracted by an external force such as pre-tension or a mass distribution above the water which leads to hydrostatic balance across the entire system. An example of the mismatch in center of gravity and center of buoyancy is shown in our WECCCOMP example, where the only float in the water had a center of gravity in front of the center of buoyancy, but this moment was counteracted by an arm connection out of the water behind the float.

I hope this explanation shines a bit more light on what is being calculated in the restoring force output or at least points you to the location in the code where you can track what is being calculated.

Please let me know if you still have questions or need other clarification.

[image]

nathanmtom, 2023-01-03:
@brittlyd I wanted to inquire if the answer above sufficiently answered you question? If we do not hear from you by then end of this week we will assume the question is resolved and we will close this issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2023-01-06:
Closing this issue due to inactivity, but @brittlyd you can reopen this issue if you have outstanding questions.

closed 2023-01-06 by nathanmtom