WEC-Sim/WEC-Sim #974: [WEC-Sim Applications]

akram3232, 2022-12-18:
i tried to run wecccomp and it run coorectly but (force actuation and power )are zeros and i dont knew the resoan 
please tell me how can i solve this problem

nathanmtom, 2022-12-19:
@akram3232 Thank you for trying to use our WECCCOMP model. Unfortunately, we are requiring that all inquiries follow the issues template and please answer every section as requested. You can either modify your issue here or open a new issue.

[image]

nathanmtom, 2023-01-03:
@akram3232 I am closing this issue because of inactivity. Please open a new issue if you still need assistance.

closed 2023-01-03 by nathanmtom