WEC-Sim/WEC-Sim #975: [WEC-Sim Applications] wecccomp

ahmedbahaa840, 2022-12-19:
when i try to run the simulink model this message appears to me as shown int the screenshots that i sent.
[image]
[image]

nathanmtom, 2022-12-19:
@ahmedbahaa840  Thank you for trying to use our WECCCOMP model. Unfortunately, we are requiring that all inquiries follow the issues template and please answer every section as requested. You can either modify your issue here or open a new issue.

[image]

nathanmtom, 2023-01-03:
@ahmedbahaa840 I am closing this issue because of inactivity. Please open a new issue if you still need assistance.

closed 2023-01-03 by nathanmtom