WEC-Sim/WEC-Sim #976: [Theory or Implementation] Unbalanced hydrostatics and incorrect mass/displaced volume when using call_capytaine.py

wpoulter, 2022-12-22:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_

I have run several simulations in WECsim using hydrodynamic data generated using the call_capytaine.py script included in the WECsim examples. This script has been generating incorrect hydrodynamic data that causes the body to oscillate about the free surface at increasing speed over time. This behavior occurs in both the noWave and regular wave cases.
The mass of the body is also wrong. When using the 'equilibrium' setting for the body mass this is smaller than expected.  

**Describe the theory or implementation approach that you have a question**
_Please clearly describe the assumptions, equations, or code script that you would like clarification. Also, please provide references to other research papers, open-source material, or other information that may be relevant in the WEC-Sim developers understanding your question._

I have the origin of my model set at the center of mass of the body. When using the body.show() method in call_capytaine.py it shows that the model is being trimmed correctly. After using the body.keep_immersed_part() method, the upper half of the body mesh is removed. When using 'equilbrium' as the mass in WECsim the mass does not match what would be expected of the displaced volume of half the body mesh. 

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._

I am attempting to model the power generation of a VAPWEC inside a cylindrical shaped buoy with spherical ends. I will be using regular waves for now. I have been using several different pendulums. I have constrained the buoy to move in heave and pitch only to approximate mooring, I would use mooring matrix but I do not know the parameters of my mooring. 

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._

I want to be able to generate hydrodynamic data using the call_capytaine.py function that is hydrostatically stable. What parameters do I need to specify or how to do I need to format my mesh model to ensure this occurs?
I would also like to be able to input a body mass to call_capytaine.py or otherwise be able to get the correct waterline on the buoy for hydrodynamic calculations. Do you have any suggestions on how to implement this? I have attached my attempt to modify the call_capytaine.py script to include accept a mass input but I get the error 

xarray.core.variable.MissingDimensionsError: cannot set variable 'mass' with 2-dimensional data without explicit dimension names. Pass a tuple of (dims, data) instead.

when running. 

**Additional context**
_Add any other context or screenshots about the feature request here._

Heave response of buoy in WECsim:

[image]

Project files:

[VAPWEC sim files.zip](https://github.com/WEC-Sim/WEC-Sim/files/10283126/VAPWEC.sim.files.zip)

akeeste, 2023-01-04:
Hello @wpoulter 

I will do my best to address your questions below:

> When using the 'equilibrium' setting for the body mass this is smaller than expected...
> ... I want to be able to generate hydrodynamic data using the call_capytaine.py function that is hydrostatically stable. What parameters do I need to specify or how to do I need to format my mesh model to ensure this occurs?

WEC-Sim's equilibrium body mass (`body.mass = 'equilibrium'`) is determined by the displaced volume of the mesh in the BEM code. If the equilibrium mass is not as expected, first check the displaced volume in the BEM solution. In this case, your mesh actually has two surfaces: external and internal surfaces of the thin hollow body. This set-up is incorrect because BEM software typically only uses the external, wetted surfaces. It is treating the internal surface as wetted, so the displaced volume only considers the thin shell of the body. If you remove the internal mesh of the hollow body, your displaced volume and equilibrium mass should be as expected. This seems to be the core of the current issue, but have commented on a few other things that may be problematic.

>I have the origin of my model set at the center of mass of the body. 

Capytaine technically uses a convention that the origin of a mesh should be at the still waterline. If your body's cg is at the still waterline, the set-ups are identical. However you should be cautious about setting up simulations at different vertical positions. Follow-up on Capytaine documentation about locations of bodies.

> I would also like to be able to input a body mass to call_capytaine.py

I am not sure that this is possible or accurate. Generally, BEM bodies are given at their equilibrium position and the mass can be calculated from the displaced volume of the mesh. This is a question for the Capytaine forum if the first comment does not resolve your issue.

I noticed that your ``VAPWEC BEM.py`` only lists three frequencies: ``bem_w = np.linspace(1.571, 1.570, 3)``. Regardless of the above changes, this is not enough frequencies to get accurate hydrodynamic data. I recommend a minimum of 100 frequencies and a range approximately covering 0-10 rad/s to give stable WEC-Sim simulations.  

As a last comment, I very highly recommend analyzing your hydrodynamic data in BEMIO before moving to WEC-Sim simulations. The ``plotBEMIO()`` function will visualize the data and contains comments on expected behavior. If the input data is not sound, the simulation cannot be stable.

Please let me know if you have any more questions.
Adam

wpoulter, 2023-01-09:
Using a model of just the exterior surface in Capytaine solved the buoyancy issue I had. I am just using regular waves for now but if I use an irregular wave case in future I will use a wide range of wave frequencies in Capytaine. Thank you for your help Adam.

closed 2023-01-09 by wpoulter

akeeste, 2023-01-09:
@wpoulter I am glad this has resolved your issue. I just want to note that the small frequency range will work for WEC-Sim's `regular` wave if the accuracy of the BEM data has been verified, but will not work for WEC-Sim's `regularCIC` wave. The `regularCIC` wave calculates radiation damping and excitation forces from the respective IRFs, which would require a large frequency range. 

Adam