WEC-Sim/WEC-Sim #977: wave maker

bjfromzjou, 2022-12-22:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
wave gauge
**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
how can i monitor the free surface in WEC-SIM,? the wave maker only can create  undisturbed wave evolution, if i simulated the wave-body interection,  can the actual wave height be acquired?

nathanmtom, 2022-12-22:
@bjfromzjou Thank you for your question about the wave elevation within WEC-Sim. WEC-Sim will only plot the undisturbed wave elevation using any of our visualization features. In order to see the disturbed wave field, from the diffracted and radiated waves, your BEM solver would need to output those wave fields separately and one would need to superimpose on top of the undisturbed incident wave field. Given the additional inputs required and that this does not influence the dynamic calculation of the hydrodynamic bodies we have chosen not to include this as an option. 

I hope this answers your question, but please let us know if you have further questions.

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2023-01-03:
@bjfromzjou We have not heard a response from your initial inquiry. Please let us know by the end of this week if you still need help answering your question. If we do not hear from you by the end of the week we will close this issue because of inactivity.

nathanmtom, 2023-01-06:
Closing this issue due to inactivity.

closed 2023-01-06 by nathanmtom