WEC-Sim/WEC-Sim #978: [Theory or Implementation] Nemoh & WEC-Sim COG for a two body point absorber

timmaflan, 2022-12-22:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_

**Describe the theory or implementation approach that you have a question**
_Please clearly describe the assumptions, equations, or code script that you would like clarification. Also, please provide references to other research papers, open-source material, or other information that may be relevant in the WEC-Sim developers understanding your question._

Hello, I am currently working on modeling a point absorber wave energy converter in WEC-Sim. This is a two body system, containing a spar and a pod. The system is being modeled in NEMOH to calculate the hydrodynamics, and then the BEMIO file is used in WEC-Sim. I am trying to accurately account for the center of gravity for the spar (-16m) and the pod (-7m). However, when I change the center of gravity from (0,0,0) to (0,0,-16) the whole mesh for both objects is shifted down away from the water line. 

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._
I am designing a point absorber wave energy converter, and am hoping to calculate accurate body dynamics using NEMOH 2 >> BEMIO >> WECSIM.

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._

I wonder if I have misunderstood the theory, or the method of implementation, of the center of gravity of the device and am hoping to learn how to best account for an accurate center of gravity/center of buoyancy in the workflow mentioned above. 

**Additional context**
_Add any other context or screenshots about the feature request here._

nathanmtom, 2022-12-23:
@timmaflan Thank you for submitting your issue and I will do my best to respond. 

To clarify, WEC-Sim requires all hydrodynamic coefficients to be calculated at the center of gravity of the hydrodynamic body. Normally this means that the mesh that is used in NEMOH or WEC-Sim will have the center of gravity centered at 0,0,0 rather than oriented how it would look in hydrostatic equilibrium. In this way, if you change the COG in your NEMOH input file, WEC-Sim will read this value and shift the meshes down appropriately. If you mesh was set-up to output the coefficients at 0,0,0, you will likely need to adjust your input mesh before re-running NEMOH or WEC-Sim. If you need additional references, please see the examples we have for NEMOH [**here**](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/NEMOH).

The COB shall be referenced from the COG and the COB will be determined from NEMOH based on the supplied mesh. 

I hope this answers your question, but please let me know if you still have questions or need clarification.

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2023-01-03:
@timmaflan I wanted to check with you to see if my answer above has answered your initial question. Please let us know by the end of the week if this issue should remain open. If we do not hear from you by the end of the week we will close this issue because of inactivity.

nathanmtom, 2023-01-11:
Closing this issue due to inactivity.