WEC-Sim/WEC-Sim #980: [Feature request]

Aiswariak, 2023-01-08:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
Yes. I am working on Wave energy converters in irregular waves. After running the wecSim code we are getting the response, velocity etc as a time series rather than a spectrum even in case of irregular waves. It would be very useful and time saving if the graphs are provides as a spectrum when using irregular waves.

**Describe the solution you'd like**
I would like to get the responses and forces graph as a spectrum rather than a time series for easiness in computation.

**Describe alternatives you've considered**
I am using matlab codes to convert from time domain to spectrum.

**Personal project context**
I am working on floating Wave energy converters in irregular waves.

**Additional context**
_Add any other context or screenshots about the feature request here._

akeeste, 2023-01-13:
Hi @Aiswariak 

This is a useful feature to have, but our WEC-Sim development teams thinks that custom post-processing may be better suited for the [MHKiT project](https://github.com/MHKiT-Software/MHKiT-Python). MHKiT is an open-source marine energy software that includes the type of data processing you are suggesting. Implementation in MHKiT vs WEC-Sim would help our teams avoid duplicating efforts and makes those functions available to a wider audience. MHKiT is available in both MATLAB and Python. Once installed you could easily call the MHKiT-MATLAB functions in the `userDefinedFunctions` or other scripts as needed. MHKiT-Python also contains functions to import the WEC-Sim output structure into Python if that is preferable.

``mhkit.wave.resource.elevation_spectrum()`` seems to be the closest fit at the moment. We could make some minor adjustments in MHKiT to turn that into a general FFT function for time series outputs. @ssolson @rpauly18 thoughts? I could get an MHKiT PR started to make these updates. And moving beyond this thread, I wanted to record a couple other items the WEC-Sim team has discussed that would be appropriate for MHKiT:
- RAO function
- Return phase information from FFTs (elevation_spectrum does not do this, and I don't think ``scipy.signal.welch()`` even has the ability to because of the method it uses)

Aiswariak, 2023-01-16:
Hi @akeeste 

Thank you for the response. I will try working with MHKiT project. 

Yes, It would be better if the suggestions are incorporated.

akeeste, 2023-02-08:
@Aiswariak I am moving forward with implementing these post-processing functions in MHKiT where they can be more widely used, in addition to with WEC-Sim outputs. I will tag you in that upcoming PR for your awareness and to reference if you want to convert to MATLAB more quickly. I will close this WEC-Sim issue unless there is further discussion needed.

closed 2023-02-08 by akeeste

Aiswariak, 2023-02-09:
Thank you so much @akeeste. Looking forward to it.