WEC-Sim/WEC-Sim #982: OSWEC Free Decay Tests

vizzvanath, 2023-01-20:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Hello WEC-Sim team! 

I'm trying to model an oswec based on the experimental results of Alan Henry's PhD thesis (shorturl.at/hioqy). Width of the flap is 6m, height is 9.5m, and thickness is 1.8m. PTO stiffness is set to zero and damping is  1 MNsm/rad. When I performed decay tests under these conditions with nonlinear hydrodynamics I got a natural period close to 14s which is agreeing with experimental results. Then I simulated the device in 1m tall regular waves with same PTO parameters for periods ranging from 1s to 20s expecting to see a maximizing of power near 14s. However peak has shifted to 10s rather than 14s. If my understanding is right I should see some resonance near 14s. I tried making the mesh finer and increasing the number of frequencies in the Nemoh hydrodynamic analysis suspecting a faulty in that. still getting the same results though.  Is my understanding correct? 

**WEC-Sim file(s)**
[WEC-Sim.zip](https://github.com/WEC-Sim/WEC-Sim/files/10467668/WEC-Sim.zip)

**To Reproduce**
For Free decay test - set FreeDecay = 1;
                                - run wecSim for waves = waveClass('noWave')

For power curve - Run wecSimMCR for regular waves (waves.height = 1;  waves.period =1:20; )
                           - Run plot(mcr.Avgpower)
   
**Expected behavior**
Maximizing of power near 14s.

**Screenshots**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release][R2021b]
 - [WEC-Sim Release][v4.1]

**Additional context**
N/A

dforbush2, 2023-01-25:
Hi @vizzvanath 

I have looked at your BEM data, and it looks pretty good, but be aware that your frequency discretization near your frequencies of interest, 10 - 14 seconds, is fairly coarse. This is because you specified a linear frequency spacing, and period is 1/frequency. I would not expect good simulation results near these frequencies when using the 'regular' wave class, which looks up (interpolating if necessary) the regular wave excitation coefficients and uses a single set of these coarsely spaced values. I would recommend the 'regularCIC' wave class which will use the convolution integral formulation based on the impulse response function, and your impulse response functions look nice. I would similarly suggest you use the 'noWaveCIC' free-decay. 

I re-ran your free-decay with 'noWaveCIC' and, with your damping, I see a damped decay frequency of 12 s, and, with no damping, I see a natural frequency of 11 s. From these results, I would expect to see your power peak at the natural frequency of 11 seconds for a fixed damping value. 

I re-ran your cases at 9:1:14 with your damping and 'regularCIC' and in fact that is what we see. The decay test you ran with PTO damping did not give you the natural resonance frequency but the damped frequency. That, and the use of 'regular' and 'noWave' instead of the CIC methods with a fairly coarse frequency discretization around the area of study accounts for your issue.

Please let me know if you have further questions. I saw no structural problems with you model.

vizzvanath, 2023-01-28:
Hi @dforbush2 

Thank you for your insight. I don't think it possible to specify a non linear frequency discretization in Nemoh, or is there a way?. 

In both the final report of NumWEC project and Alan Henry's PhD thesis the natural period of the oswec seems to depend only on inertia and stiffness. That is why I was hoping a maximization of power near resonance irrespective of damping value as the pitch RAO maximizes at resonance. Let me know what you think.
 
[image]

Thanks again!

dforbush2, 2023-01-30:
What you are saying is correct, that the natural frequency does depend only on stiffness and inertia. The equation you present is just an expansion of the canonical form https://en.wikipedia.org/wiki/Natural_frequency with constituent terms broken out. The addition of damping, however, is expected to change this frequency to a damped frequency, which is always less than the frequency of free vibration. (See free vibration with damping https://en.wikipedia.org/wiki/Vibration)

 The maximization of an RAO in the power-capture degree of freedom does not imply a maximization of potential power capture. Reading through the dissertation you have recommended, there are several remarks to this end. To paraphrase, the RAO is maximized when there is NO damping force applied. But this means there is also power being extracted from the motion. As a counter point, an arbitrarily large damping will lead to large forces acting at the PTO, but there will be no motion. There are analytical ways to pursue this for a linearized system model (see https://www.sciencedirect.com/science/article/pii/S2405896320315780, particularly explore the works cited) but I would caution you that the results of Dr. Henry show significant non-linearity. In experiment, this is typically done by varying the damping level for a particular sea-state. Your approach seems to be the opposite, varying the sea-state for a particular damping. But there is no guarantee that this damping level will be optimal for any sea state.

closed 2023-02-14 by dforbush2