WEC-Sim/WEC-Sim #983: [Feature request] BEMIO readNEMOH.m Update to NEMOH 3.0

lloraras0510, 2023-01-23:
Hello the WEC-Sim team,

Thanks for all the amazing work you have done!

As you may know, NEMOH was updated to v3.0 Dec 2022, and with that, some format changes were made to Nemoh.cal. Noticeably, on line 27 of Nemoh.cal, there is an additional number at the beginning denoting the unit output of NEMOH.
e.g., `1 50 0.1	2.0			! Freq type 1,2,3=[rad/s,Hz,s], Number of wave frequencies/periods, Min, and Max`

I made the following changes to readNEMOH.m to make it work, from line 92 to 97
92 `    if isempty(strfind(raw{n},'Number of wave frequencies'))==0`
93 `    tmp = textscan(raw{n},'%f %f %f %f');                       % added one more %f`
94 `    hydro(F).Nf = tmp{2};  % Number of wave frequencies         %tmp(1) ->2       `
95 `    hydro(F).w = linspace(tmp{3},tmp{4},tmp{2});  % Wave frequencies        %tmp2,3,1 ->3,4,2`
96 `    hydro(F).T = 2*pi./hydro(F).w;  % Wave periods`
97 `   end`

Also, I noticed a potential bug on line 173 of readNEMOH.m. 
173 `    if isempty(strfind(raw{n},'Diffraction force'))==0`

It is scanning for 'Diffraction force' in ExcitationForce.tec? I generated my ExcitationForce.tec from NEMOH, and by default, it is 
e.g. 8 `Zone t="Excitation force - beta =   0.000 deg",I=   100,F=POINT`
So the potential solution here is to change line 173 to  `    if isempty(strfind(raw{n},'Excitation force'))==0`

And in all of the WEC-Sim examples for BEMIO NEMOH, they have 'Diffraction force' in the ExcitationForce.tec. I'm not sure if it is done on purpose or a potential bug, so I hope maybe someone from the development team can clarify. 

Thanks again for building WEC-Sim. I'm excited to learn and research with this tool!

Best,
Yi

kmruehl, 2023-01-25:
Hi @lloraras0510 thank you for bringing our attention to changes made to NEMOH v3.0 output, and the issues with BEMIO's `readNEMOH` script. No one on the WEC-Sim team has run NEMOH v3.0 (yet), and as a result we don't have any test cases. Can you share a NEMOH v3.0 case with our team?

Also, since you've already proposed a solution to making `readNEMOH` compatible with NEMOH v3.0, would you like to submit a pull-request to add this functionality?

Thanks again.

yman95, 2023-01-30:
Hello @kmruehl,

I just committed a readNEMOH.m file to the dev branch from GitHub Desktop. Not sure if that's how it works since it's my first time contributing. Please let me know if you can see it from your end or if that's the correct branch to contribute to. Thanks. 

Yi

kmruehl, 2023-02-08:
@yman95 thank you! I see your PR and will review it.

kmruehl, 2023-02-08:
@yman95 it looks like there was an issue with the PR you submitted, so I re-opened a new one: https://github.com/WEC-Sim/WEC-Sim/pull/990.

Do you have a a NEMOH v3.0 case you could share? Our team hasn't used Nemoh v3.0, so we don't have a way to test this revision.

yman95, 2023-02-08:
@kmruehl Thanks for getting back.

I just submitted two folders to the dev branch. It's supposed to be a simple sphere with radius of 20 meters with 2/3 of its body submerged. 

sub2 is the folder with results generated from running NEMOH 3.0. In it, the BR folder contains the skeleton files generated by BEMRosetta. sphere_sub folder contains the files for my attempt to run with wecSim from MATLAB. 

Though, for my sub2\results\ExcitationForce.tec, on line 8, I changed 'Excitation' to 'Diffraction', as to follow the examples in WEC-Sim, and did not change line 173 of readNEMOH.m on my end. Sorry if there's any confusion. 

My attempt for simulation is not successful at the moment. This leads to my question on the WEC-Sim. 

MATLAB suggests there's singularity and a degenerate mass distribution. I lowered the time step to 0.01s. And tried the suggestion on the trouble shooting page by adding an inertia component to the .slx file, and a dummy body(2) and declaring it in the wecSimInputFile.m, but both did not work. Maybe could you take a look at what might went wrong? Thanks for any help.

Yi

kmruehl, 2023-02-08:
@yman95 I can't see where you submitted the folders. Can you upload a ZIP of the cases here?

yman95, 2023-02-08:
[sphere_sub.zip](https://github.com/WEC-Sim/WEC-Sim/files/10690636/sphere_sub.zip)
[sub2.zip](https://github.com/WEC-Sim/WEC-Sim/files/10690637/sub2.zip)

Hope this works.

kmruehl, 2023-02-09:
@yman95 thank you, I'll take a look at the cases and work on the PR to include this feature

dforbush2, 2023-03-03:
Hi @yman95 

Can you please provide the _unmodified_ NEMOH v3.0 output file? I understand you changed the data table start string from "Excitation force" to "Diffraction force" but I will need the precise string to ensure the implementation is correct. 

Please try the attached readNEMOH.m file on the _unmodified_ file and advise

[readNEMOH.txt](https://github.com/WEC-Sim/WEC-Sim/files/10877389/readNEMOH.txt)

Currently debugging WECSim run, will follow up

dforbush2, 2023-03-03:
Hi @yman95 

A few things on your BEM first:
* it seems like you have a lot jumps in added mass, radiation, and excitation for the sphere you are analyzing. These are either 1) irregular frequencies related to internal reflections 2) you need a more discretized mesh. Checking your *cal files, I see only 260 panels...that's a relatively small number for this geometry, and you may have better luck if you increased your panel discretization.
* your IRF functions don't decay to zero in the calculated time. More panels/frequencies will help with this, but I also recommend you extend the calculation time in your BEMIO call. 

Now looking at your model, I'm a bit confused as to what is happening here: you have a 3DOF floating joint, followed by a 1 DOF translational PTO separated by a 1 kg mass, which is exceptionally small compared to the mass of the sphere (~2.5e4). Effectively, this is going to act just like a 3DOF joint, but the solver is going to struggle around this 1 kg mass b/c you have effectively created a very stiff problem. If this is modeled as intended (which I suspect it is not), you may have better luck with a stiff (e.g. ode15s) or moderately stiff solver (ode23tb).

Let me know what you were going for here and I might have some different suggestions for modeling this. This model runs totally fine with the mass removed as a 1 DOF or 3 DOF system, with the mass removed and either the PTO or the 3DOF constraint connected from the frame to the body.

yman95, 2023-03-11:
> 
Hello,

Sorry for the delay. I was busy with another project.

I have attached the unmodified outputs from NEMOH v3.0 and ran the new script. From my end, everything seems to be working. The MATLAB code graphed six figures of its calculations and wrote the .h5 file. Let me know if I have included all the NEMOH outputs for your testing. Thanks. 

[Nemoh 3.0 unmodified results.zip](https://github.com/WEC-Sim/WEC-Sim/files/10949544/Nemoh.3.0.unmodified.results.zip)

Will follow up on the second post shortly.

yman95, 2023-03-14:
The original idea was to test something really simple, a regular sphere in my case, in regular wave condition, just to understand how the program works. 

- I used a simple mesh thinking it might save computing time and is easier to run. I was not aware of how low panel counts can affect the simulation. I'll use a higher res .stl file.

dforbush2, 2023-03-14:
Hi @yman95 

Point by point responses:

* You're right, a simple mesh is easier and faster, but it will affect your BEM results. Since WEC-Sim relies on these coefficients, and you only have to run BEM once, it's worth doing a longer run. It is also worth exploring the options for irregular frequency removal, since this will still be a problem for any floating body regardless of panel discretization. 
* Great. You may find you don't need to go that long w/ the IRF calculation with a more refined BEM model. 
* By imperfect BEM I meant BEM that is "jumpy" or has spikes, or shows a really different result from one frequency to the next. If you are using "regular" waves, the solver will take your BEM coefficients _from exactly that frequency_ and use it to determine hydrodynamic forces. If you happen to select a frequency at which you have a spike or a "bad" point in your BEM data, your simulation results will be pretty far off, even if the rest of your BEM is good! Using "regularCIC" waves, the solver calculates hydrodynamic forces based on the impulse response function, which is determined by the convolution integral of _all frequencies of your BEM data_, so it can be more robust to BEM that has some spikiness in it. 
* The CG and CB values WEC-Sim uses by default will be read from your *h5 file from your BEM data. You need to be careful running your BEM so that the origin of the mesh used here is at the CG of the body. I am not a NEMOH expert, though.

dforbush2, 2023-03-22:
Solved by #990

closed 2023-03-22 by dforbush2