WEC-Sim/WEC-Sim #984: [Feature request] A question about how to do a Monte-Carlo Simulation on a WEC-Sim numerical model

Doudou-Li, 2023-01-25:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
N/A._

**Describe the solution you'd like**
N/A._

**Describe alternatives you've considered**
How to change the modelling parameters during the multiple case run or do a sensitivity analysis? ._

**Personal project context**
I would like to do some Monte-Carlo Simulations for doing a sensitivity analysis based on the WECCCOMP simulation platform. 
The numerical modelling information can be found from the paper 'Numerical Model Development and Validation for the WECCCOMP Control Competition' and 'AN ENERGY-MAXIMISING MPC SOLUTION TO THE WEC CONTROL COMPETITION' ._

I think it is good to do some Monte-Carlo Simulations to obtain the likelihood of a range of results of occurring, testing the designed estimator and controller for a WEC system at the same time.

Something details related to Monte-Carlo Simulation can be found in attachments.
[Observer-Based Unknown Input Estimator of Wave Excitation Force for a Wave Energy Converter.pdf](https://github.com/WEC-Sim/WEC-Sim/files/10500029/Observer-Based.Unknown.Input.Estimator.of.Wave.Excitation.Force.for.a.Wave.Energy.Converter.pdf)

[Last thesis.pdf](https://github.com/WEC-Sim/WEC-Sim/files/10500033/Last.thesis.pdf)

**Additional context**
N/A._

It would be great help from you to me.

Many thanks
Doudou

dforbush2, 2023-01-25:
Hi @Doudou-Li 

I'm looking at the WECCCOMP model WaveStar.slx, and my understanding of your request is that you would like to run a set of simulations where each simulation uses a different set of parameters, or unique state estimators/controllers. 

Depending on what parameters you would like to change, you can use wecSimMCR to loop through most of the parameters that one can specify in the wecSimInputFile. Using the *mat file workflow allows the most flexibility. I would encourage you to look at the MATLAB function "fullfact" if you have a large number of changing parameters. But this won't work for everything. I can suggest some alternative workflows if you would can give me more detail about what you would like to change.

Doudou-Li, 2023-01-27:
Hello dforbush,

Yes, I'd like to run a set of simulations with a different set of parameters.  But It seems that the WEC-Sim new feature 'Multiple Condition Runs' only allows to set a range of sea states for doing the multiple condition simulations. There don't have many parameters which can be changed in the 'wecSimInputFile'.

Many thanks
Doudou

dforbush2, 2023-01-27:
Hi @Doudou-Li 

Using the *mat file workflow, you are allowed to change more parameters than just the sea states. I've attached an example "makeMCR" that changes the linear damping for each body and the mooring stiffness in the RM3 Mooring example. 

1) run `makeMCR`, this will create a file called exampleMCR.mat.
2) run `wecSimMCR` from the command line. This will reference the attached wecSimInputFile.  

This is still somewhat limited, so let me know if there are modifications you'd like to make that are not allowed in this way. 

[mcrFileExample.zip](https://github.com/WEC-Sim/WEC-Sim/files/10520299/mcrFileExample.zip)

Doudou-Li, 2023-01-30:
That's really great. I will start to look through your provided files and see if I can do the same thing using the *mat file workflow.

Best regards

Doudou

dforbush2, 2023-02-08:
I am going to close this issue for now, but feel free to re-open it if you have further questions.

closed 2023-02-08 by dforbush2

Doudou-Li, 2023-02-08:
That's very good.

Thanks.