WEC-Sim/WEC-Sim #987: Error reading h5 files [BUG]

hresing, 2023-02-01:
**Describe the bug**
The .h5 files that I generate using BEMIO.m on the .AH1 and .LIS files I get from AQWA aren't able to be read when I run wecSim.

When I use the OSWEC.out file or the OSWEC.AH1 and OSWEC.LIS files from the tutorial and examples/BEMIO/AQWA/OSWEC folders respectively, BEMIO and then wecSim run without issue. 
However, if I try to use .AH1 and .LIS files that I have generated from AQWA, they get through BEMIO fine, behaving as expected with an output hydro structure, a .h5 file, and plots that look as expected. Then, when I try to use that .h5 to run wecSim (using the same .slx and input files as before) I get the error message in the screenshot below.
I have run my own AQWA files through wecSim successfully before, so I am not sure what has changed, other than the fact that I am now working on a new computer, so had to re-clone the repository and such again on the new computer.

**WEC-Sim file(s)**
These are the .AH1, .LIS and resulting .h5 files that wecSim won't run.
[AQWAFiles.zip](https://github.com/WEC-Sim/WEC-Sim/files/10560820/AQWAFiles.zip)

**To Reproduce**
1. Generate the .h5 file using BEMIO with readAQWA and my .AH1 and .LIS files
2. Edit the wecSimInputFile.m to use the OSWEC tutorial simulink file, geometric set up, and the appropriate .h5 file.
3. Run wecSim

run via command line

**Expected behavior**
I would expect for wecSim to run the same way with my files as it does with the ones from the tutorial/bemio examples on github.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10
 - MATLAB/Simulink Release R2022b
 - WEC-Sim Release v5.0.1 

**Additional context**
I have installed the large file storage via git lfs install, which seemed to be the answer to other people having issues with reading h5 files, and tried changing settings that I thought could be causing an issue with not having enough data from AQWA (for example, calculating hydrodynamic parameters at 100 frequencies rather than any fewer)

jtgrasb, 2023-02-09:
Hi @hresing, thank you for creating an issue and providing the corresponding files. After running BEMIO using your files, it appears that the issue may be related to the number of bodies. Your AQWA files only seem to be for one body while the OSWEC example is currently set up for a 2-body BEM run. You have a few options that I think would make this work:
- Add the base body to your AQWA run
- Remove the base body from the WEC-Sim model entirely
- Change the base to a non-hydro body (an example for which can be found in [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications))

Please let me know if you have further questions.

Best, Jeff

jtgrasb, 2023-02-22:
Closing due to inactivity.

closed 2023-02-22 by jtgrasb