WEC-Sim/WEC-Sim #989: [Theory or Implementation]About OpenFAST-WECSim coupling

hachikoi1, 2023-02-07:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
    Dear @dav-og, I plan to couple OpenFAST with WEC-Sim,
similar to https://github.com/WEC-Sim/WEC-Sim/issues/956. 

**Describe the theory or implementation approach that you have a question**
    To my knowledge, OpenFAST (or `ElastDyn`) could provide only the translational and rotational **motions** of a platform (floating body), including position, velocity and acceleration, to WEC-Sim, but not the loads. Based on the platform motions, the hydrodynamic and mooring **load** results calculated by WEC-Sim will be transferred back to OpenFAST.

**Personal project context**
   Coupling OpenFAST with WEC-sim, where OpenFAST solves for the platform **motions** of a FOWT and WEC-Sim (containing MoorDyn) solves for the **hydrodynamic and mooring loads**
    
**Describe the type of conclusion or resolution you are looking for**
    **The key question for the coupling is whether WEC-Sim (or Simscape Multibody) supports specifying the motion of the floating body/platform as a condition for the current time step.** If I specify the motion of a reference point on the floating body/platform, will the calculation crash due to the misaligned platform-mooring connections.

**Additional context**
    Could you please describe the coupling strategy between WEC-Sim and MoorDyn? Does WEC-Sim transfer the floating body motions (position, velocity and acceleration) to MoorDyn and MoorDyn transfers the mooring loads back to WEC-Sim, or WEC-Sim transfers the floating body loads to MoorDyn and MoorDyn transfers the motions back to WEC-Sim? 

    Look forward to hearing from you. Many thanks!

kmruehl, 2023-02-08:
@hachikoi1 Thanks for your question about coupling WEC-Sim with OpenFAST. This sounds like a very interesting project. 

The coupling strategy between WEC-Sim and MoorDyn is the former: WEC-Sim transfer the floating body motions (position, velocity and acceleration) to MoorDyn and MoorDyn transfers the mooring loads back to WEC-Sim. 

However, for coupling with OpenFAST you should be able to specify a position, instead of a force. This is similar to how our different [PTO blocks](https://wec-sim.github.io/WEC-Sim/master/user/code_structure.html#pto-blocks) are formulated. We have both motion and force actuated PTO blocks. I'm not sure how easily this approach would work for a body, but it should be possible. 

Best wishes, and let us know how it goes.

hachikoi1, 2023-02-10:
Dear Kelley
 
Thanks for your reply. As you mentioned, the issue we are concerned about is the motion of the substructures that have translational motion to the platform (e.g, the following platform type). Our motivation for developing the OpenFAST-WECSim coupling is the powerful constraint library of WEC-Sim. However, it is very difficult to specify the motion of platform substructures based on the platform motion from OpenFAST.  

[image]
    I have the following confusion. Could you please provide some advice or comments?
    **1. OpenFAST can only transfer translational and rotational displacement, velocity and acceleration of platform to WEC-Sim. Can WEC-Sim solve for hydrodynamic loads based on the platform motion transferred from OpenFAST? 
1.1 Is acceleration necessary? According to the solve algorithms in OpenFAST, WEC-Sim will be called 8 times in one time step to correct acceleration. If WEC-Sim calculation does not involve acceleration, the coupling of the two would be much simpler. 
1.2 Can WEC-Sim automatically calculate the motions of substructures and platform-mooring connections based on the motions transferred from OpenFAST?**  
    **2. Another coupling solution is to modify the source code of OpenFAST to transfer the loads of the upper structures of platform to WEC-Sim. This approach is similar to the MoorDyn-WEC-Sim coupling and seems more feasible than 1.** 
Look forward to hearing from you. Many thanks!

kmruehl, 2023-04-13:
@hachikoi1 Thank you for your patience, and my apologies for the slow reply. I'll attempt to answer your questions. 
1, 

    **1. OpenFAST can only transfer translational and rotational displacement, velocity and acceleration of platform to WEC-Sim. Can WEC-Sim solve for hydrodynamic loads based on the platform motion transferred from OpenFAST?**
I can't think of a reason this would cause any issues, but I'm not sure how to best implement this in a WEC-Sim body block. It would be much easier to apply an additional force/moment to a WEC-Sim body than it would be to apply additional translational and rotational displacement, velocity and acceleration to WEC-Sim. Is it possible for OpenFAST to pass the force and moment instead?

**1.1 Is acceleration necessary? According to the solve algorithms in OpenFAST, WEC-Sim will be called 8 times in one time step to correct acceleration. If WEC-Sim calculation does not involve acceleration, the coupling of the two would be much simpler. **
Yes, this has to do with the way we handle added mass. Please refer to our [documentation on added mass](https://wec-sim.github.io/WEC-Sim/dev/developer/advanced_features.html#added-mass) for more information. 

**1.2 Can WEC-Sim automatically calculate the motions of substructures and platform-mooring connections based on the motions transferred from OpenFAST?**  
This relates to 1. I think it should be possible, but it would require breaking the body block library and manually added additional translational and rotational displacement, velocity and acceleration to the body. I don't recommend this approach, but if you have success, please let us know!

**2. Another coupling solution is to modify the source code of OpenFAST to transfer the loads of the upper structures of platform to WEC-Sim. This approach is similar to the MoorDyn-WEC-Sim coupling and seems more feasible than 1.** 
Based on the prior questions I think this is a more straightforward approach. WEC-Sim can pass loads to OpenFAST, and/or OpenFAST can provide loads to WEC-Sim (one or two-way coupling). I recommend exploring this avenue.

hachikoi1, 2023-04-15:
> **1. OpenFAST can only transfer translational and rotational displacement, velocity and acceleration of platform to WEC-Sim. Can WEC-Sim solve for hydrodynamic loads based on the platform motion [...]

Dear [kmruehl](https://github.com/kmruehl)
Thank yuou for your reply.
We had intended to transferre the hydrodynamic loads from WEC-Sim to OpenFAST. But this approach has fatal problems, that are, the added mass force is wrong and B2B force is unavailable. We eventually transferred acceleration to OpenFAST, in which acceleration is converted to load.
Beat regards,

kmruehl, 2023-05-11:
@hachikoi1 I'm glad to hear that you were able to find a way to couple WEC-Sim and OpenFAST using acceleration. That sounds like a good approach. I'm going to go ahead and close this issue.

closed 2023-05-11 by kmruehl