WEC-Sim/WEC-Sim #992: [Feature request] How to change the power take-off system in a WEC-Sim model  (WECCCOMP)

Doudou-Li, 2023-02-14:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
N/A._

**Describe the solution you'd like**
N/A._

**Describe alternatives you've considered**
I want to replace the linear motor by a rotational generator or a hydraulic power take-off (PTO) system in the WECCCOMP simulation platform.

**Personal project context**
I have designed and tested my wave excitation moment estimator and energy-maximising controller on the WECCCOMP model and thought the results are quite good. Then I would like to test my designed method with the different PTO systems. To see  if the generated power will be different or any interesting thing can happen. 

**Additional context**
N/A._

That would be great helpful if you can help me to take out the linear motor and put any other PTO system.

Many thanks
Doudou

nathanmtom, 2023-02-14:
@Doudou-Li Thank you for your inquiry about extending the WECCCOMP example to have a different PTO unit. However, the WECSim team cannot decide on the type of PTO you want updated within the WECCCOMP model. Any new PTO would need to be designed to fit within the system of joints that defines the mechanical motion of the Wavestar float. 

If you want to review the existing options within PTOSim, we can assist your development or provide comment on how to use that toolbox. But at this time we will not be taking the lead on developing a new WECCCOMP model with alternate PTO. 

Let us know if you have any questions, comments, etc.

Cheers, 
Nathan

Doudou-Li, 2023-02-14:
Yes, I'd like to use one of the existing PTO units from the PTO-Sim.
[image]

So I think I need your assist for designing a PTO system it within the system of joints from the WEC-Sim source. 

BR
Doudou

nathanmtom, 2023-02-15:
@Doudou-Li Thank you for the follow up response. The WECSim team can provide assistance if you run into errors or problems incorporating a new PTO into the base WECCCOMP model but we will not be leading development of an example case for you at this time. 

If you want to collaborate more directly on such an effort we could explore having you request support through the TEAMER program, https://teamer-us.org/product/sandia-national-laboratories-wave-energy-converter-simulator-wec-sim/. Contact information to start this discussion can be found on that web page if desired. 

I am going to close this request at the end of the week as there is no issue here and this is more of a request for model development. 

Cheers, 
Nathan

Doudou-Li, 2023-02-16:
OK, that's very helpful.  You can close this request now.
I will try to read more references related to the existing PTO units cos I don't have this background, and see if I can do anything based on them for energy-maximising control purpose. I will contact the TEAMER program to have a collaboration if necessary.
 
Anyway, a wave to wire model can be useful in real application control design.  Model Predictive Control (MPC) might be limited in designing with it but a robust controller can be OK based on a wave to wire model and brings some benefits against the system uncertainties as well.

Many thanks.
Doudou

closed 2023-02-17 by nathanmtom