WEC-Sim/WEC-Sim #993: Sign mismatch in Rotary To Linear Adjustable Rod

jwills801, 2023-02-14:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I am using the check valve power take off for the RM5 as is done in the applications directory of wec-sim and is documented in the WEC-SIM PTO-Sim documentation page. I wanted to investigate the sign convention used (i.e. does positive velocity mean the the cylinder is moving toward side A or B?), so I added a To Workspace block to look at the position and velocity of the hydraulic cylinder in matlab. They did not seem to match so I integrated the velocity signal and compared it to the position signal. They were mirrors of each other. Changing the sign of one of them resulted in a near perfect match between position and integral of velocity.

**WEC-Sim file(s)**
The Rotary to Linear Adjustable Rod library block in PTO-Sim

**To Reproduce**
1. Go to WEC-Sim/WEC-Sim_Appications/OSWEC/OSWEC_Hydraulic_PTO
2. Adjust the .slx file so that translational position and velocity of the hydraulic cylinder can be observed.
3. run wecSim from command line
4. Plot translational position and velocity

**Expected behavior**
The integral of velocity should be position

**Screenshots**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: macOS Monterey 12.6.1
 - MATLAB/Simulink Release: 2021a
 - WEC-Sim Release: 5.0.1

**Additional context**
N/A

jleonqu, 2023-02-15:
Hello @jwills801 
Can you upload your working files? I would like to see where and how you are measuring the speed and the position of the cylinder. Thanks!

jwills801, 2023-02-15:
Hello @jleonqu 

Yes. Here is the repo: [https://github.com/jwills801/OSWEC_Hydraulic_PTO](https://github.com/jwills801/OSWEC_Hydraulic_PTO)

Thank you for the help!

jleonqu, 2023-03-07:
Hello @jwills801 

Yes, I changed the sign of position. As you mentioned, there was an error in the sign. I have checked the equations for velocity and that part is ok. I have also checked the other motion conversion blocks, and the equations are correct. I have submitted a PR ( #1002 ) to fix this bug.

The correct sign convention for the hydraulic cylinder is positive when the cylinder rod is in extension:
[image]

jwills801, 2023-03-07:
Okay. Thank you!

closed 2023-03-08 by jwills801