WEC-Sim/WEC-Sim #994: [Theory or Implementation] WECCCOMP WAMIT.out file parameter query and request for WAMIT inputs

Robert6codes, 2023-02-15:
**Is your question request related to a problem? Please describe.**

_Regarding the WECCCOMP WAMIT out file, I am confused about some of the WAMIT inputs, particularly XBODY, ZBODY, XCG, ZCG. According to the WEC-Sim manual, within WAMIT, the origin of the body coordinate system and mesh must be at the centre of gravity. According to WAMIT, XBODY, YBODY, ZBODY are the dimensional (X,Y,Z) coordinates of the origin of the body-fixed coordinate system relative to the global coordinate system. Therefore, it is my understanding that XBODY, YBODY, ZBODY should be set as +x and +z such that +x and +z represent the centre of gravity. In the WAMIT force control file according to the WAMIT manual, I must specify XCG, YCG, ZCG in terms of the body coordinate system. Therefore, after following the WEC-Sim requirements if XBODY, YBODY, ZBODY is already set to the Centre of gravity, then XCG, YCG, ZCG should be set as 0 but its not. I don't understand why? Please refer to the attached photo as an aid._

**Describe the theory or implementation approach that you have a question**

_The two sources used are the WAMIT manual and the WEC-Sim manual. The requirements of the two softwares to my understanding does not match the inputs to WECCCOMP in the form of the WAMIT.out file. There have been no assumptions or equations and the code script in question is the WAMIT.out file from the WECCCOMP project which lists some of the inputs to WAMIT used to gain WAMIT.out._

_Within WEC-Sim I am modelling the Wavestar as a float with all the material properties and a massless connection rod which is different to what was done in WECCCOMP treating the float and arm separately. I believe this is where the confusion is likely to be. Apart from this, modelling the Wavestar in roll, and the scale, there is no difference between my model and the WECCCOMP model_

WAMIT manual: https://www.wamit.com/manual.htm_

**Personal project context**

_I am a masters of Mechanical Engineering student at Edinburgh University. My dissertation involves a cross-code comparison of WEC-Sim to OceanEd (https://github.com/D-Forehand/OceanEd) which I am altering to include nonlinear terms to the cummins equation. The model I am using is a 1:40th scale Wavestar and therefore I have been using the WECCCOMP 1:20th scale Wavestar for guidance. The WAMIT.out file requirements are different for the two softwares and I am struggling to understand the requirements for WEC-Sim as there is a miss-match between the WAMIT and WEC-Sim manual and what was done to get the WAMIT.out file for the WECCCOMP project. The end goal is not only to improve OceanEd but to assess the convolution term methods within the two softwares._

**Describe the type of conclusion or resolution you are looking for**

_I am unclear about the WAMIT inputs used within the WECCCOMP project and how modelling the arm and float separately influences this compared to modelling the float with all the material properties and arm as massless. An explanation of the XBODY, ZBODY, YBODY and XCG, YCG, ZCG  inputs to WAMIT used within the WECCCOMP competition would be great. If possible also, getting the WAMIT inputs used would be great.  Also unrelated to the rest of the points, is there anywhere I can look to get more information on the WECCCOMP Simulink setup? Much appreciated!_

**Additional context**

The inputs to WAMIT I am referring to from the WECCCOMP WAMIT.out file:
[image]

nathanmtom, 2023-02-15:
@Robert6codes Thank you for following the template, this is very helpful in parsing out your request and the type of feedback you are looking for. 

In response to your questions here is my responses:

- In WECSim we need to have all hydrodynamic coefficients defined at the center of gravity of any body that is interacting with the fluid domain. In the WECCCOMP model this only includes the float and you are correct that rather than having two separate bodies we could have had a single body that included both float and arm. Given that model was developed over 5 years ago now I might be forgetting exactly why we chose to separate the bodies, my assumption at the moment was we were having some issues achieving the desired draft line of the float and decided to separate the two to troubleshoot the model. Once a satisfactory model was developed we chose not to combine those two components. 
- In terms of XBODY vs XCG, if you were to run WAMIT is XBODY set to 0 and XCG set to the center of gravity WAMIT will output the hydrodynamic coefficients at the 0,0,0 (origin) which corresponds to the still water line. Therefore, for WECSim we keep XCG set to zero and then we colocate the origin of our mesh file with the center of gravity of the body. WAMIT then shifts the mesh to the position defined by XBODY and we get the hydrodynamic coefficients out at the center of gravity. If you look at our WAMIT BEMIO examples, https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/WAMIT, you can see how we've done this for other examples. 
- Re WAMIT inputs, we should have those available but will need to check out records as I could not find them on my initial search. @kmruehl might have quicker access to them than I. 
- Re Simulink set-up: We should be able to but what specific features are you hoping for? 

Please let us know if this has helped answer your question, if not please feel free to follow up with additional questions. 

Cheers, 
Nathan

Robert6codes, 2023-02-16:
Hi Nathan,

Thank you very much for the help! Really appreciated. I will give that a go. I would appreciate it if this issue is kept open for a little longer in case I get stuck again down the line regarding this challenge.

cheers,

Robert

Robert6codes, 2023-02-18:
Hi Nathan,

I hope you are well. Regarding your XBODY vs XCG point, It was my understanding to set XCG to 0 based on your description and the description of the issue above, so therefore on the WECCCOMP.out WAMIT file the corresponding output should be center of gravity = 0,0,0. I have had a look at the examples and this is the case in all the .frc WAMIT file examples. However, in the WECCCOMP.out WAMIT file, the center of gravity is: -0.05, -0.0017, 0. Why is the center of gravity here not 0,0,0? 

Regarding Simulink, I'm a very new user. A general description of the simulink system would be great if possible!

appreciate the help!

Robert

nathanmtom, 2023-02-24:
@Robert6codes Thank you for following up with your questions related to this issue and I'll do my best to answer your question. 

You are correct that the center of gravity listed in the .out file is not 0,0,0 and rather the X,Y coordinates of the center of gravity align with the X,Y coordinates line up with the center of buoyancy. Since we are not defining the center of gravity in the .frc file, then it would make sense that since WAMIT is only getting a mesh file describing the displaced volume of fluid that any center of gravity calculation would align with most of the center of buoyancy calculations, the mismatch in Z is escaping me at the moment and I would need to dive back into the WAMIT manual to find a more accurate reason for the mismatch. 

In the BEMIO parser, only the XBODY coordinates are being read into WEC-Sim to define the center of gravity, which the center of buoyancy calculation is relative to XBODY so the two are added in order to find the center of buoyancy globally. Therefore, the values found in the center of gravity line are not used by WEC-Sim. I will mention that as shown in the image below that the center of gravity and center of buoyancy for the float were offset given the spherical shape is symmetric and there were additional weights placed on the float to shift the center of gravity in front of the center of buoyancy. 

[image]

I was also able to find the WAMIT input files used to generate the WAMIT .out file used during WECCCOMP.

[WECCCOMP_Hydro.zip](https://github.com/WEC-Sim/WEC-Sim/files/10829010/WECCCOMP_Hydro.zip)

Lastly to your question about Simulink, if you want general information reviewing the trainings that Mathworks have will be the best rather than any information we have on how WEC-Sim utilizes Simulink features.

Cheers, 
Nathan