WEC-Sim/WEC-Sim #995: [Theory or Implementation]

dforbush2, 2023-02-15:
This issue is opened on behalf of @Dadidal and is the second half of issue #988 

**Is your question request related to a problem? Please describe.**
About nonlinear viscous damping Fv [https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#viscous-damping](https://github.com/WEC-Sim/WEC-Sim/issues/url).
If I use C_v and C_D coefficients I have to define a 6x6 matrix, right? What velocities are then used to determine these forces in the rotational degrees of freedom (pitch, roll, yaw) - translational or angular velocities? Can I model yaw rate drag using this option? [https://www.orcina.com/webhelp/OrcaFlex/Content/html/Vesseltheory,Currentandwindloads.htm#VesselTheoryYawRateDrag](https://github.com/WEC-Sim/WEC-Sim/issues/url)

Can WEC-Sim account for Manoeuvring load? [https://www.orcina.com/webhelp/OrcaFlex/Content/html/Vesseltheory,Manoeuvringload.htm](https://github.com/WEC-Sim/WEC-Sim/issues/url)

dforbush2, 2023-02-15:
Hi @Dadidal

The linear damping and quadratic drag coefficients are described in the bodyClass. While the resulting forces are always 6 x 6 (assuming you do not have any generalized body modes) for consistency, the linear damping "body.linearDamping" is directly defined as a 6 x 6 matrix, typically users define the cd coefficients body.quadDrag.cd and body.quadDrag.area are each 6 element vectors, which are placed along the diagonal of body.quadDrag.drag.

Alternatively, a user can directly define the body.quadDrag.drag field as a 6 x 6 matrix. This will over write the above definition based on the body.quadDrag.cd and body.quadDrag.area. See lines 289 through 296 of bodyClass.m.

The velocities used to calculate linear viscous or drag forces are the body velocity in the global frame. For the rotational modes, these have units of radians/ second. I also encourage you to explore the Morison Elements options for some more flexibility http://wec-sim.github.io/WEC-Sim/master/theory/theory.html#morison-elements.  

All of the above are implemented equivalently for each degree of freedom, so you can use it to model yaw drag rate if desired. That link you provided has some great info on how to select these coefficients, though I caution you that the underlying theory is not as valid when considered in the context of an oscillating motion like that of a wave field. 

With respect to maneuvering load, by default WEC-Sim does not include these load components. If desired, they could be imposed as an external force sources acting on the floating body constraint.

Dadidal, 2023-02-22:
Thank you for explaining this part. Now these questions are clear to me

closed 2023-02-22 by dforbush2