WEC-Sim/WEC-Sim #996: [Theory or Implementation] Translational PTO not outputting anything

Robert6codes, 2023-02-23:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**

I am using a singular Rotational PTO and when I run simulations there are no outputs of angular position, velocity, acceleration, or torque.

**Describe the theory or implementation approach that you have a question**

I am modelling only the float of a Wavestar in roll. I understand that the rotational PTO by default only has rotation about pitch, not roll. Therefore, I've included: pto(1).orientation.y=[1 0 0] into the input file. My simulink block is extremely simple, shown below, and the subsystem for the seabed is taken directly from the OSWEC simulink model. Within the input file, I have offset the PTO location from the free surface by: pto(1).location = [0 -0.112 0.141];. 

[image]

I have complete all the troubleshooting tests with the following conclusions:
1. The model has hydrodynamic stability
2. Decay tests settle, however the model settles in a strange location. The first image shows the model when ran with no initial displacement and the second shows the model when ran with the displacement in the code line below the second image.

[image]

[image]
command used: body(1).initial.displacement = [0 0 0.03];

3. Including the viscous drag terms does visually have an effect on angular position.
4. The BEMIO file graph outputs all follow the expected trend. Ie, the state space representation being used is correct. 

I have ran the PTO with and without damping and stiffness values. 

**Personal project context**

This is the same project as my previous issue. It is a masters dissertation project. 

**Describe the type of conclusion or resolution you are looking for**

1. An explanation of if just putting the rotational PTO block in the simulink model is enough or do I have to create a new block as is with OSWEC to explicitly output values. 
2. A solution to the above problem

**Additional context**
Thank you very much for the help! Really appreciated.

Robert6codes, 2023-02-23:
Also, these are the output files I currently get. The pto1_out is not present in any of the examples.

[image]

when I open it and open "signals" it looks like this:
[image]

Robert6codes, 2023-02-23:
Apologies, there was in issue with the Simulink model causing this. The simulation gets PTO outputs now however I get the following error when typing wecSim into the command window:

[image]

I found in the issues a user who faced a similar issue. I tried to just rename the userdefinedfunctions script however the problem still persists unlike with issue #741.

Robert6codes, 2023-02-23:
Apologies, there was in issue with the Simulink model causing no outputs. The simulation gets PTO outputs now however I get the following error when typing wecSim into the command window:

[image]

I found in the issues a user who faced a similar issue. I tried to just rename the userdefinedfunctions script however the problem still persists unlike with issue https://github.com/WEC-Sim/WEC-Sim/issues/741.

The issue regarding free decay tests described in the photos above still persists too.

dforbush2, 2023-02-23:
Hi @Robert6codes 

Can you please share your full case directory as a *zip? In particular we will need your *h5 file. I suspect your BEM data is the problem with the settling location you are seeing: since you are looking at roll, a DOF that is not by default plotted in your BEMIO output the issue may have eluded your check above. 

With respect to your error, this is either a path issue or you have added a "clear" somewhere you shouldn't have...perhaps in your 'userDefinedFunctions'?. The projectRootDir variable is created on line 30 of 'initializeWecSim' and _should_ exist until cleared on line 69 of 'stopWecSim'

Robert6codes, 2023-02-23:
Hi there,

thank you for the quick response. Please find attached the zipped document
[Wavestar140run-simplified_model2.zip](https://github.com/WEC-Sim/WEC-Sim/files/10817281/Wavestar140run-simplified_model2.zip)

appreciate it,

Robert

dforbush2, 2023-02-24:
To resolve your bug, remove the 'clear' at the beginning of your WEC-Sim input file.

The reason you are seeing that the orientation of the body at equilibrium is a function of your initial displacement is that you have already constrained the body to be a particular distance from the PTO (the body CG is currently located] unless otherwise specified) 

body(1).centerGravity = [0; 0.1120; 0.1000]; % this is from your *h5 file
pto(1).location = [0 -0.1120 0.141]; % this is from your input file

So when you give an initial vertical displacement to body, Simscape is rotating the body to try and meet both this and the distance constraint between the body and PTO. 

Try adding something like this to your input file to make sure that your initial displcament falls along the radius defined by the PTO position. This suggestion won't get you to perfection but its along the right track!

%% initial disp
hypLength = norm(pto(1).location-[0, 0.1120, 0.100]); % latter vector is body CG
rotAngle = pi/18; % in radians
initAngle = atan(abs(pto(1).location(2))/abs(pto(1).location(3)));
body(1).initial.displacement=[0, -hypLength*(sin(rotAngle + initAngle)-sin(initAngle)), -hypLength*(cos(rotAngle + initAngle)-cos(initAngle))];
body(1).initial.axis = [1 0 0];
body(1).initial.angle = [rotAngle];

Robert6codes, 2023-02-28:
Thank you very much for this!

closed 2023-02-28 by Robert6codes