WEC-Sim/WEC-Sim #997: [WEC-Sim Applications]

davidfdomingos, 2023-02-23:
Dear developers,

**Issue:**
I am running WEC-Sim with hydro data from Capytaine for a floating body in beam waves. However, when I define waves.direction = 90 it produces the following error:

"Error using initializeWecSim (line 224)
waves.direction outside of range of available hydro data"

When I do _waves.direction = pi/2_, it runs smoothly. However, the documentation suggests that degrees should be used:
[image]

**Question:**
Does it really matter the exact units provided to _waves.direction_ as far as it is consistent with the provided hydrodynamic data?

Looking forward to hear from you, and thank you for developing this great open source tool!!

Cheers,
David Domingos

dforbush2, 2023-02-23:
Hi @davidfdomingos 

Within WEC-Sim, all the units of wave direction should be in degrees (this is apparent if you look at waveClass line 431 (among others)). 

However, the headers of the direction table containing the BEM excitation data will be pulled from the BEM output files: for WAMIT, this is input and reported in degrees, but Capytaine input/output these values in radians.

**There is currently an error in BEMIO/example/capytaine/oswec (and maybe others). The readCapytaine script is also wrong because it is expecting degree output, but if capytaine is run properly, this should be radians** 

The error you have highlighted is just checking to see if the waves.direction falls within the bem headings that are used: this is expecting both arguments to be in degrees!

**Use caution with readCapytaine if used for non-zero headings**

We will prioritize this fix.

dforbush2, 2023-02-23:
please used the attached readCapytaine script to recreate your *h5 file that should resolve this issue for you. 

this is included in PR #999 (thanks @akeeste!): awaiting example result updates before merge. 
[readCAPYTAINE.txt](https://github.com/WEC-Sim/WEC-Sim/files/10818556/readCAPYTAINE.txt)

davidfdomingos, 2023-02-24:
> please used the attached readCapytaine script to recreate your *h5 file that should resolve this issue for you.
> 
> this is included in PR #999 (thanks @akeeste!): awaiting example result updates before merge. [readCAPYTAINE.txt](https://github.com/WEC-Sim/WEC-Sim/files/10818556/readCAPYTAINE.txt)

Thanks @dforbush2 and @akeeste, it runs with this script :)

dforbush2, 2023-02-24:
Glad this works. This issue is resolved by PR #999 and will remain open until this is merged.

kmruehl, 2023-03-22:
PR merged, closing the issue

closed 2023-03-22 by kmruehl