WEC-Sim/WEC-Sim #1000: [Feature request] - Versatile Constraint Block

davidfdomingos, 2023-02-24:
Dear developers,

I am currently trying to simulate a float body in beam waves and I want to constraint every DOF except roll and sway. I'm having a hard time to achieve this. I think it would be very useful to have a versatile constraint block that would allow to specify which DOF we want to constraint, in this case the input to this block could be something like:
[0, 1, 0, 1, 0, 0]

Perhaps this could even replace some, if not all, the currently available constraint blocks.

Looking forward to hear from you.

Best Regards,
David Domingos

dforbush2, 2023-02-24:
Hi @davidfdomingos 

Congratulations on being our 1000th issue/PR!

You can assemble 1 rotational constraint in series with a translational constraint and adjusting their rotation orientation: For this specific case, it would look like:

```
constraint(1)= constraintClass('PTO1'); % this one should be rotational
constraint(1).location = [0 0 1] ; % I made up a location
constraint(1).orientation.y = [1 0 0]; % sets rotation axis to about roll

constraint(2) = constraintClass('PTO2'); % this one should be translational
constraint(2). location= [0 0 1]; % again, made up, but should match above
constraint(2).orientation.y = [0 1 0]; % sets translation axis to sway
```

See Simscape multibody documentations for more detail.https://www.mathworks.com/help/smlink/ref/inventor-constraint-simmechanics-joint-correspondence.html;jsessionid=49ad7a7875867f7e3a9b7de871de

davidfdomingos, 2023-02-27:
Thanks @dforbush2, I was not sure if the 1000th issue was reason to celebrate ahah 
Jokes aside, congrats to all the developers for building, maintaining and making this code available to everyone (with access to a matlab license), definitely a stepping stone for research in the offshore field. Keep up the good work!

Indeed this worked, thanks for the hint. I was assuming that series of constraints wouldn't work because the first would filter the info needed for the second. In any case, I think having this generalized type of constraint would simplify the model. If I get some spair time I might look into it.

dforbush2, 2023-02-27:
Hi @davidfdomingos 

Assembling complex joints using a combination of joint primitives is the recommended method of doing this in Simscape multibody. With the provided WS constraints (translational, rotational, spherical, 3DOF and 6DOF floating), one could assemble the joint you propose and many others w/o modifying the block library. 

WEC-Sim will totally work with the joint primitives as well if you need to  just make sure you keep the coordinate systems aligned with WEC-Sim conventions!

closed 2023-02-27 by dforbush2