WEC-Sim/WEC-Sim #1001: [Theory or Implementation] Incompatible SLX files with MATLAB R2022b

h7ossein, 2023-02-27:
**Is your question request related to a problem? Please describe.**
I'm trying to use WEC-Sim with the new version of MATLAB (R2022b). In particular, I'm using the OSWEC example files from the WEC-Sim master, The SLX file for OSWEC seems empty, and when I tried to make a new file according to instructions on the website, I was not successful. I tried to use an old SLX file from a former project of mine, but it gave an compatibility error.

**Describe the theory or implementation approach that you have a question**
N/A

**Personal project context**
N/A

**Describe the type of conclusion or resolution you are looking for**
N/A

**Additional context**
N/A

dforbush2, 2023-02-27:
Hi @h7ossein 

I cannot reproduce this error in 2022b. I suspect what may have happened is that your updated installation of MATLAB does not have the same path configuration as previous versions, and so when loading OSWEC.slx it does not know where to look for the library blocks that model is referencing.

As a quick fix, close all your models and please run `addWecSimSource` from your WEC-Sim source directory. For a longer term solution please see https://wec-sim.github.io/WEC-Sim/master/user/getting_started.html#install-wec-sim

h7ossein, 2023-02-28:
Thank you for the response,
So I just checked and I was using the "Tutorial" files not the "Example" ones, maybe this was the reason, because now I don't have that problem anymore,

closed 2023-03-01 by dforbush2