WEC-Sim/WEC-Sim #1003: [WEC-Sim Applications] body(i).nonlinearHydro = 1 instantaneous body position calculation lag

Robert6codes, 2023-02-28:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**

I have already performed a convergence study on the STL float being used and the model works as expected when running body(i).nonlinearHydro = 2 (sine wave in regular waves) however when using body(i).nonlinearHydro = 1 there is a point midway through the sine wave in which the model freezes and does not follow the expected trend. When running a decay test with body(i).nonlinearHydro = 1 the result follows the exact same path as body(i).nonlinearHydro = 2, it is only when a regular wave is registered that this happens. I have not tried irregular waves.

[image]

**To Reproduce**

Please refer to issue 996 for the full model. The only difference is that the nonlinear hydro terms have been added and any regular wave is ran. The issue is irrespective of the regular wave. 

Once the model is ran, the issue can be seen on an output plot of roll against time. 

**Expected behavior**

As it is a heaving hemisphere, I am expecting more or less a sine wave when running regular waves irrespective of the wave steepness (within reason).

**Screenshots**

Please find a screenshot of the problem above.

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10 
 - MATLAB R2022a, newest Simulink release.
 - WEC-Sim: 5.0.1

**Additional context**

As always I appreciate the help!

Robert6codes, 2023-03-01:
Also, my decay test results between the nonlinear and linear cases are significantly different:

[image]

I'm not sure why there is such a difference. When I run nonlinear waves, sometimes the float jumps slightly up before settling but this is only for steeper wave runs and doesn't happen at wave runs of lower steepness.

dforbush2, 2023-03-01:
Hi @Robert6codes 

I will dig more into the mass calculation, since that may not be acting as intended, but a couple of things right off the bat:

1) When using a wave field, if you want to run w/ nonlinearites, you will want to use body(i).nonlinearHydro = 2 . This includes both non-linear buoyancy and and non-linear froude-krylov forces. Using body(i).nonlinearHydro =1 only uses the non-linear buoyancy but the linear excitation...I wouldn't expect there to be a physically meaningful result in this case. 

2) I would expect a very different free decay for a hemisphere between the linear and non-linear case. The linear case is treating the restoring hydrostatics as a constant value, regardless of the fact that as your hemisphere displaces in heave the surface area is changing drastically. body(i).nonlinearHydro =1 or body(i).nonlinearHydro =2 (they are equivalent in this case since there is no exciting wave to exert non-linear froude-krylov forces) calculates the hydrostatic restoring stiffness at each instant based on the current cut water plane area of the body. I would expect a result like you are showing here, where the apparent stiffness (seen perhaps most clearly in the changing frequency) depends on motion amplitude. 

Will follow up on the mass issue.

dforbush2, 2023-03-04:
Hi @Robert6codes 

Your STL file looks good...you could honestly probably reduce the refinement here to speed the simulation. The way mass is calculated with the 'equilibrium' flag differs with liner and non-linear hydrodynamics. The latter actually looks at the real-time submerged volume and calculates the mass based on that, while the former uses the displacement volume from the BEM estimate (property body.dispVol). This explains the distinct start-up behavior between these run methods, especially when there is a non-zero initial displacement. 

With respect to the expected sinusoid response, try again with a longer wave and maybe some PTO damping. The excitation force seems to be significant enough to cause the body to have some non-linear responses. I would not expect a perfect sine wave forcing a hemispherical body!

dforbush2, 2023-03-05:
@Robert6codes 

I ran your model today with a few different solver configurations using the lower discretization of your model. After some initial stabilization and high-frequency ringing, the expected sinusoid was present. (see figures)

The first few oscillations are related to the body trying to achieve equilibrium with some panels alternating (timestep to timestep) between in and out of the water. This is a solver artifact that is improved with the higher discretization, but it is non-physical: I would recommend you perform your analysis on the portion of data after this high frequency has subsided. Adding some damping will also help! That said, I have reattached your model files as a *zip. I don't believe I made any substantive changes, but everything appears to be working as expected. 

[Wavestar140run-simplified_model2 - Copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/10890495/Wavestar140run-simplified_model2.-.Copy.zip)

Robert6codes, 2023-03-06:
I see! Thank you very much for the guidance, it's really appreciated.

closed 2023-03-06 by Robert6codes