WEC-Sim/WEC-Sim #1004: [Theory or Implementation] Implementation of a Submerged One-Body Point Absorber

mo-meek, 2023-03-04:
**Is your question request related to a problem? Please describe.**
I have attempted to implement a submerged one-body point absorber with a single tether mooring using the cable block. The simulation runs without error, but I am unsure if I have some values defined correctly or if I have included all the needed parameters for it to simulate this WEC correctly as the power plot taken from the ptos.powerInternalMechanics is lower than expected and some location/initial displacement values are not what I expect (I modified values based on the visualizer). 

**Describe the theory or implementation approach that you have a question**
I have not seen prior work done on submerged one-body point absorber modeling in WEC-Sim, or many examples of the cable block implementation. I referenced the WecSimInputFile and Simulink model for both the RM3 and the MBARI (in the Cable WecSimApplications Github). I could not find a way to build the Simulink model with only a float body, translational PTO block, and cable. Therefore, I attempted to combine the two concepts by creating a “dummy” PTO cylinder to create a translational PTO constraint between the float body and the PTO (nonhydro body). Then I connected the “dummy” PTO to the cable with a translational constraint (follower) and created a fixed constraint at the seafloor at [0 0 -100] (base). 

There are some discrepancies between what values I expected to set and what at least looks “correct” in the Mechanics Explorer.
1. The float depth of submergence is supposed to be 6.5m (from the top of the float) and the float height is 5m so I expected to set the body(1).initial.displacement = -6.5 but the float already seems to be submerged without setting this value. 
2. I planned to have the “dummy” PTO cylinder (height of 2.5m) be placed centered at the float’s bottom plane (z=-11.5m). I expected to set body(2).initial.displacement = -11.5 to accomplish this but it seems like -16.5 brings it to be “flush” with the bottom plane of the float. 
3. Similar to above, I expected to connect the cable from the bottom of the “dummy” PTO body which would be 6.5m from the surface, plus 5m for the float height, and another 1.25m for half the PTO height, so -12.75, but -17.75 seems to do this better. 
4. Due to note 3, I also modified the cable(1).length from 87.25 to 82.25 m to match (100-17.75=82.25). 
5. I was planning to set a pretension value, but was not sure what the headers are for the output.cables.totalForce are defined, but it seems like the simulation runs okay without a pretension value. 
6. I saw on the WEC-Sim API page that paraview = 1 can be set for visualization but I do not see anything in the visualizer. 

**Personal project context**
This is for an OSU CE511/411 class project. Our end goal is to take a base model and run a PTO damping optimization using WEC-Sim’s MCR feature. We intend to set the PTO stiffness to 0 N/m and then sweep a range of PTO damping values.
Describe the type of conclusion or resolution you are looking for
I am uncertain if this model is implemented correctly, either in Simulink or the parameters set in the WecSimInputFile based on the output power plot and the trial & error method for setting some of the parameters. Any recommended modifications or clarification on the parameter specifications would be greatly appreciated. 

**Additional context**
I have attached the entire simulation folder with geometry, hydroData, wecSimInputFile and Simulink model. 

Thank you for your time! 

[PA.zip](https://github.com/WEC-Sim/WEC-Sim/files/10887168/PA.zip)

jtgrasb, 2023-03-13:
Hi @mo-meek,

Thank you for your questions and providing the context and files to understand the problem. First, I want to note that we also have a [mooring feature](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#mooring-features) in WEC-Sim which may be more suited for your application. The cable block was originally created to couple two moving bodies, but has the advantage that it can go slack (which a mooring matrix doesn't account for, but moordyn can). 

As for the Simulink model, the constraint between the cable and the pto (body 2) is a translational constraint, meaning it can 
move freely in the z-direction. This can be seen by plotting the cable displacement and force (cable doesn't stretch at all, while PTO body is oscillating). 

[image]

Thus, your cable doesn't stretch at all and the constraint allows for free heave motion instead. I would recommend changing that constraint to a fixed or a spherical constraint depending on your application, which would mean the cable is fixed (or allowed to rotate) to the pto body. 

I'm not sure if this would solve your low power issue though. It is difficult to say what would cause the power to be lower than expected without knowing the full design, but one thing I noticed is that the body mass is set to equilibrium. This may be intentional, but I would expect a design like this to have a smaller mass, creating a buoyancy force which could be offset by the mooring/cable force.

Lastly, for the visualization, the paraview setting simply exports the results to a format which can be used by [Paraview](https://www.paraview.org/), a separate software which can be downloaded. We also have 2 built-in visualization features which may be simpler to use: [Wave Markers](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#wave-markers) and [Save Visualization](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#save-visualization). I would recommend the Save Visualization because it can be easily called from the userDefinedFunctions script as part of post-processing.

Please let me know if this answers your questions!
Best,
Jeff

jtgrasb, 2023-03-30:
Closing due to inactivity

closed 2023-03-30 by jtgrasb