WEC-Sim/WEC-Sim #1005: [Developer Issue]

dforbush2, 2023-03-05:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
Non-linear hydrodynamics and matlab version 2022a compability, potential memory leak. 

**Describe the issue**
When running 2022a , Windows 10, with non-linear hydrodynamics = 2 using a highly discretized body, MATLAB/Simulink appears to create large *.dmr dump files that are not properly removed after the simulation completes. This *.dmr files, for me, end up in /<UserName>/$Recycle.bin, which is by default a hidden folder. These *dmr files can be GB in size for a relatively short simulation. 

Unclear if this is an issue particular to my matlab/hardware setup. If other users, particularly WEC-Sim developers could please inform if they have experienced this issue it might help dial down if this is an issue with our implementation.

kmruehl, 2023-04-13:
@dforbush2 I just ran [this nonlinear hydro case ](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Nonlinear_Hydro/ode45/Regular) from WEC-Sim Applications, with `body(1).nonlinearHydro = 2;`, and was not able to replicate this issue. I'm running 2020b on Windows 10, so not the same env, but close. We aren't [running the applications in 2020a on GitHub Actions](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/workflows/run-tests-dev.yml), but all of the nonlinear hydro tests are passing there. 

Two thoughts:
1) try upgrading MATLAB to a newer version
2) try calculating the [nonlinear hydro forces at a different timestep](https://wec-sim.github.io/WEC-Sim/dev/user/api.html#objects.simulationClass.nonlinearDt), using `simu.nonlinearDt`. I think this may have to do with the size of the data stored each timestep, especially if the STL mesh is "highly discretized"

dforbush2, 2023-05-03:
Thanks @kmruehl for the feedback. Those are great tips for reducing the dump file size, and upgrading versions (also reinstalling) may help these dump files land in the right place and be deleted as intended. Generally this does not seem to be a widespread issue and is contained to my install on this machine.

closed 2023-05-03 by dforbush2