WEC-Sim/WEC-Sim #1006: [Theory or Implementation]On the workflow of WEC-Sim

hachikoi1, 2023-03-07:
Dear everyone
    I'm a bit confused about the starting point of the simulation and the iterative logic of WEC-Sim. 
   At the beginning of simulation, initial position of floating body is specified in input file, and the position is transferred to [_Rigid Transform_]([image]) in [_Sturcture_]([image]) module. So the starting point of the simulation is [_Sturcture_]([image]) module, and initial velocity of floating body is zero by default?
    Another question is, initial acceleration of floating body is always non-zero. How does WEC-Sim get the acceleration at initial moment? and where are position and velocity of floating body updated?

hachikoi1, 2023-03-07:
[image]
[image]

jtgrasb, 2023-03-08:
Hi @hachikoi1,

Thank you your inquiry into using WEC-Sim theory. The initial position and velocity are zero unless otherwise defined, while the acceleration is handled by the dynamics solver.

Also, we are requiring that all inquiries follow the issues template and please answer every section as requested without deleting the section headers so we know which questions on the template are being addressed by your statements. You can either modify your issue here or open a new issue.

[image]

hachikoi1, 2023-03-09:
> Hi @hachikoi1,
>
> Thank you your inquiry into using WEC-Sim theory. The initial position and velocity are zero unless otherwise defined, while the acceleration is handled by the dynamics solver. [...]

Thank you very much for the reminder.

> 

The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:

Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.
Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.

Is your question request related to a problem? Please describe.
About the starting point of the simulation and the iterative logic of WEC-Sim. 

Describe the theory or implementation approach that you have a question
We find the initial displacement of floating body specified by users are the results at t = 0, and velocity is always zero. However, the acceleration is not zero for free decay conditions. **Whether the initial displacement is used as an input or output for t=0? And where the WEC-Sim calculation starts?**

Personal project context

We try to understand the workflow logic of WEC-Sim and the implementation of the ode4 solver

Describe the type of conclusion or resolution you are looking for
I am confused about the calculation at t = 0. I inserted the _scope_ module into the _Hydrodynamic Body_ to detect the acceleration at t = 0 and found that the acceleration input seems to be zero, while the acceleration result is non-zero. In terms of the calculation procedure for _Hydrodynamic Body_, are two calculations performed at t = 0? For the first calculation, acceleration input is zero and only the hydrostatic restoring force is available. the acceleration result is obtained through _Structure_ module in _Hydrodynamic Body_. The twice calculation calculates the additional mass force based on the acceleration obtained by the first calculation. All other time calculations are performed only once.
Another issue is, for the ode4 solver, does WEC-Sim loop _Hydrodynamic Body_ for each update? and therefore all loads that need to be extrapolated are time-dependent?

Additional context
I am confused about the workflow of WEC-Sim and would like to get your guidance. If there is a mistake in the above description, please correct me!

Look forward to hearing from you. Many thanks!

jtgrasb, 2023-03-09:
Hi @hachikoi1, 

Thank you for using the template and your questions regarding the solvers. It sounds like your general understanding is correct. The simulation and dynamics are solved by the dynamics solver Simscape Multibody, which uses the [Solver Configuration](https://www.mathworks.com/help/simscape/ref/solverconfiguration.html) block in Simulink to define the solver settings. This block handles the initial conditions computation, the details for which can be found [here](https://www.mathworks.com/help/simscape/ug/how-simscape-simulation-works.html#bql2kkl-5). WEC-Sim performs a time-domain simulation, meaning the dynamics are solved sequentially at each timestep.

Best,
Jeff

hachikoi1, 2023-03-13:
Dear @jtgrasb
    
    Another question is the calculation of hydrostatic restoring force. I found hydrostatic restoring force contains gravity of floating bodies, and F_NetBuoyancy = F_gravity - F_buoyancy + F_rotational_buoyancy. However, float mass is body.hydroForce.storage.mass but not body.mass, and body.hydroForce.storage.mass is significantly different from body.mass.
So what is body.hydroForce.storage.mass and why not use body.mass?

[image]

jtgrasb, 2023-03-13:
@hachikoi1, 

The body.hydroForce.storage.mass variable is used during the simulation to store the actual value of the mass of the body. On the other hand, the body.mass variable includes an adjusted mass value based on the added mass (more info on this can be found in issues #161 and #308. 

During setup, body.hydroForce.storage.mass is set equal to the mass, then body.mass is changed to account for added mass (thus it can't be used for restoring force). After the simulation is complete, body.mass is set back equal to its original value and body.hydroForce.storage.mass stores the value from the adjusted mass matrix.

jtgrasb, 2023-03-13:
@hachikoi1 

I am not sure the exact terms used in AQWA, but in WEC-Sim you can specify the mass as any value desired. Keep in mind that the body will likely have a non-zero equilibrium displacement with different mass values.

hachikoi1, 2023-03-15:
> @hachikoi1
> 
> I am not sure the exact terms used in AQWA, but in WEC-Sim you can specify the mass as any value desired. Keep in mind that the body will likely have a non-zero equilibrium displacement with different mass values.

Dear @kmruehl
I browsed Issue https://github.com/WEC-Sim/WEC-Sim/issues/161. If I understand correctly, adjustMassMatrix works at the beginning of simulations and restoreMassMatrix works during post-processing, and process results are wrong? When I insert a Scope block to monitor total force in Hydrodynamic Body, I found the monitored total force is not the same as the results in the workspace.
So how can I get the right process results?

Best regards,

hachikoi1, 2023-03-16:
> > @hachikoi1
> > I am not sure the exact terms used in AQWA, but in WEC-Sim you can specify the mass as any value desired. Keep in mind that the body will likely have a non-zero equilibrium displacement with different mass values.
> 
> Dear [jtgrasb](https://github.com/jtgrasb) I browsed Issue #161. If I understand correctly, adjustMassMatrix works at the beginning of simulations and restoreMassMatrix works during post-processing, and process results are wrong? When I insert a Scope block to monitor total force in Hydrodynamic Body, I found the monitored total force is not the same as the results in the workspace. So how can I get the right process results?
> 
> Best regards,

I found monitored F_Excitation, F_RadiationDamping, F_Restoring, F_MorisonAndViscous and F_LinearDmaping are the same as the results in workspace, and only F_AddedMass is wrong, which causes the wrong total force in Scope. So may be I could individually calculated F_AddedMass = obj.hydroForce.storage.fAddedMass*extrapolated acceleration?

jtgrasb, 2023-03-16:
@hachikoi1 Yes, if you'd like to see the correct added mass and total force during the simulation, you can calculate the correct added mass separately as you've suggested.

jtgrasb, 2023-03-30:
Closing due to inactivity.

closed 2023-03-30 by jtgrasb