WEC-Sim/WEC-Sim #1007: MoorDyn Crashing the MATLAB session

aliSHaider, 2023-03-12:
### Is your feature request related to a problem? Please describe.
When MoorDyn is used in WEC-Sim design, it is called as an external DLL executable library, and it runs on a separate console.exe under MATLAB's main process, with its separate process-ID(PID) in the windows task manager. The issue is related to the following scenarios on the Windows platform:
1>> if the lines data text file has an error in it
2>> MoorDyn fails to converge
3>> MoorDyn is launched, but the model has some other error
Under such a scenario, MATLAB fails to terminate the MoorDyn Console and hangs instead, and either crash immediately or crashes if we press the close button on the MoorDyn console window. Either way, there seems to be no way the MATLAB session can recover from MoorDyn failure without crashing. It becomes a major issue if we are trying to tune mooring lines and want to try different lines configurations and stiffnesses, etc., to find an optimal mooring system because some configurations will not work, which will keep crashing MATLAB sessions during the test and trial. 

### Describe the solution you'd like
The goal is for MoorDyn's failure not to crash the MATLAB session but instead MoorDyn dll execution to close and the MoorDyn console to close. I've found a solution as described in the next section.
### Describe alternatives you've considered
For scenarios 1 and 2, the following Fix#01 seems to work on the Windows platform. And to cover scenario 3, an additional Fix#02 is needed.
### Proposed Fix#01
In the MoorDyn block's 'InitFcn' call, making the following suggested changes check for MoorDyn Initialization failure and terminates the MoorDyn.dll console without crashing the MATLAB Session.
The idea is to keep a check on the returned value of the Lines.dll and kill MoorDyn's Conhost.exe process if the return value is less than 0 (an error scenario). This is done by identifying the PID of the most recently launched Console.exe process and terminating it.

**if libisloaded('Lines')
    calllib('Lines','LinesClose');
    unloadlibrary Lines;
end

moorDynError=0;
tim=clock; % record current time
tim=[tim(4) tim(5)];% hours minutes
PIDmoordyn=[];
pros = System.Diagnostics.Process.GetProcessesByName('conhost'); % get info for process named 'conhost'
for i=1:pros.Length
    % total minutes since process started
    tp=[(tim(1)-pros(i).StartTime.TimeOfDay.Hours)*60 + tim(2)-pros(i).StartTime.TimeOfDay.Minutes] ;
    if(tp<=1)
        PIDmoordyn=pros(i).Id  % the most recent (process started less than 1 minute ago) is moorDyn
    end
end
clear pros tim tp

loadlibrary('Lines.dll','MoorDyn.h');

if calllib('Lines','LinesInit', [0 0 0] ,[0 0 0 0 0 0])<0
    moorDynError=1
    unloadlibrary Lines;   % a stuck window does not execpt close line library call
    eval([strcat('!taskkill /f /pid',32,num2str(PIDmoordyn))]) ; %forcefully Kill task with PID
    disp('-------------Error: Mooring Line Initialization failed in MoorDyn Block')
    disp('------------MoorDyn Console Window is forcefully closed by MoorDyn Block ')
end

try
    if moorDynError==1
        if ispc
            eval([strcat('!taskkill /f /pid',32,num2str(PIDmoordyn))])
        end
    end
end**

[image]
 

### Proposed Fix#02
Suppose the MoorDyn console has been launched, but there is an error in the Simulink model. In that case, MATLAB throws an error and terminates the Simulink execution without terminating the MoorDyn console window. A try-catch statement in the wecsim file can address this issue. If a Simulink model error occurs, we can print the error and terminate the MoorDyn Console process.

**try
sim(simu.simMechanicsFile, [], simset('SrcWorkspace','parent'));
catch e %e is an MException struct
    fprintf("------Error in the Model, Causes are :\n");
    for i=1:size(e.cause)
        fprintf(e.cause{i}.message);
        fprintf('\n');
    end
    if ispc
    eval([strcat('!taskkill /f /pid',32,num2str(PIDmoordyn))])  % kill moordyn console window
    end
 end**

**Personal project context**
I'm working on modeling and control design for the WECs with initial testing of the plant simulated in WEC-Sim.

**Additional context**
N/A

dav-og, 2023-04-05:
This issue has been addressed by @AlixHaider 's PR: https://github.com/WEC-Sim/WEC-Sim/pull/1012

closed 2023-04-05 by dav-og

treppe8k, 2023-04-21:
I am using Matlab 2020b, when i try install WecSim last version, I got this error

####################

  .>> slLibraryBrowser
Error using load_system (line 21)
Model 'WECSim_Lib_Moorings' was created with a newer version (R2022a) of Simulink
To disable this error message, use Simulink preferences.
To create a model that is compatible with this version of Simulink, load the model in Simulink
R2022a and select Save > Export Model to > Previous Version.

####################

You can try  re-export the 'WECSim_Lib_Moorings.slx' with the the MatLab 2020b version ?
According  to guide , that is the version in use for WecSim 5.0.1

thanks in advance