WEC-Sim/WEC-Sim #1008: [WEC-Sim Applications] - Results not matching RAOs from hydro data

davidfdomingos, 2023-03-13:
**Describe the bug or issue**
I have computed hydro data using capytaine and provided it to WEC-Sim. When looking at the results, the sway  and roll motions from WEC-Sim do not match the RAOs computed from the provided hydro data. 
I'm considering beam waves. 
The floating body is only allowed to roll and sway.

**To Reproduce**
1 - Run the script 'Read_ncResultsFile_v1.py'  in WECSim_RAOIssue_Git_v1\Auxiliary, to provide you with .csv files containing the RAOs in sway and roll; from the hydro data provided to WSim. Determine the RAO for both motions for the frequency of 0.52 [rad/s]
2- Open the folder WECSim_RAOIssue_Git_v1 in matlab and type 'wecSim' to run it for a case of regular waves at 0.52 rad/s and 1 m amplitude (2m high) 
3 - The resulting plots will show you the comparison between the WSim results and the respective RAOs previously computed

**Expected behavior**
That the amplitude of the motions caused by 1 m amplitude waves match the RAO results for the respective frequency

**Screenshots**
[WECSim_RAOIssue_Git_v1.zip](https://github.com/WEC-Sim/WEC-Sim/files/10960264/WECSim_RAOIssue_Git_v1.zip)

[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release] R2021b
 - [WEC-Sim Release]-Up to date with the master branch.

nathanmtom, 2023-03-29:
@davidfdomingos Thank you for posting your question and I apologize for the delay in our response. 

I have had a moment to download your .zip file and independently verify the RAO calculation and WEC-Sim's performance above. I do not have an immediate answer, but hope to have an update within the next week. 

I will acknowledge that the sway response is of higher priority as the roll response is on the order of 0.00x so the slight mismatch is likely from slight numerical discrepancies. But have you selected a different wave period and observed larger discrepancies in the roll response amplitude operator?

Cheers, 
Nathan

nathanmtom, 2023-04-10:
@davidfdomingos Thank you for your patience as I believe I've found the source of the miss match in outputs. 

I believe the mismatch is from estimating the RAOs using the 6 degree of freedom response against the 2 degree of freedom response (sway and roll). In your Capytaine run you us the 'cpt.post_pro.rao' function which I believe is calculating the RAO assuming all 6 DOF are oscillating which leads to lower RAO values. When I quickly reduced the 6DOF to a 2DOF equation of motion, a result of your constraints, then the final values are greater that the 6DOF and align much better with WEC-Sim's outputs (see image below). After comparing the amplitudes from the reduced degree of freedom model against your WEC-Sim results, I am content that the results are nearly the same and there is no bug in WEC-Sim.  

[image]

To help you navigate the RAO discussion, I am attaching a custom script that I used to complete the reduced order RAO and compare against the full 6DOF RAO calculation. 

[bemio_1008.zip](https://github.com/WEC-Sim/WEC-Sim/files/11194264/bemio_1008.zip)

Please review and let me know if you agree with this conclusion. 

Cheers, 
Nathan

davidfdomingos, 2023-04-21:
Hi @nathanmtom, 
First of all thank you for your detailed explanation and patience. I want to go through the script you provided. However, I'm having very limited time at the moment and only next week I will be able to look into this. I understand if you want to close the issue in the meantime. If there is anything relevant coming up I can contact you and the issue can be re-opened.
Cheers,
David

nathanmtom, 2023-05-01:
Thank you for the response @davidfdomingos, we will keep this issue open for the rest of the week but if we have not received a response back we are going to close the issue. 

Cheers, 
Nathan

davidfdomingos, 2023-05-05:
Hi Nathan,

I was now having a closer look at your results and script. Indeed they become closer to what I was obtaining. Can we say that this difference is caused by disregarding cross-coupling effects (e.g.: B45) between DOF when constraining the body to only sway and roll?

Cheers,
David

nathanmtom, 2023-05-05:
@davidfdomingos Thank you for following up with your issue, I do believe that the difference is a result of constraining degrees of freedom which eliminates certain cross-coupling effects. If it is specifically the B45, or roll-pitch damping, I am not sure as in a beam sea run where you are assuming the ship only sways and rolls I wouldn't anticipate much pitch motion. However, I did not look particularly close to the values of the cross coupling terms and it could be from B45.

Cheers, 
Nathan

davidfdomingos, 2023-05-08:
Interesting to see the significance of the cross-terms in such conditions, I didn't expect it. I have no further questions for now, thank you for looking into this @nathanmtom! Feel free to close this issue.

Cheers,
David

nathanmtom, 2023-05-08:
Appreciate the follow-up @davidfdomingos, I'll close this issue but please reach out with other questions you might have.

closed 2023-05-08 by nathanmtom