WEC-Sim/WEC-Sim #1010: [BUG] Unrecognized method, property, or field 'plotEta' for class 'waveClass'."

sung2799, 2023-03-15:
**Describe the bug**
I had reinstalled WEC-SIM (due to computer crash) and am now unable to fully run any previously working simulations. It runs partially, and I get the output file before the error occurs, so the error seems to occur after the output data structure is created.
I'm running on MATLAB R2022a. It looks like something to do with the 'slx' file, though I have a pretty simple model. I've attached a screenshot of the simulink file, and cut and paste the whole error message:
"Error using wecSim
Error evaluating 'StopFcn' callback of Frames block (mask) 'test/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
Error using wecSim
Unrecognized method, property, or field 'plotEta' for class 'waveClass'."

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._
Based on the error message, I "suspect" it's due to the 'slx' file, but not sure.
[image]

**To Reproduce**
_Steps to reproduce the behavior:_
Simply running newly installed WECSIM, using scripts and geometries had run in a previous version of WECSIM without problem.
_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
Running WEC-SIM in MATLAB R2022a using command line.

**Expected behavior**
I expected it to fully run, providing the usual graphs and animation.  Instead, I get an output variable which is good, but an error message occurs and the run stops.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]
- OS: Windows 10 Enterprise
- MATLAB R2022a
- WECSIM: WEC-Sim v5.0.1

dforbush2, 2023-03-15:
Hi @sung2799 

I appreciate you re-opening this issue. When you reinstalled WEC-Sim I see v5.0.1, but it is probable that you previously had v4.4 installed. v5.0 was a major code overhaul a number of functions were re-named and re-organized. 'plotEta' does not exist any longer, and was replaced with 'plotElevation'.

Changing this function call in your userDefinedFunctions will solve this issue. You may also wish to revise previous wecSimInputFiles to the v5.0.1 naming convention. Please see the relevant release notes and review documentation: most of these changes are minor but greatly help code consistency and readability.

sung2799, 2023-03-15:
Thx Dom - I had noticed the new naming convention and had updated the script, but I'm not sure where "plotEta' for class 'waveClass'" was accessed so that I can address this issue.  In my wecSimInputFile.m, I have a simple assignment of 
waves = waveClass('regular');           % Initialize Wave Class and Specify Type                                 
waves.height = 0.75*2*(0.0254);        % Wave Height [m] (peak - trough)
freq_Hz=.67;                           %Resonance 
waves.period = 1/freq_Hz;  

There is no mention of 'plotEta' and I'm not sure where else to look,  This machine is new since the last one crashed and died, so it should not have any legacy v4.4 on it.  Thanks again for the help on this.

Li

sung2799, 2023-03-17:
Dom - thank you.  found the culprit functions and updated them to v5.0.1, and I'm running again.