WEC-Sim/WEC-Sim #1013: [Theory or Implementation] How can I validate the weakly non-linear solution?

ameduri2312, 2023-03-16:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_How do we validate if our weakly non-linear solution is correctly being evaluated compared to the linear_

**Describe the theory or implementation approach that you have a question**
_I am building a weakly non-linear model similar to the one in the examples. I have created a .stl file of a  cylinder with around 5000 panels and attempting to simulate regular wave cases. However, in most cases, the weakly non-linear amplitude is 2 times the linear one, which I am not sure is correct. I would like to understand how to validate this method, or if I am making any implementation mistake in my code _

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._

**Describe the type of conclusion or resolution you are looking for**
_I would like to understand how the non-linear solution is validated and if there is an error in my implementation that is giving such a large discrepancy ._

**Additional context**
_Both attached pictures are for H = 2m, T = 5s. My input files can be found in the github link https://github.com/ameduri2312/WEC_Sim.git_

[image]
[image]

dforbush2, 2023-03-20:
At present there is a potential bug in the Non-linear froude-krylov implementation that is giving non-physically large amplitudes in some cases. A fix is in progress

dforbush2, 2023-03-22:
We are planning to revert #910 which introduced this bug. Until this is done, please modify "nonLinearBuoyancy.m" to the version that was present prior to this merge. This will largely improve your non-linear hydro results. In long waves, this should look quite similar to the linear case.  

Reviewing your model, I have one more recommendation. Your *stl file appears to have been made from your BEM mesh, and includes only panels that are at or below the free-surface (I am assuming this because the cylinder appears truncated at the free surface: it is possible this is the intended geometry, in which case, please disregard). When running non-linear hydro, because panels that are initially above the free-surface may, due to body displacements, contribute to the non-linear buoyancy and Froude-Krylov forces, you may consider modifying your *stl file to include some of the geometry that is above the nominal free surface.

akeeste, 2023-04-05:
Hi @ameduri2312

Do you have any more questions about this bug or @dforbush2 's suggestion for the mesh? If not, I will close this issue at the end of this week.

nathanmtom, 2023-04-19:
Closing issue due to inactivity.

closed 2023-04-19 by nathanmtom