WEC-Sim/WEC-Sim #1014: [BUG] 6 DOF constraint

hachikoi1, 2023-03-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
**Dear all,
For free decay cases, e.g. /WEC-Sim_Applications-master/Free_Decay/1m, I specified a regular wave condition and replace 3 DOF constraint with  6 DOF constraint. Why output.bodies.position(:,2:2:6) are alway zero? And why output.bodies.position(:,1) always oscillate around x = 0, rather than gradually moving in the positive direction of x?**

**WEC-Sim file(s)**

**To Reproduce**

**Expected behavior**

**Screenshots**
[image]
[image]
[image]
[image]
(Figure 4: bodies.position(:,1))
**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 11]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2022a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v5.0.1]

**Additional context**

kmruehl, 2023-03-22:
@hachikoi1 it looks like your comment was edited and accidentally deleted. Can you please update it to include the original post?

hachikoi1, 2023-03-23:
Dear [kmruehl](https://github.com/kmruehl)

I found it's mu fault. Could you please delete this post?

Best regards,

kmruehl, 2023-03-24:
Will do, thanks!

closed 2023-03-24 by kmruehl