WEC-Sim/WEC-Sim #1015: [Feature request]

dforbush2, 2023-03-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
A request from the WECOptTool team. Many developers have experimental data or identified dynamic systems representing their WEC that are in the form of intrinsic impedance models, which represent the N x N input/output relationship between velocity and total force. These models can be calculated from information in BEM output data and are effectively a summation of these constituent forces, but can also be derived in other ways that may not have resolved the individual terms (added mass, radiation damping, etc) in a manner sufficient for a WEC-Sim input. 

**Describe the solution you'd like**
A version of the "body" block that calculates dynamics based upon a model of intrinsic impedance rather than BEM data. 

**Describe alternatives you've considered**
I personally made custom blocks to accomplish this, but had to remove much of the model from the WEC-Sim environment to do so. There is not a streamlined workflow to accomplish this. 

**Personal project context**
A request from the WECOptTool team. This will make GB very happy.

kmruehl, 2023-05-17:
closing the internal feature request since it's now listed on the project board

dforbush2, 2023-07-26:
@salhus and @dforbus have considered and discussed this development. Within the WEC-Sim environment this is difficult to do. The model must actually calculate based on admittance (1/impedance) to properly handle excitation force. With an output velocity, this would then need to drive a custom 6-dof actuator, and correct for the connected body inertia (which would otherwise be double-counted since it is included in the admittance model). The determination of the total (non wave) forces exerted on the admittance body would depend on how it is connected to other elements: i.e., we would need a separate implementation for each case (for most users this might only be 2 or 3).

dforbush2, 2023-07-28:
An additional thought: the latter proposed approach also opens the possibility of using a "hybrid" h5 data structure where some DOFs are informed by BEM and others informed by an impedance/admittance model. This is likely practical: most experimental SID efforts are only focused on a subset of the degrees of freedom. 

What is not, however, encompassed by a body-type impedance block is relative-motion type impedance models often critical to PTO design for multi-body systems (for example see https://link.springer.com/article/10.1007/s40722-021-00197-9).

salhus, 2023-07-28:
@dforbush2 Thanks for that paper, seems to be very helpful. I will read through it.
The other thing (though trivial) I thought was that since the joint blocks (PTO/Constraint) are being actuated by a prescribed motion, one could just use 6DOF joint blocks and prescribe 0s as motion for prescribed DOFs. A standardised approach will then always use 6DOF joints and prescribe motion as necessary. 
It could be implemented something like a WAMIT's flag enable scheme by setting joint DOFs as a matrix of ones, e.g., [0 0 1 0 0 0], will represent a joint that allows relative heave motion. The block can be set to display this flag matrix as a caption, so the user can still look at the model and infer what the joint block is doing.

In my head, an implementation looks something like this, 
1. The model is a daisy-chain of joints where each joint (say, $J_n$) connects to some Solid Block ($S_n$) and the next joint block ($J_{n+1}$),
2. The kinematics of $J_n$ will be the prescribed motion. Looking at the paper you showed, the force coming from the sensor of the joints can be added to the excitation force from the existing body block. Now since the joint blocks are 'sensing' relative motion of the $J_{n}$ and $J_{n+1}$, we should probably 6DOF constraint blocks between world and the $S_n$ frame. This won't add any new forces but will allow to measure the absolute kinematics.

closed 2023-08-09 by dforbush2