WEC-Sim/WEC-Sim #1019: [Theory or Implementation] Hydrostatic and gravitational restoring coefficients

hachikoi1, 2023-03-26:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
Dear all,
1. In _Linear Hydrostatic Restoring Force_ BLOCK, F_NetBuoyancy = F_gravity - F_buoyancy + F_rotational_buoyancy. However, in bodyClass.m, I also found `obj.mass = obj.hydroData.properties.volume * rho;`. So essentially F_NetBuoyancy = F_rotational_buoyancy, right?
2. I found WEC-Sim calculate mass based on MESH BASED DISPLACEMENT. For a floating wind turbine platform, specifically DeepCWind, we only know the overall draft (includes platform, turbine and mooring), and the mass and COG of platform. How should I set the platform parameters in AQWA/WAMIT to get the right results? If I use overall draft, the platform mass calculated based on mesh displacement is not what I want.

**Describe the type of conclusion or resolution you are looking for**
For question 2, should I modify the platform draft, or the platform mass and COG in AQWA? Subsequently I would like to add wind turbines to the platform.

hachikoi1, 2023-03-28:
> **The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
> 
> 1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
> 2. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
> 3. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
> 4. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.
> 
> **Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**
> 
> **Is your question request related to a problem? Please describe.** Dear all,
> 
> 1. In _Linear Hydrostatic Restoring Force_ BLOCK, F_NetBuoyancy = F_gravity - F_buoyancy + F_rotational_buoyancy. However, in bodyClass.m, I also found `obj.mass = obj.hydroData.properties.volume * rho;`. So essentially F_NetBuoyancy = F_rotational_buoyancy, right?
> 2. I found WEC-Sim calculate mass based on MESH BASED DISPLACEMENT. For a floating wind turbine platform, specifically DeepCWind, we only know the overall draft (includes platform, turbine and mooring), and the mass and COG of platform. How should I set the platform parameters in AQWA/WAMIT to get the right results? If I use overall draft, the platform mass calculated based on mesh displacement is not what I want.
> 
> **Describe the type of conclusion or resolution you are looking for** For question 2, should I modify the platform draft, or the platform mass and COG in AQWA? Subsequently I would like to add wind turbines to the platform.

For a WEC with mooring, I think F_buoyancy = F_gravity + F_mooring. That is, F_buoyancy =  volume * rho * g, but F_gravity is directly readed from frequency-domain results: 
```
if (isempty(strfind(raw1{ln},'HYDSTIFFNESS'))==0 |...
            (isempty(strfind(raw1{ln},'MASS'))==0 & isempty(strfind(raw1{ln},'ADDEDMASS'))==1))
        if isempty(strfind(raw1{ln},'HYDSTIFFNESS'))==0 f = 0; else f = 1; end
        for j=1:hydro(F).Nb
            for i=1:6
                tmp = str2num(raw1{ln+(j-1)*6+i});
                if length(tmp)>6 tmp(1)=[]; end
                if f==0
                    hydro(F).Khs(i,:,j) = tmp; % Linear restoring stiffness
                elseif f==1
                    M(i,:,j) = tmp; % This is the mass we want
                end
            end
        end
    end 
```

jtgrasb, 2023-03-30:
Hi @hachikoi1,

Thank you for your question. It seems you have the correct understanding. WEC-Sim uses the equilibrium setting for mass by default, which is applicable for some cases but definitely not all. As you mentioned, the 'equilibrium' setting only accounts for forces of gravity. With a mooring pre-tension, a different mass is required to balance the forces on the body. If you know what the mass actually is, then you can set the mass to that value in WEC-Sim. If you only know the overall draft and mooring pretension, you can balance the forces (F_gravity, F_buoyancy, and F_mooring) to determine the correct mass. 

Best,
Jeff

closed 2023-04-17 by jtgrasb