WEC-Sim/WEC-Sim #1020: [Theory or Implementation] How to create h5 file with mean drift forces?

Dadidal, 2023-03-30:
**Is your question request related to a problem? Please describe.**
I do not understand how to create a hydrodynamic data file (*.h5) with mean drift forces.

**Describe the theory or implementation approach that you have a question**
Firstly I calculated BEM data in ANSYS Aqwa. This BEM data includes mean drift forces. I know that according to documentation, WEC-Sim can read the mean drift forces only from WAMIT. Therefore I used BEMRosetta to change the BEM output format: I imported results from AQWA and then exported them to WAMIT format. Then I used BEMIO to read "WAMIT" *.out file and tried to simulate my model with "body(1).meanDrift = 1" option but contained an error:
```
Error using h5readc
The HDF5 library encountered an error and produced the following stack trace information:

    H5G__traverse_real    component not found

Error in h5read (line 93)
[data,var_class] = h5readc(Filename,Dataset,start,count,stride);

Error in readBEMIOH5 (line 98)
    hydroData.hydro_coeffs.mean_drift =  reverseDimensionOrder(h5read(filename, [h5BodyName '/hydro_coeffs/mean_drift/control_surface/val']));
```
It seemed that my h5 file does not contain mean drift forces even though they exist in BEM data. I checked h5 file and observed that all matrixes of mean drift forces have 0 values.
Could you give some advice on how to fix it?
I have attached the following files (BEM data.zip):
1. BEM output from ANSYS Aqwa
2. Converted BEM data to WAMIT format
3. Generated .h5 file

**Personal project context**
I am trying to include 2nd order wave forces in my simulation

**Describe the type of conclusion or resolution you are looking for**
I want to clarify and understand what I should do in order to include mean drift forces in WEC-Sim simulation

**Additional context**
[image]
[BEM data.zip](https://github.com/WEC-Sim/WEC-Sim/files/11108894/BEM.data.zip)

nathanmtom, 2023-04-07:
@Dadidal Thank you for your issue and I'll do my best to provide options to resolve the problem.

- First off, I want to let you know that we do have an applications case with an example of the .out file format from WAMIT with mean drift. https://github.com/WEC-Sim/WEC-Sim_Applications/blob/master/Mean_Drift/WAMIT/outputFiles/sphere.out
- A quick comparison between the two output files shows that the headers for the pressure integration option, your file has an extra 'PRE' at the end. While in our 'readWAMIT' script on line 222 we are looking for header without that which is why the parser is filling the val with zeros. 
- I've only briefly looked at the rest of the output format, but perhaps if the other variables are importing correctly then it is likely a formatting issue in your .out file.

If you do not mind attempting to make the section headers consistent with said format and rerun 'readWAMIT' that will let us know if that was the issue or if somehow the parser is still miss identifying data in the .out file.

Dadidal, 2023-04-12:
@nathanmtom Thank you for your response. I deleted 'PRE' word and also checked other formatting issues regarding headings and didn't find any. But the problem is not solved. Parser still cannot identify mean drift forces im my BEM data. All values are zero even after modifying headers. Here I attach another BEM data that was transformed to WAMIT .out format just for the case.
[ANALYSIS.zip](https://github.com/WEC-Sim/WEC-Sim/files/11208312/ANALYSIS.zip)

nathanmtom, 2023-04-24:
@Dadidal Apologies for the delay in my response, but after looking at the code it does appear that the mean drift forces are being read in my the readWAMIT.m file, but upon writing the .h5 file and importing prior to the run the 'pressure integration' implementation is not being saved. This should be a straight forward implementation to update a few files and hope to have files to share with you by the end of the week before submitting a pull request. However, let me know if there have been any changes to this issue in the last few weeks.

nathanmtom, 2023-04-27:
@Dadidal PR #1046 is in response to this issue. If you review the changes implemented you should be able to test if this now gives you nonzero outputs for the mean drift forces. Another team member will need to verify the implementation prior to being pulled into WEC-Sim, so hopefully in the mean time you can implement the proposed changes to see if your issue is resolved.

nathanmtom, 2023-05-05:
@Dadidal PR#1046 has been merged into the 'Dev' branch of WEC-Sim. Please confirm that the new implementation is now allowing you to import you mean drift forces.

Dadidal, 2023-05-07:
@nathanmtom I checked the new implementation and now I can import mean drift force coefficients. Thank you!
Also just for clarification do you plan to add import of mean drift forces from another BEM solvers (ex. NEMOH 3.0)?

nathanmtom, 2023-05-07:
@Dadidal Thank you for confirming the changes now allow you to import the mean drift force coefficients for the pressure integration method. We are aware that the latest version of NEMOH does have the option to now output the mean drift forces and hope to update the readNEMOH.m in the near future to read in those forces as well. I am going to close this issue now, but please reach out if you have any other issues with the mean drift implementation.

closed 2023-05-07 by nathanmtom