WEC-Sim/WEC-Sim #1025: [BUG] PTO.damping command not registering when using nonlinearhydro

Robert6codes, 2023-04-06:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I am comparing nonlinear models to linear models to a physical experiment and I believe that damping is not being applied when running the nonlinear models. I am using the PTO(1).damping command, and when running with a given damping (to represent friction in this case), this plot below is produced which compares the linear model, the nonlinearhydro = 1, and the nonlinearhydro = 2 model to a physical experiment, 'real'.

[image]

However, when running 0 damping, the nonlinear models output doesn't seem to change drastically

[image]

I've checked my input file and it runs fine and is very similar to the nonlinearhydro examples inputs. 

**please not I did make a mistake when running this trying to implement coulomb friction, anyhow I used body().lineardamping but its my understanding that if selecting the correct mode of motion it should have the same effect at PTO().damping. furthermore, it is constant throughout all of the models where damping is applied.**

**WEC-Sim file(s)**
I've checked my input file and it runs fine and is very similar to the nonlinearhydro examples inputs. Will upload the files in the comments shortly.

**To Reproduce**
_Steps to reproduce the behavior:_
Just run the model and you should see the same issue.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
I am running the model from the input file, not from simulink.

**Expected behavior**
Given a damping, I would expect the RAO to reduce for all models, not just the linear case.

**Screenshots**
Please see above.

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10
 - 2022a MATLAB
 - [WEC-Sim Release: 5.0.1 I believe, if not the second most recent. Installed the software December 2022]

**Additional context**
_Add any other context about the problem here._

Robert6codes, 2023-04-06:
Hi, 

I have just read issue 1013 and understand this issue may be related considering the graphs outputs are very similar between mine and the user of the issue in 1013 for. However, is the bug exclusive to if damping is applied? The WEC-Sim version I am using is identical between the production of both the graphs 

thank you for the help, highly appreciated as always :)

nathanmtom, 2023-04-07:
@Robert6codes Thank you for your issue, I have the following questions and requests for you to help with this issue.

- The bug with the nonlinear hydro was reverted back on March 22, so if you have not pulled the latest version of WEC-Sim please do. 
- The impact of damping will be dependent on the size of the oscillating body and damping you are using. In the previous issue, the difference between no damping and being overdamped should impact the phase between the body response and wave elevation making the impact of the nonlinear hydro term clearer. 
- In the text above you mention attaching your WEC-Sim files to assist us in debugging, but they do not seem attached. Please do attach them or else we will need more information on the exact model you are running, .stl files, etc.

Robert6codes, 2023-04-08:
[Wavestar140run_rightI - Copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/11183075/Wavestar140run_rightI.-.Copy.zip)

Hi Nathan,

Please find the file attached. 

I have just downloaded the newest version and I am experiencing another strange result. The linear model has an RAO 4x that of the nonlinear model. 

Also, when running nonlinear hydro 2, if I set the body(1).mass to equilibrium I get a certain RAO output. If I then take from the simulation report the mass that the 'equilibrium' command calculated the WEC to be and implement that directly as body(1).mass = *copy and pasted value from simulation report*, then I get a different RAO output. 

I am also still experiencing a drastic sudden 'jump' when running nonlinear hydro 2 or 1 compared to if running the purely linear model. I understand this can happen because its ultimately calculating hydrodynamic stability differently, however I was not expecting it to this extent.

To my understanding, I've gone through all the checks such as making sure that the state space model is represented correctly.

cheers

nathanmtom, 2023-05-01:
@Robert6codes Apologies for the delay in my response, but here is my update to your questions.

- For `equilibirum` mass when you are using the linear option, the mass is equal to the displaced volume times the fluid density. If you are using either of the `nonlinearHydro=1 or 2` the mass is equal to the integrated volume of the .stl file you are using. In your case for the linear option the mass is 0.3509 while for the nonlinear case the mass is 0.3034. Because of the differences in mass I expect there to be differences in RAO, thus no bug.
- Linear vs nonlinear model has an RAO of approximately 4x greater is also on the same order of magnitude I got running with your pto(1).damping value of 0.399; however, when running with a damping of say 100*0.399 the answer is much closer which is likely do to the undamped simulation resulting in much larger motion amplitudes that are corrected when using the nonlinearHydro option, thus no bug and WEC-Sim appears to be working as expected. If you took the wave amplitude very small, I would expect the two solutions to converge and could be an additional check to make sure your model is working as expected.
- No impact of damping on simulation appears to be because you went from 0.399 to 0, which is already a fairly underdamped system. In the two images below, I ramped up the damping to be 100 times the 0.399 value you included in your example files. Based on the results the increased damping does reduce the amplitude of motion as expected. 
 
[image]
[image]

- For the upstart transient of the nonlinearHydro simulations, it appears that when you select `equilibrium` the expected mass is smaller than desired causing extra buoyancy that is likely causing the rotation to move up out of the water in the positive direction. This seems to be something you will want to check within your .stl file to see if this can be corrected based on the initial placement in the water.

I believe these answers address all of your comments, but please let me know if I missed anything. Based on the above comments I do not see any bug within WEC-Sim and the code is working as coded. 

Please let us know if you still have questions about the simulation you are running.

Cheers, 
Nathan

nathanmtom, 2023-05-03:
@Robert6codes Thank you for reviewing my responses and providing additional information about your particular case. My responses and feedback are below:

- I agree that the magnitude of the damping is dependent on body weight and other mass properties so high or low is situation dependent. The implementation of a very large damping was to illustrate that the pto(1).damping is influencing the simulation. I really cannot comment on if the case you've built that would be underdamped or damped, but the results do show that the motion amplitude for the 100x cases have about 0.001 which is incredibly small and without some static friction any wave will generate some motion. 
- Re Linear model in OceanEd and WEC-Sim: I am having difficulty observing the differences as the image you provided above does not include WEC-Sim L, where I assume L=linear. 
- Re natural period: If WEC-Sim linear is having a natural period that is twice that of the experiments, then either the added mass is too large or the hydrostatic stiffness is to small. For a second order system the natural frequency is proportional to the square root of spring over mass. If the hydrostatic restoring coefficient was over defined (i.e. too large this should actually reduce the natural period). But perhaps I am interpreting your response incorrectly. 
- The added mass and the hydrostatic coefficients in the hydrodynamic model can both be influenced by incorrectly setting the center of gravity. It would be worth double checking these parameters. 

Apologies as I am finding it difficult to provide more checks on the WEC-Sim side to try and get to the bottom of the mismatch in response between WEC-Sim, OceanEd, and experiments. But please let us know if you have questions about variables, coordinates, etc. that may help in your troubleshooting.

Cheers, 
Nathan

Robert6codes, 2023-05-03:
Hi Nathan,

thank you very much for your ongoing help, I really appreciate it. Unfortunately my project finished a while back but I will store these responses should future work be conducted to let a future user know to pick up where I finished. I did struggle with getting the centre of gravity in the right spot on the STL, I think it was correct as it took me a few goes however could indeed be the core cause given the last point.

cheers again,

Robert

closed 2023-05-03 by Robert6codes