WEC-Sim/WEC-Sim #1026: [Theory or Implementation] Implementation of Coulomb friction

Robert6codes, 2023-04-06:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**

I am trying to reproduce the friction from a real experiment in WEC-Sim using the Tustin model. However, I cannot figure out how to implement coulomb friction correctly into my model. I believe a method to do so would be through PTO translational Torque block, however other than that I am rather stuck on how to do it. Is there a method that I am missing?

**Describe the theory or implementation approach that you have a question**
Please see above.

**Personal project context**
Please see above. It is for a dissertation project.

**Describe the type of conclusion or resolution you are looking for**
Just a method to do it. I've seen an academic paper mention they did it using the previously mentioned block and was hoping the WEC-Sim team had further insight.

**Additional context**

thank you!

Robert6codes, 2023-04-06:
For reference, it is this paper that I am referring to (from WECCCOMP) and I notice some of you are authors:

https://doi.org/10.1115/OMAE2018-78094

The difference here is that I have calculated the coulomb force experimentally and therefore want to implement it through this manor. Through the WECCCOMP Simulink block, I see something similar has been done for the controller:

[image]

nathanmtom, 2023-04-07:
@Robert6codes Thank you for your question and I'll do my best to respond and give some more information for you to review.

First off, I wanted to direct you to another WECSim applications case, https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/WECCCOMP/WECCCOMP_Fault_Implementation, that has implemented joint friction and has an associated report to help you navigate how that was implemented for this particular case. 

In terms of your follow-up question, you are correct that in the controller block you have shown you can add additional damping (linear or nonlinear) in the loop to mimic any other external damping. However, I am having difficulty understanding if you are asking how to use a Tustin model block within Simulink or you have some type of if/else statement that provides an output given a certain input? If it is the later option, then you would just need to go into the controller and find where position or velocity outputs are found, implement said if/else statement to generate the desired moment you want and add to the output signal.

nathanmtom, 2023-05-01:
@Robert6codes We wanted to follow-up to see if my suggestions above helped answer your question about implementing Coulomb Friction into the WECCCOMP model. Please let us know by the end of the week or else we will close the issue assuming this has been resolved.

Robert6codes, 2023-05-01:
Hi Nathan,

I hope you are well. I ended up using the "linear motor" block in the image without the controller using a number of if statements dependent on the motors position only. The advice did help thank you! I will close this issue.

closed 2023-05-01 by Robert6codes