WEC-Sim/WEC-Sim #1027: [BUG]

tedman223, 2023-04-06:
Hello I am an MECC team member. I am having a few issues with my wecSim model. In my sim mechanics my water line is only showing up between my float and spar but no where else. In addition there is no vertical motion in the float. I am also having a very low Power output which I believe is related to this lack of vertical motion. 

[image]
![code](https://user-images.githubusercontent.com/126424529/230451114-96e7f7bf-4245-4d37-ac42-3
![phase](https://user
[image]
-images.githubusercontent.com/126424529/230451151-58fc7470-4ded-4607-a2ad-60df5305daac.jpg)

[image]
038c564a7d5.jpg)

nathanmtom, 2023-04-06:
@tedman223 Thank you for your question. We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**WEC-Sim applications specific issue or question**" template and include more details on your setup and error message, We are happy to help. Please maintain the template structure and the issue type in the title.

**Refer to WEC-Sim issue templates and encourage user to re-submit issue (long)**
Thank you for your question. When creating an issue, please use the GitHub templates:
[image]
[image]
[image]

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

nathanmtom, 2023-04-17:
@tedman223 Thank you again for your question. I wanted to reach out to ask if you are able to edit or open a new issue following one of the issues templates. If we do not hear from you or see edits by the end of the week we will close this issue.

tedman223, 2023-04-17:
> CAUTION: This email originated from outside of the University System. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I was actually able to resolve my issue, and was adressing the wrong questions for it, so I think it would be best to close the issue. I am however having some issues with getting my linear generator values correct and was wondering if you had any advice on where to research these?

cheers, Ted

nathanmtom, 2023-04-17:
Thanks for the update @tedman223 and glad to hear you resolved your first issue. 

If you are still having questions about your generator values, I am going to recommend closing this issue and you resubmitting a new issue while following the issues templates as illustrated in my first response. Apologies if this is an inconvenience, but following the templates helps ensure we have enough information to provide useful feedback.

closed 2023-04-18 by tedman223