WEC-Sim/WEC-Sim #1028: [BUG] Hydrostatic restoring matrix

hachikoi1, 2023-04-11:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
1. In hydrostatic restoring matrix, C(6,4) and C(6,5) should always be zero and C(4,6) and C(5,6) may be non-zero. Automatically calculated body.hydroForce.linearHydroRestCoef seems to be the transpose of the correct result if I do not specify obj.hydroStiffness.
[image]
2. In AQWA theory manual, C(4,4), C(5,5), C(4,6), and C(5,6) use displacement mass but not body mass, but those in WAMIT use body mass. For floating bodies with moorings, these two results are not the same. Therefore, a correction of C(4,4), C(5,5), C(4,6), and C(5,6) is necessary for AQWA.

[image]
[image]

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
4. _Scroll down to '....'_
5. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

nathanmtom, 2023-04-11:
@hachikoi1 Thank you for your inquiry and I will try to do my best to respond. 

In terms of the hydrostatic stiffness symmetry, I cannot tell where your first image is pulled from but one would expect that their should be symmetry as motion in one degree of freedom is influencing another degree of freedom and the they are coupled.

After reviewing your issue, I believe you are concerned that the mass of the object is needed to be defined in order to correctly calculate the hydrostatic restoring coefficient. This is true if you are calculating the hydrostatics at a different location than the center of gravity. Per WEC-Sim's requirements if the forces are calculated with respect to the center of gravity, then there is not moment arm generated by the gravitational component, causing `z_{g}` above to be zero. 

Based on my responses, I do not believe there is a bug in the code, but please consider my responses and let us know if you still have questions about the implementation.

Cheers, 
Nathan

nathanmtom, 2023-05-05:
@hachikoi1 Apologies for the delay in my response, but will attempt to answer your latest set of questions and comments.

- In response to the (6,4) and (6,5) indices in the hydrostatic restoring matrix, you are correct that currently the readBEMIOH5.m file does a symmetric transpose. The PR #1032 is being implemented to zero those indices that will occur with the symmetric transpose. We agree this change needs to be implemented based on the AQWA and WAMIT definitions.
- In response to recalculation of C(4,4), C(4,5), and C(4,6), C(5,5), and C(5,6), we do not believe there needs to be a change because of how we calculate the hydrodynamic coefficients about the center of gravity. In this case, there is not restoring moment arm generated by z_{g} as you are calculating the coefficients about the center of gravity. Therefore, the mass term shown in the equations above drops out of the equation. The WEC-Sim team had discussed this question last fall and after comparing results against WAMIT outputs and analytical expressions we were confident that our current implementation is correct as the hydrostatic components are then only dependent on geometry and displaced volume (which is reliant on geometry).

PR #1032 should be merged within the next week after which we will close this issue. Thank you for bringing this to our attention.

hachikoi1, 2023-05-06:
Dear [nathanmtom](https://github.com/nathanmtom)

    Thank you for your replies. 

    I generally agree with your work. For the first question, I think it would be better to exchange (6,4) and (6,5) for (4,6) and (5,6), especially for eccentric devices.

Best Regards,

nathanmtom, 2023-05-17:
@hachikoi1 Please let us know if our recent PR has addressed your concerns. If we do not hear from you by the end of this week, 5/19 we will close this issue.

hachikoi1, 2023-05-18:
> Hi @hachikoi1 ,
> 
> re: exchange (6,4) and (6,5) for (4,6) and (5,6) Originally, the transpose did exchange those values. The PR #1032 ensures that the roll row becomes a row of zeros after the said exchange,
> 
> re: the ρg∀zg−mgzg term in WAMIT,
> 
> [image]
> 
> The difference between the terms is because WAMIT allows the user to modify the inertia matrix used for the calculations of hydrostatics coefficients. If the FRC file is written in alternative 1 (by setting IALTFRC = 1), the mass is calculated using the mass of the displaced fluid, while if the FRC file is written in alternative 2 (by setting IALTFRC = 2), mass is defined by the user in the user-defined external inertia matrix set by the EXMASS part of the FRC (alternative 2) file. (see 3.1, 4.3, 4.4, WAMIT manual).
> 
> Does that address your question?
> 
> Cheers, Sal

Dear [salhus](https://github.com/salhus)
I'm very sorry, I've only just received a reply reminder.
- I have tested the modified codes, but the modifications seem to make both (6,4), (6,5), (4,6) and (5,6) to be zero.
For example, the HYDSTIFFNESS calculated by AQWA is
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
0.0000E+00  0.0000E+00  4.4558E+06  3.9265E+02 -2.3044E+01  0.0000E+00
0.0000E+00  0.0000E+00  3.9265E+02  2.1960E+09  2.5056E+03 -1.3018E+05
0.0000E+00  0.0000E+00 -2.3044E+01  2.5056E+03  2.1960E+09 -1.8571E+04
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
but body.hydroForce.linearHydroRestCoef is
0	0	0	                                0	                         0	                                0
0	0	0	                                0	                         0	                                0
0	0	4455640.97787295	        392.635986795146	-23.0431775874375	0
0	0	392.635986795146	        5112817523.19772	2505.51057815846	        0
0	0	-23.0431775874375	2505.51057815846	5112817523.19772	        0
0	0	0	                                0	                         0	                                0
- As mentioned in https://github.com/WEC-Sim/WEC-Sim/issues/1019#issue-1640906559, for a floating body with mooring systems, rho*g*V might larger than mg. For the FRC file written in alternative 1, I presume that WAMIT identifies the weight of the floating body as rho*g*V, i.e. $\rho g\iint_{S_b}y^2n_3dS+\rho g\forall (z_b-z_g)$ in C44 but not the formula of $\rho g\iint_{S_b}y^2n_3dS+\rho g\forall z_b-mg z_g$. Therefore, I think you are right.
Many thanks,

salhus, 2023-05-18:
Hi @hachikoi1,

1. The two matrices you shared, have different coefficients for the roll, and pitch modes - the $(4,4)$ , and the $(5, 5)$ entries. So the coefficients corresponding to the interaction between the roll/pitch with yaw being different could perhaps be stemming from that. I checked the cases where $(4,6)$, and $5,6$ were not zero after the update was made. Are they the same body? If possible can you share the mesh you used?
2.  As a quick, I would recommend looking at the $.hst$ file generated by WAMIT to check if WAMIT even calculates the same hydrostatics for the $4,6$, and $5,6$ - the roll-yaw and pitch-yaw coefficients 
3. Did you implement the external forces in WAMIT, like you did in AQWA?

hachikoi1, 2023-05-19:
> * I have tested the modified codes, but the modifications seem to make both (6,4), (6,5), (4,6) and (5,6) to be zero.
>   For example, the HYDSTIFFNESS calculated by AQWA is
>   0.0000E+00  0.0000E+00 [...]

Hi [salhus](https://github.com/salhus)
Sorry for my fault.
The HYDSTIFFNESS calculated by AQWA is
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
0.0000E+00  0.0000E+00  4.4558E+06  3.9265E+02 -2.3044E+01  0.0000E+00
0.0000E+00  0.0000E+00  3.9265E+02  5.1130E+09  2.5056E+03 -1.3018E+05
0.0000E+00  0.0000E+00 -2.3044E+01  2.5056E+03  5.1130E+09 -1.8571E+04
0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00  0.0000E+00
body.hydroForce.linearHydroRestCoef is
0	0	0	0	0	0
0	0	0	0	0	0
0	0	4455640.97787295	392.635986795146	-23.0431775874375	0
0	0	392.635986795146	5112817523.19772	2505.51057815846	0
0	0	-23.0431775874375	2505.51057815846	5112817523.19772	0
0	0	0	0	0	0
The corrseponding AQWA results are below:
https://github.com/hachikoi1/UMaine.git

salhus, 2023-05-22:
@hachikoi1, 
Thanks for the follow-up. 
Thanks to your observations, I think I found the issue. BEMIO was reading the AQWA data correctly but the writeBEMIOH5.m function was not doing the necessary transpose.  Could you please try this branch and check if this works as expected https://github.com/salhus/WEC-Sim/tree/aqwa_Hydrostatics

For your reference, here is a check that readAQWA function was working correctly.

I used the GitHub link you shared and ran readAQWA function of the BEMIO module, and the roll-yaw, and pitch-yaw terms correctly populated,
[image]

* In the images that follow, the black background images are from MATLAB,  and white background images are from the H5 file viewer.

[image]

However, when I looked at the h5 file in an h5 file viewer, the writeBEMIOH5 function should have taken a transpose.

[image]

Following the fix I made, here is how the hydrostatics matrix looks, confirming that the writeBEMIOH5 and readBEMIOH5 are working as intended.

[image]

and here is the h5 file viewer,

[image]

@nathanmtom This is a better fix to the PR #1032, as the need to force zeros on the roll row becomes unnecassry.

Cheers,
Sal

hachikoi1, 2023-05-23:
Dear [salhus](https://github.com/salhus)
This fix looks better!
The key issue is the transpose of hydro.Khs(:,:,i)
```
writeH5Parameter(filename,['/body' num2str(i) '/hydro_coeffs/linear_restoring_stiffness'],hydro.Khs(:,:,i)','Hydrostatic stiffness matrix','');        
```
Many thanks for your work.
Best regards,

salhus, 2023-05-23:
Thanks @hachikoi1 for noticing and helping through the process! 
I am glad you think that was helpful. 
re: AQWA Drift,
I think AQWA related development is usually slower because of the paywalled licenses. At the moment, I do not have a license for AQWA. The main issue with forming import scripts based on a limited number of files is the lack of confidence for an approach based on limited access. 
However, if you could send some AQWA files with drift forces, I may be able to give some recommendations on how to incorporate it with BEMIO.  
I would also encourage you to make PRs to WEC-Sim - the more user-engagement and help from the community we get the better :)

The PR should be processed soon. Let me know if we can close this issue after the PR is merged.

Cheers,
Sal

hachikoi1, 2023-05-24:
Dear [salhus](https://github.com/salhus)
I think the isses has been resolved and can be closed.
I will be working on reading the AQWA mean drift at a later stage. Looking forward to communicating with you.

kmruehl, 2023-05-31:
relates to https://github.com/WEC-Sim/WEC-Sim/pull/1065

closed 2023-06-05 by salhus