WEC-Sim/WEC-Sim #1033: [WEC-Sim Applications] Poor power output with direct linear PTO

tedman223, 2023-04-18:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
I am currently simulating a two body point absorber with the direct linear genrator PTO. I am getting results, how ever it is not good data. I believe that I am having some incorrect numbers in my PTO variables that are resulting in this. I have carefully selected what I beleive are resonable values and have iterated with all these values to see what works best. I am hoping to get a more resonable power, voltage, and current outputs as I have not been able to produce anything about 0.9 Watts. 

**To Reproduce**
_Steps to reproduce the behavior:_

N/A

**Expected behavior**

I am expecting at least 10 Watts of Power is not more. 

**Screenshots**
[image]

[image]

[image]

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 11]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [ 2022a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.1]

**Additional context**
N/A

dav-og, 2023-04-19:
Hello @tedman223,

Its a bit difficult to say if/why your power output is too low without knowing more around your model inputs. Would it be possible to share your input files? (Feel free to email them if you don't feel comfortable uploading here).

Also, I'm wondering:
- If you tried a linear PTO model at first? (i.e. just constant coefficients for stiffness and damping)
- How you determined the parameters for your direct drive PTO?

Cheers,
Dave

tedman223, 2023-04-19:
I have been trying various different constant coefficients for the stiffness and damping and I have only tried the Direct Linear Drive PTO. I found equations for all the variables and estimated based on our device design. I had a difficult time estimating the Flux Linkage of the stator and am unsure of what this value should be close to. When it was too high I had issues with micro oscillations so I belive it should be a lower input. I tried emailing you my files I got a response that the h5 was too large to post but if you can confirm that you recieved the files that would be great. Also here is a rough draft of the equations I went off of. 
[DDPTO equations.pdf](https://github.com/WEC-Sim/WEC-Sim/files/11275476/DDPTO.equations.pdf)

dav-og, 2023-04-19:
Haven't received anything, try this address: wec-sim@outlook.com

tedman223, 2023-04-21:
> CAUTION: This email originated from outside of the University System. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I believe this issue is that the h5 file is too large even in the compressed folder, is there another email I can send this too or should I just leave out that file?

tedman223, 2023-05-05:
Hello,

I have not heard back from you guys in a week in regard to my power output issue and was wondering if you have any recommendations on what to do?

cheers,
Teddy

dav-og, 2023-05-10:
Hi Teddy,

I've taken a look at your model but nothing looks obviously wrong to me.

I recommend using the linear (translational PTO) option to explore the PTO stiffness and damping space - and find out which values work well for this system. Then once you've determined the 'ideal' stiffness and damping parameters you could try to aim for equivalent stiffness and damping with your direct drive model.

Cheers,
Dave

closed 2023-05-17 by dav-og