WEC-Sim/WEC-Sim #1037: [BUG]  WECSim_Lib_Moorings incompatible matlab 2020b

treppe8k, 2023-04-21:
I am using Matlab 2020b, when i try install WecSim last version, I got this error

####################

.>> slLibraryBrowser
Error using load_system (line 21)
Model 'WECSim_Lib_Moorings' was created with a newer version (R2022a) of Simulink
To disable this error message, use Simulink preferences.
To create a model that is compatible with this version of Simulink, load the model in Simulink
R2022a and select Save > Export Model to > Previous Version.

####################

You can try re-export the 'WECSim_Lib_Moorings.slx' with the the MatLab 2020b version ?
According to guide , that is the version in use for WecSim 5.0.1

thanks in advance

jtgrasb, 2023-04-21:
@treppe8k Thank you your inquiry into using WEC-Sim's Mooring Library and notifying us of the issue. We are working on addressing this issue now. It seems to be related to PR #1012, but I am getting some issues with MATLAB crashing when trying to update it. I've created issue #1038 to work through the bug. 

If you have any other issues, please note we are requiring that all inquiries follow the issues template and please answer every section as requested without deleting the section headers so we know which questions on the template are being addressed by your statements. We prefer to stick with the templates as it helps us track issues historically and new users can better parse the knowns and unknowns of a posted issue.

[image]

closed 2023-04-21 by jtgrasb

treppe8k, 2023-04-22:
**the issue persist but with other paths** 

> the 2 examples (verified) and tutorials (maybe)

Warning: Model 'OSWEC' was created with a newer version (R2021a) of Simulink
To create a model that is compatible with this version of Simulink, load the model in Simulink
R2021a and select Save > Export Model to > Previous Version. 
> In load_system (line 21)
In simulationClass/loadSimMechModel (line 194)
In initializeWecSim (line 463)
In run (line 91)
In wecSim (line 43) 
Warning: Cannot load version 'R2021a_202009111111' of family 'sltp_mm'.

jtgrasb, 2023-04-24:
Hi @treppe8k,

We only require our WEC-Sim Library files to be saved in R2020b per discussion in PR #919. The newer models should run fine in 2020b. This error message can be disabled in Simulink's preferences or using the following command (which is also called when you run wecSim.m):

set_param(0,'ErrorIfLoadNewModel','off')