WEC-Sim/WEC-Sim #1041: Singularity in Generalized Modes Simulation

corbinlenderink, 2023-04-25:
I am receiving the following error when attempting to run a simulation including generalized deformation modes for a flexible, boat-shaped buoy:

Error using wecSim (line 45)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 45)
    ['Test_Model_GBM/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.0024124388727062521. Stopping simulation.  There may be a singularity in the
    solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)
        Error using wecSim (line 45)
        ['Test_Model_GBM/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'Test_Model_GBM/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass distribution on its base side.
    Error using wecSim (line 45)
    Error evaluating 'StopFcn' callback of Frames block (mask) 'Test_Model_GBM/Global Reference Frame'.
    Callback string is 'run('stopWecSim')'
        Error using wecSim (line 45)
        Index exceeds the number of array elements. Index must not exceed 1.

The WAMIT simulation for my case ran without issue, but after creating the .h5 file using Bemio, WEC-Sim will not run properly. I followed the WEC-Sim flexible barge example and believe I set up my model in the same way as this example. Additionally, the body file generated by my WEC-Sim input file also appears to match up with the WAMIT .out file. I am not sure what is causing the singularity and would appreciate any insight into how to fix this issue. Thank you!

akeeste, 2023-04-25:
@corbinlenderink Thank you for your question. We now require users to open GitHub issues using an appropriate template. This helps the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**Bug report**" template and include more details on your setup and error message, I am happy to help. Please maintain the bolded template structure and the issue type in the title.

closed 2023-04-26 by corbinlenderink

pkalog01, 2023-05-05:
Hello, I happen to come across the same problem, is the issue resolved?

akeeste, 2023-05-05:
@pkalog01 This issue was moved to #1042, where you can read about some of the proposed solutions.