WEC-Sim/WEC-Sim #1042: Generalized Modes Simulation Issue[BUG]

corbinlenderink, 2023-04-26:
**Describe the bug**
I am attempting to run a flexible buoy simulation from a WAMIT simulation that included four generalized deformation modes. Although I set up the simulation by following the example for the flexible barge, my simulation runs for several hours without any convergence. I have run the simulation excluding the deformation modes with a rigid body block successfully, but cannot get the simulation to run properly when I include the deformation modes with a flexible block.  Additionally, the body file generated by my WEC-Sim input file appears to match up with the WAMIT .out file. I am not sure what could be causing the simulation to not run properly and would appreciate any insight into how to fix this issue. Thank you!

**WEC-Sim file(s)**
Flexible body block (Simulation ran fine excluding the generalized modes using a rigid block)
**To Reproduce**
To reproduce this issue, I would need to provide the .h5 file and the WEC-Sim files I used to run the simulation (I would be happy to provide these)

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
Simulink with input file
**Expected behavior**
I would expect to get similar results as the rigid body simulation I ran for the same buoy.

**Screenshots**

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10
 - [MATLAB/Simulink Release]: 2021b
 - [WEC-Sim Release] v4.1

**Additional context**
The mode shapes were determined using an Ansys Modal analysis and interpolated to a WAMIT mesh using a Matlab code. The WAMIT simulation ran with similar results to a rigid buoy of the same shape.

corbinlenderink, 2023-04-26:
Here is the output file for the WAMIT run:
[WAMIT_Output.zip](https://github.com/WEC-Sim/WEC-Sim/files/11335289/WAMIT_Output.zip)

corbinlenderink, 2023-04-26:
Here is the .h5 file:
[h5_file.zip](https://github.com/WEC-Sim/WEC-Sim/files/11335370/h5_file.zip)

Here are the WAMIT input files:
[WAMIT_Input.zip](https://github.com/WEC-Sim/WEC-Sim/files/11335367/WAMIT_Input.zip)

Here is the WAMIT output file:
[WAMIT_Output.zip](https://github.com/WEC-Sim/WEC-Sim/files/11335373/WAMIT_Output.zip)

Here are all the WEC-Sim files:
[WEC_Sim.zip](https://github.com/WEC-Sim/WEC-Sim/files/11335378/WEC_Sim.zip)

I think that should be everything. Please let me know if there were any files I may have missed. Thank you!

corbinlenderink, 2023-04-26:
To answer your question, I was surprised by the heave and surge response for the flexible case I ran. Below, I have included the rigid body results for the same buoy below.

[image]
[image]

The buoy is 2m in length, and I was expecting to get similar outputs when I added the four flexible modes.

dforbush2, 2023-04-26:
Got it thank you. Those comparison plots give me a better idea as to the expected trends. I will dig into this and follow up soon. My prediction at this time is that the magnitudes/phases of the GBM terms, especially the off-diagonals cross-coupled to surge and heave, are not behaving as expected and may need some additional damping or inertia imposed to be physically representative.

dforbush2, 2023-04-27:
@corbinlenderink 

do check out the 'dev' branch of WEC-Sim if you get a chance. plotBEMIO functionality here is expanded for additional GBM.

Initially, I notice that while you have 600 frequencies in your BEM runs, you have a relatively coarse discretization around your surge and pitch resonance ~ 3.5 rad/s (this is a common problem when specifying a linear wave period spacing in wamit input files, since frequency ~ 1/period. I would suggest altering your frequency vector to better discretize the frequency range of interest. 

Compared to these rigid body modes, the added mass and radiation damping information for the GBMs are 3-4 orders of magnitude smaller, while the excitation force magnitude is only 2-3 orders smaller. This implies that we expect GBMs to show substantially larger displacements than the rigid modes.  This is probably not what was intended. 

Further, the cross-coupling modes between the heave mode and the gbms show a negative added mass and negative damping that is of a consistently larger magnitude than the gbm added mass and damping . This is not necessarily wrong, but creates a situation in which sufficient heave excitation may cause the coupled gbm modes to act unstably (i.e., they have effectively negative inertia and damping.).

Looking at your WAMIT input files, it seems that you are specifying the stiffness, mass and damping of these associated GBM modes via IALTFRC = 2 . To my knowledge (this is limited with WAMIT so please check for yourself) these parameters are not included in the hydrodynamic coefficients that are estimated in the *.out file, which is ultimately parsed via BEMIO tools into your *h5 file, which is the information used to calculate the hydrodynamic forces in WEC-Sim. All that to say, it may be necessary to manually add these stiffness/damping/inertia contributions as specified in your *.frc file to their respective modes in your model. 

Let me know if you need help doing this in the Flexible body block, since this may involve breaking the library link. 

So in summary:
1) consider refining your frequency discretization in WAMIT to yield better resolution around your resonant frequencies. 
2) Ensure that your stiffness/damping/inertia applied in your *.frc file is being represented appropriately in WEC-Sim.

corbinlenderink, 2023-04-27:
Thank you for the detailed feedback! I will look into the options you suggested and update the post tomorrow based on the results.

corbinlenderink, 2023-04-27:
Since your last message, I have tried a couple things to fix the issues mentioned above, but the simulation is still showing the same trend as before. I have attached the updated simulation files based on the changes discussed below.

Here is the .h5 file:
[h5_file.zip](https://github.com/WEC-Sim/WEC-Sim/files/11346870/h5_file.zip)

Here are the WAMIT input files:
[WAMIT_Input.zip](https://github.com/WEC-Sim/WEC-Sim/files/11346872/WAMIT_Input.zip)

Here is the WAMIT output file:
[WAMIT_Output.zip](https://github.com/WEC-Sim/WEC-Sim/files/11346882/WAMIT_Output.zip)

Here are all the WEC-Sim files:
[WEC_Sim_Input.zip](https://github.com/WEC-Sim/WEC-Sim/files/11346886/WEC_Sim_Input.zip)

First, for the mode definition, I am using Ansys modal analysis to determine mode shapes for the structure and reading these mode shapes (interpolated from the Ansys mesh to the WAMIT mesh) into the DEFMOD Fortran file in WAMIT. Additionally, I include the mass and stiffness values for each mode shape in the .frc file. I realized I did not scale down the deformations for these mode shapes previously, so I tried scaling them to represent a smaller deformation in my last simulation. I also scaled the modal mass and stiffness values as well in an attempt to decrease the GBM displacements. However, this did not seem to make any difference in the results. Do you know anything that could be causing the GBMs to show substantially larger displacements than the rigid modes? I was able to recreate the flexible barge example using this same methodology, so I believe I am setting up the WAMIT files correctly.

Second, I looked at the stiffness/damping/inertia contributions from my .frc file, and it appears the data is being saved in the .h5 correctly as shown below. As you mentioned, the flexible barge example did not require manually adding this data so I don't believe that is the issue.

WAMIT FRC Screenshot:
[image]

h5 File Screenshot:
[image]
[image]

I will try running the simulation again with a more fine discretization, but I have not done so yet since the simulation would take longer to run in WAMIT.

dforbush2, 2023-05-02:
Hi @corbinlenderink 

Apologies for the delay I had to pull in the rest of the team on this one to make sure I had this right. The short of it is that the GBM barge example is NOT correct in how it handles the stiffness, inertia, and/or damping included in the *frc file. WEC-Sim  does not use these properties when it calculates dynamics in the GBM modes, using only the radiation and excitation coefficients to do so. This will also affect any modes with non-zero cross coupling terms to the GBM modes.

Because this is a serious inconsistency I have added the "bug" label and will work to revise the documentation and applications case as soon as possible and modify the flexible body block to accommodate the additional terms since this is a feature we ought to incorporate for a variety of reasons. At this time, WEC-Sim does ***not*** support additional stiffness, damping, and inertia specified in the *frc file of WAMIT and they will have to be added appropriately to the flexible bodies block if they are to be included in the WEC-Sim simulation. You'll likely need to break the library block to do this.

The quickest way to see this for those interested is to compare the RAO (output of the *.4 file) in WAMIT with substantially different values of the above specified in the *frc file with the same RAO calculated from multiple WEC-Sim runs. WAMIT will have RAOs differing between the runs (as it should be) while WEC-Sim RAOs will be identical. 

I will target having a working version of this block to you within the next couple of days to get you moving, followed by a more complete PR and documentation.

corbinlenderink, 2023-05-02:
Thank you for getting back to me and looking into it! I will monitor this page this week for the updated version of the block! Thank you for all of your help!

dforbush2, 2023-05-02:
Hi @corbinlenderink 

My previous comment was actually incorrect: WEC-Sim has this implementation in place and as long as the WAMIT input and output files appear as bemio expects, these values _should_ be pulled into their respective places in the  Flex Body/FlexibleBody/body.hydroForce.gbm.[stifffness/damping/mass_ff_inv]. 

I will review BEMIO implementation, can you please provide a more complete set of the WAMIT output files? Particularly there is a *.mmx file I was missing on the previous upload. In your case the *h5 file does look like it contains the correct elements in the right places and they are ending up in the right places in the model. I will work on your particular model today.

corbinlenderink, 2023-05-02:
I have made some changes to the model the last couple of days in an attempt to fix the issue, so I am not sure which simulation I sent previously. I have attached all the files for the most current simulation below including additional output files from WAMIT like you suggested.

WAMIT Input Files:
[WAMIT_Input.zip](https://github.com/WEC-Sim/WEC-Sim/files/11375824/WAMIT_Input.zip)

WAMIT Output Files:
[WAMIT_Out_1.zip](https://github.com/WEC-Sim/WEC-Sim/files/11375825/WAMIT_Out_1.zip)
[WAMIT_Out_2.zip](https://github.com/WEC-Sim/WEC-Sim/files/11375826/WAMIT_Out_2.zip)

h5 File:
[h5_file.zip](https://github.com/WEC-Sim/WEC-Sim/files/11375829/h5_file.zip)

WEC-Sim Files:
[WEC_Sim_Files.zip](https://github.com/WEC-Sim/WEC-Sim/files/11375835/WEC_Sim_Files.zip)

dforbush2, 2023-05-02:
Are your WEC-Sim models running to completion? With the files provided I get a nonfinite derivative at the 3DOF joint constraint at  ~2.9 seconds. The error is due to an instability in added mass force.

dforbush2, 2023-05-02:
This is definitely a BEM issue...which unfortunately means my support will be limited. What is driving the instability is what is noted in my first comment: the relative magnitude of the cross-coupling terms with the GBM modes.
Some tips:
1) make sure your modes have the proper sign convention, I'm not familiar with the ANSYS tool, but I've seen this cause similar issues in the past.
2) You might try meshing the free surface and using the IRR option, though since your rigid modes seem to work, I'm not sure this will make much of a difference. 
3) Though your mesh looks good, depending on your mode definition you may have better luck if it is further refined.
4) If you have a decent physical argument why this may be true, you could manually set the cross-coupling terms between the GBM modes themselves or the rigid modes to zero. 
5) Your surfaces have a low mass and stiffness for their geometric size (***note that these are output to the mmx file as non-normalized values, unlike the hydrodynamic coefficients***), at least compared to the barge example. You could modify this manually by adding linear or quadratic viscous damping to your wecSimInputFile or by changing the mass and stiffness in the h5 files.

corbinlenderink, 2023-05-02:
Thank you for the recommendations. I feel the issue is likely due to the mode shape sign convention and the mass issues you mentioned. I will work to resolve these issue and rerun the simulation. Thank you for your help!

dforbush2, 2023-05-02:
To speed things up significantly, since you have some impression of the frequency range of interest, you can also make this modification in the attached model and add the following to your wecSimInputFile. 

This relies on the control system to
[mods.zip](https://github.com/WEC-Sim/WEC-Sim/files/11378139/mods.zip)
olbox, but you can achieve the same thing without this toolbox by using a 1D TF function for each degree of freedom between a demux and mux block.

dforbush2, 2023-05-03:
I will leave this issue open for the time being. If you would be so kind as to keep us informed of the BEM-side solution you arrive at that would be of great help.

corbinlenderink, 2023-05-03:
I will be running additional BEM simulations the next couple of days and will let you know if I am able to resolve the issue. There are a few potential areas where I may have made some errors in the setup that I will be fixing. 

In the meantime, I had one additional question. Currently, the COG of the body I am simulating is located slightly below the water line in my BEM simulation. In the Bemio readWAMIT file, I noticed the COG of the body is read in by looking at the XBODY line of the output file as shown below.
[image]

The way this code is written, the COG is always located at 0,0,0 since these are the values in the XBODY section of the output file shown below.
[image]

However, for my simulation this would not be the case as the center of gravity is located at a different location. Would any issue arise from altering the code to read in the center of gravity from the appropriate location in the WAMIT.out file?Additionally, should my .stl geometry be oriented the same as my WAMIT simulation or should I orient it so the origin of the mesh is located at the center of gravity. Please let me know your thoughts. Thank you!

dforbush2, 2023-05-04:
Ah ok this might be the problem. The GBM applications case you are following does in fact have a CG coincident with the water line, so the origin of the gdf file is here and the inputs in the *pot and *frc files specifying CG location is 

You can check out the ./examples/BEMIO/WAMIT/RM3 for an example of how the GDF file and WAMIT inputs need to look for bodies with a CG that is not at the waterline.

WEC-Sim requires BEM coefficients that are provided about the CG of the body. That implies that the origin of the body is moved to the origin to align w/ WAMITs coordinate system. Your GDF mesh needs to be shifted so that 0,0,0 is at the CG of your body, and this offset between the free surface and your CG called out in your *pot and *frc input files. There are a combination of possible ways to do this noted in the WAMIT manual, but whatever you do the critical things to ensure  are that WAMIT is 1) solving for the CG at that is at the origin of your GDF file. and 2) the distance to the free-surface (where your mesh should be cut) and your CG is noted correctly. 

This very well might solve all of your problems on the BEM side. 

With respect to visualization, if I recall correctly the STL file coordinates are read directly from the STL file and applied to the WEC-Sim coordinate frame. So this means if you want to have your visualization appear totally correct you may have to shift so your intended-for-WAMIT mesh to this new origin. In your case however, this offset is slight enough I wouldn't worry about it.

corbinlenderink, 2023-05-04:
I was able to get the model to successfully run with the generalized modes included, but I made some changes that I am not sure are reasonable.
First, in the WAMIT simulation, I altered the POT file to include only the heave values in the simulation. Additionally, I ran the WAMIT simulation with the GDF file positioned at the waterline, so the CG location was not 0,0,0. In WEC-Sim, I then altered the .out file that was read in by BEMIO so the center of gravity values read in from the XBODY line were the correct center of gravity values as shown below. For the .stl geometry file,. I positioned the geometry so the origin was at the center of gravity. 
[image]

Finally, I double checked the signs of the mode shapes as you had mentioned and altered them to match up with a paper I have been using as a reference. I believe this could have been another source of error in the model.

These changes seemed to lead to a successful simulation, but I am not sure if the COG fix was the correct way to go about it. I will look into the example you mentioned to double check the method I used.  If you had any insight concerning the workaround I used for the COG, please let me know. I am now planning to rerun simulations with different settings to try to identify the source of the problem. Regardless, it appears to be functioning properly now.

corbinlenderink, 2023-05-09:
Since I last posted, I have been able to get simulations running with the flexible body block and generalized modes included. However, I have noticed the simulations have not shown any difference between the same simulation without the generalized modes included. As you can see in the image below, the heave response is identical between the two simulations.
[image]

For this simulation, I was modeling a disk shaped buoy with two generalized modes. Below, I have included the Ansys Data for the mode shapes as well as screenshots of the modal deformations. The mode shapes included in the model are modes 2 and 3 in the below screenshots.

[image]
[image]
[image]

I have also attached the files for this simulation in case they were needed to look at the details.

WAMIT Inputs:
[WAMIT_Input.zip](https://github.com/WEC-Sim/WEC-Sim/files/11436144/WAMIT_Input.zip)

WAMIT Outputs:
[WAMIT_Outputs_1.zip](https://github.com/WEC-Sim/WEC-Sim/files/11436123/WAMIT_Outputs_1.zip)
[WAMIT_Outputs_2.zip](https://github.com/WEC-Sim/WEC-Sim/files/11436130/WAMIT_Outputs_2.zip)

WEC-Sim Files:
[WEC_Sim_Files.zip](https://github.com/WEC-Sim/WEC-Sim/files/11436149/WEC_Sim_Files.zip)

dforbush2, 2023-05-10:
So looking at this, I see that this is a different geometry than run previously. From your GDF, FRC and POT files, the CG should be located at 0,0,0 and coincident with the SWL. But looking at the startup portion of your run this does not actually seem to be the case and the equilibrium heave position of the COG is about 0.7 m. (Interestingly, the GBM mode 7 mirrors this startup behavior...this might explain your identical heave results w/ and w/o gbm modes) 

Please ensure that this was what was intended, since it differs quite a bit from your previous approach. 

I generally recommend 'regularCIC' calculation methods rather than 'regular' unless you're really confident in your bem results at every point. Your BEM here does look really nice (except for maybe the COG offset described above), though, so you might be ok. 

On your comparison point, do you have a rigid-body *h5 run that you can provide?

corbinlenderink, 2023-05-11:
Yes, you are correct that this is a different geometry than the geometry I posted about previously. This geometry was a simpler shape to model, but I am experiencing the same issue with the geometry I posted about previously. Could you explain in more detail where you are finding the COG offset in the model startup? From what I see in the h5 file, the COG is set at 0,0,0. 

I have attached the rigid body h5 for comparison:
[Rigid Files.zip](https://github.com/WEC-Sim/WEC-Sim/files/11453077/Rigid.Files.zip

corbinlenderink, 2023-05-16:
I just wanted to follow up with my previous post to see if I could get a response. I am still unsure of the COG issue mentioned in the previous response. Please let me know,. Thank you!

dforbush2, 2023-05-16:
Hi @corbinlenderink 

The issue is probably minor at this point but I have slight concerns with your BEM setup. This is based on 2 observations:
1) at WEC-Sim model start, the z-position of the CG moves to ~ + 0.7m. This appears to be because the mass of the buoy and the displaced volume are in equilibrium at this position. 
2) WAMIT will estimate hydrodynamic coefficients based upon the modeled position of the submerged geometry (i.e., it is solving these coefficients assuming a zero-amplitude vibration about the modeled position) 

Because the actual equilibrium submerged geometry (from WEC-Sim hydrostatics) is different from the modeled submerged geometry (in WAMIT), your hydrodynamic coefficients are not as accurate for your device as they could be. 

The solution is to 
a) adjust the buoy mass so that the equilibrium position in WEC-Sim is about equal to the submerged geometry in WAMIT.
or
b) maintain mass as it is and slice the WAMIT geometry to correspond to the actual submerged geometry at equilibrium. The position of your CG in the geometry will not change, it should still be at the origin, but you'll have to adjust your *pot and *frc file to tell WAMIT where the CG position is relative to the free-surface. 

I suggest you refer to the WAMIT v7 manual and check out some of their examples on this. You can set ILOWGDF = 1 in their test-case *cfg files to produce a GDF file that you can open to look at in something like Rhino.

corbinlenderink, 2023-05-16:
You have mentioned that "at WEC-Sim model start, the z-position of the CG moves to ~ + 0.7m". Can you explain where you are seeing this behavior occur? From the data output by WEC-Sim, I have only seen a COG position of 0,.0,0, so I am a little confused as to where you are seeing the z-position moving to 0.7 m.

dforbush2, 2023-05-16:
Look at the first couple of seconds of the "Heave Position" plot you have provided last week. That is the heave position of the CG of the body. Your h5 file should list the CG of the body with respect to the still water line (origin of the coordinate system in WEC-Sim). My concern is that your mean heave position in this plot does not align well with your H5 file, which is CG in the WAMIT run. If your submerged geometry does not change markedly over this displacement range (like it was a tall vertical cylinder) I wouldn't be concerned, but because it looks like it might in this case, I raised the issue.

dforbush2, 2023-05-16:
Also, comparing the flex and the rigid H5, your heave responses can be expected to be identical because all relevant diagonal heave terms are identical and cross-coupling (3,7 and 7,3) are consistently anti-symmetric. Aside from the minor BEM issue above it seems like this is all running well now.

corbinlenderink, 2023-05-18:
I am worried I may have a misunderstanding of the flexible modeling in WEC-Sim. Today, I looked back at the flexible barge case from the WEC-Sim examples, and I set up a simulation for the barge without including the flexible, generalized body modes. After running these simulations, I plotted the heave response for the two simulations as shown below.

[image]

As you can see, the heave response is identical for this geometry as well regardless of whether additional DOFs for generalized modes were added. For the flexible simulation, is there some additional data that is included for the flexible case that I am missing? Unless I am misunderstanding the outputs, it appears to me there is no difference between the two simulations. This would mean the bending and shearing of the body that is included in the flexible case is having no influence on the results. After running the flexible simulation, I notice a flex_out file is created, but I haven't been able to find an explanation in the documentation as to what is included in this output file. Would it be possible to get a more detailed explanation of the methodology than what is included in the limited documentation? I am having a hard time understanding the benefit of the flex block if the results appear to be the same as the rigid block.

dforbush2, 2023-05-18:
The documentation is sparse on this and the flex_out structure is not as intuitive as the output data structure. I will elevate this to the rest of the team...it's high time this documentation is improved, but I'll give you a little rundown here.

The output structure contains all of the _rigid body_ dynamic information. So in the test that you ran above, the heave mode is the rigid body heave of the barge (and also for your two buoy runs): the only reason one would expect these results to be different would be if the flexible modes somehow affects/limits the heave mode through the off-diagonal hydrodynamics (If you're running WAMIT, check out rigid body heave results (index 3) for test17).

The 'flex_out' is the same structure as 'body_out', 'pto_out', etc, but the difference is the latter group gets deleted and collated into the 'output' structure if wecSim runs to completion. The easiest way to see what is happening with flex_out is to open the library block and look at the last mux block running into this flex_out data block.

[image]

From tracing these lines back we can see that order of the mux terminals is:
1) position
2) velocity 
3) acceleration
4) total force (excitation - (sum of everything else))
5) excitation force
6) radiation damping force
7) added mass force
8) restoring force
9) morison and viscous
10) linear damping

**for your flexible body modes**. This implies that the size of each group is determined by the number of modes you have defined in your BEM run. So for the barge example there are 4 GBM modes, implying that if one wanted to look at the excitation force on GBM mode 3 they would look at:

4(position) + 4(velocity) + 4(acceleration) + 4(excitation) + 3(the mode we want) = 19

the 19th data column in flex_out. The time step of this data is the same as that in the output structure. So we could plot it
```
figure; clf;
plot(flex_out.time,flex_out.signals.values(:,19));
xlabel('Time(s)')
ylabel('GBM mode 3 excitation')
```

In contrast, your buoy example has 2 GBM modes. So to look at the excitation force on mode 2 we would look at:
2(position) + 2(velocity) + 2(acceleration) + 2(excitation) + 2(the mode we want) = 10

the 10th data column in flex_out. 

We definitely need to improve our documentation here!

corbinlenderink, 2023-05-18:
Thank you so much for that rundown of everything. Your explanation was very clear, but I had some follow up questions I was hoping you could answer. For the study I am working on, I was hoping to get a better understanding of how the inclusion of the flexible modes will impact the power output of a point absorber WEC. Based on your explanation above, it seems the flex body block outputs the rigid body results to the "output" structure and the generalized modes results to the "flex_out" structure. Is there any way to couple these two sets of results together to get an understanding of how the flexible modes affect the buoy motion? I am having a hard time understanding how to get beneficial results from the flexible model if the outputs are simply the rigid body and generalized modes results completely separated. Finally, could you possibly explain the physical meaning behind the position of a generalized mode? I know I defined the mode shapes as positions in the z-direction, so could this be interpreted as the heave motion of the mode shape? Please also let me know if there is any additional documentation I could look into besides the WAMIT manual and the WEC-Sim page. Thank you so much for all of the detailed help you have provided with this!

dforbush2, 2023-05-18:
To be clear, WEC-Sim DOES use the full-matrix (including GBM) to calculate the impulse response functions and/or coefficients it uses to determine the rigid body motions. So if there is coupling between the GBM modes and the rigid body modes, it will be accounted for. Based on your BEM, this coupling doesn't have this effect b/c of its symmetry. The barge also has this character. You can check out the bodyClass to see how these quantities are calculated: check out body.hydroForce.fAddedMass, fDamping, and/or irkb depending on solution method to see how this works.

So if the addition of flexible modes _will_ affect your rigid body motion according to your BEM, wecSim should catch it. Its just in the cases we have evaluated so far, this wasn't really the case. 

The physical meaning of the gbm mode depends on the mode definition, but "one unit" of deformation in wec-sim is equal to the gbm mode definition evaluated for that deformation ==1.

corbinlenderink, 2023-05-19:
Can you explain what you mean by "this coupling doesn't have this effect b/c of its symmetry"? Is there a way to avoid this so the GBM modes actually have an effect? Otherwise, it seems like the results for rigid and flexible simulations will always be identical.

dforbush2, 2023-05-19:
That's a function of your geometry, mode definition, and BEM setup. I have numerous GBM runs that are set up similarly to these example cases that do not have symmetric off-diagonal terms and therefore don't have identical rigid-body responses. Note that in test case 17 they deliberately fix their GBM mode in the *frc file for...comparison reasons(?, I'm actually really confused by this decision) . I'll review your input files, but again, I'm not a WAMIT pro so I can't guarantee things on that end are working as intended.

Best,
Dom

kmruehl, 2023-06-14:
I'm going to close this issue