WEC-Sim/WEC-Sim #1044: [Theory or Implementation] Origin, Center of Gravity Cg, and Center of Motion CM

yman95, 2023-04-26:
**Is your question request related to a problem? Please describe.**
I'm using BEMRosetta for the STL file processing and NEMOH for the BEM hydrodynamics calculations. In my simple linear hydrodynamics case, I understand that WEC-Sim wants the origin of the mesh to be at the origin, "As in the linear case, the STL mesh origin must be at a body’s center of gravity." However, BEMRosetta's dev confirmed that translate function moves the entire geometry including the Cg and center of buoyancy Cb, so it doesn't seem possible to translate just the origin to Cg. 

**Describe the theory or implementation approach that you have a question**
From BEMRosetta, after setting the sphere mass, using Archimedes principle, the origin is set to the water level, and the adjusted Cg and Cb are shown below:
[image]

However, the exported .stl and .dat file don't remember the location of Cg. It's defined in NEMOH's Hydrostatics.dat file. 
[image]

If I define Cg to the correct location (0, 0, 2.494) from origin and CM refer to the same location, The final simulation in WEC-Sim will have the sphere float above water level, but if I leave Cg at (0, 0, 0) which is the water level. I doubt the simulation result is correct. 
[image]
Geometry floating above water level
[image]

I understand the center of motion is where the extra forces (added mass, excitation, radiation) are referred to, but I'm not clear on how to calculate such location. Does WEC-Sim take care of the location of CM in simulation? Given wave excitation force, pitching and wave latching, I assume CM will move horizontally and vertically.

Two questions:
1, Is there a way to solve or bypass the problem with Cg and origin? 
2, My most confused part: Where should I define my center of motion? I assume it's somewhere between Cb and Cg. Does it have anything to do with where I put the 3DOF constraint?

**Describe the type of conclusion or resolution you are looking for**
I would like to understand better about the center of motion for the system and have the correct geometry location relating to Cg and the origin.

Thank you for the supports,
Yi

jtgrasb, 2023-04-27:
HI @yman95 

Thank you for your inquiry into WEC-Sim's center of gravity conventions. I don't have experience with BEMRosetta directly, but I can offer insight into how WEC-Sim handles the center of gravity, origin, and center of motion. 

Your BEM run (NEMOH) should be set up such that the BEM calculation is done about the cog (it seems you have done this correctly). In NEMOH, the origin of the mesh is separate from the cog and is located at the mean water surface. WEC-Sim reads this BEM data in using the BEMIO scripts and applies the dynamics accordingly about the correct cog. The center of motion is something handled by Simulink's multibody dynamics solver and will depend on your WEC-Sim model including any additional constraints, PTO, mooring, etc. 

In terms of the mesh itself, WEC-Sim requires the .stl file's origin to be at the body's center of gravity. The .stl file is simply a set of points and connections defining the body mesh, so this can be adjusted after running NEMOH. I am not sure of BEMRosetta's capabilities, but translating the .stl to the cg is something that can be done in Solidworks or even in MATLAB with the [stlread()](https://www.mathworks.com/help/matlab/ref/stlread.html) and [triangulation()](https://www.mathworks.com/help/matlab/ref/triangulation.html) functions. It is also important to note that the .stl file is only used for visualization in WEC-Sim (unless you plan on using nonlinear buoyancy and Froude-Krylov forces) and is unnecessary for a linear WEC-Sim model if simu.explorer is set to 'off'.

With the cog correctly defined in NEMOH and the .stl translated to have its origin at that cog, the body should be defined correctly and the visualization will accurately represent the bodies position and motion within the simulation. 

I hope this is a helpful explanation and please let me know if you have further questions.

--Best,
Jeff

yman95, 2023-05-01:
Thanks for the explanations. It's great that you confirmed that the BEM calculations should be done about CG(cog).

I've tried with the MATLAB stlread() and triangulation obj for translating the vertices with something like

```
TR = stlread('Sphere_r5_achimedes.stl');
vertices = TR.Points;
translation = [0 0 -2.4935];
vertices = vertices + translation;
TR_translated = triangulation(TR.ConnectivityList, vertices);
stlwrite(TR_translated, 'Sphere_r5_translated.stl');
```

It translates every point directly by certain amount the same way BEMRosetta does it. And this method translates the water level, and center of buoyancy CB as well:
[image]
As far as I understand, NEMOH records CG in Hydrostatics.dat: 
```
 XF =  -0.000 - XG =   0.000
 YF =   0.000 - YG =   0.000
 ZF =  -0.875 - ZG =   0.000
```
With F denoting CB and G denoting CG. So, I'm not sure there's a way to just translate CG from the stl file. 

As a workaround for now and as you mentioned that the stl file is purely for visualization for the linear case, I'll put (0,0,0) for my CG and CM at the location of where CG is supposed to be.

Do you have other suggestions on how to move just the CG? Maybe Solidworks have this function?

jtgrasb, 2023-05-02:
Hi @yman95,

I think I am a bit confused on your issue. It seems you are trying to define a sphere with a cg at its physical center, which is located at (0, 0, 2.494). In NEMOH, the origin of the mesh is supposed to be at the mean water surface, and the cg is specified relative to that origin. It looks like you have done this correctly in your first picture if I am understanding your intentions correctly. After running NEMOH with this configuration, the hydrodynamic coefficients will be as desired and can be loaded into WEC-Sim. Assuming the configuration in the first picture is what you desire (most of the sphere above the water line), WEC-Sim should run as desired. You do not need to change the cg of the body.

In terms of loading an .stl into WEC-Sim (only needed for visualization and nonlinear), the .stl that was used for NEMOH will be in the incorrect location because WEC-Sim requires the mesh to have its origin at its center of gravity rather than at the mean water surface. This will mean a different mesh will need to be used for WEC-Sim vs. NEMOH.

Please let me know if I am misunderstanding your issue.

Best,
Jeff

yman95, 2023-05-03:
I apologize for the confusion. You are understanding my steps perfectly. The issue for me now is the location of CG in the stl file. The first figure in BEMRosetta was after the Archimedes function, and it seems to have all the correct locations for CG, water level, CB. But it is not saved that way. After saving the stl file and the dat file, the information on the location of CG is lost. Using the translate function in BEMRosetta and MATLAB will translate water level and CB along with CG. So, I'm wondering if there is a method to just move the CG, i.e., from figure 2 to figure 1.

Thanks again for the help,
Yi

jtgrasb, 2023-05-03:
@yman95 Thank you for clarifying. It seems your issue may be in the way BEMRosetta sets up the problem and saves the results. I am unable to provide support on BEMRosetta and NEMOH, but I can suggest checking the basic results of NEMOH. If you run BEMIO in WEC-Sim, you can look at the plots and compare them to what may be expected. You could also access the Khs value and compare that to a theoretical value for the immersed part of the sphere. Those results can help you understand if NEMOH is calculating the hydrodynamics as you'd expect. You can also compare your results to our [NEMOH examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/NEMOH) including a sphere example, which may help as comparison.

If the results from BEM are calculated about the wrong point, there is no way to easily translate them to a new location after the fact. I would suggest working further with BEMRosetta and NEMOH to figure out why the cg is being placed at 0,0,0 rather than 2.494. You may need to shift your mesh file first before loading it into BEMRosetta.

yman95, 2023-05-03:
Thanks for the advice. I'll work further with BEMRosetta to try to resolve the issue. 

Can I ask for a suggestion not related to the issue? You mentioned that to try to compare to a theoretical or analytical solution to check if the simulation results are what are expected, do you have any references you recommend for validating results? I'm interested in comparing the hydrodynamic performances of different geometries in different wave conditions using WEC-Sim, but the first step is probably validating the model.

I'm sure WEC-Sim was validated, but for a specific problem and academic writing, I may need to validate any of my results to an analytical solution or other CFD models. Can you recommend some strategies to this? Thanks.

jtgrasb, 2023-05-04:
In terms of validating the basic position and cg results from NEMOH, I would recommend comparing to our sphere example, which has also been run in WAMIT, Capytaine, and AQWA. You can use this to compare to the hydrodynamic coefficients from your NEMOH runs, validate the results are acting as expected, then change your model as desired. I would also recommend our [numerical test cases](https://wec-sim.github.io/WEC-Sim/dev/user/troubleshooting.html#numerical-test-cases), which give an incremental process to make sure your WEC-Sim model is behaving properly. 

For model validation purposes, tank test data is definitely best, but there are also some high-fidelity software that can be used. [Here](https://www.osti.gov/biblio/1774818)'s an example where Star-CCM+ was used to validate a WEC-Sim model.

yman95, 2023-05-05:
Thanks a lot for the advice and supports. They help a lot.

closed 2023-05-05 by yman95