WEC-Sim/WEC-Sim #1053: Long Simulation Runtimes Using Elevation Import - [Theory or Implementation]

ccz359, 2023-05-16:
**Is your question request related to a problem? Please describe.**

Long simulation runtimes associated with user-defined wave elevation data. A 1200 second simulation is projected to take 14 days, and there are hundreds of 1200s files to simulate (project completes end of July 2023). The simulation does appear to work without errors (70 seconds successfully simulated), just takes too long.

**Describe the theory or implementation approach that you have a question**

The model is a modified version of the RM3, with modified Hydraulic PTO Simulation. 

The time step for the simulation is set very small: 0.0001s. If it is set larger than this, significant errors occur in the model with the PTO Sim valves (which are modified from the standard WEC-Sim library block). This time step is definitely contributing to the long runtimes, however when the same model is run using regular waves, the 1200s simulation completes in a few minutes. It appears that the wave elevation field data is contributing significantly to the long runtimes. I have verified that the wave elevation field data is formatted the same as the elevationData example file. The wave elevation field data I am using is 2.5Hz, from a Sofar Spotter buoy. 

Both 'accelerator' and 'rapid-accelerator' simulation modes have been tried, which both resulted in errors or equally long runtimes. 

**Personal project context**

The project goal is to be able to simulate hundreds of 1200 second wave elevation field data files, in order to compare WEC field data performance to WEC-Sim modeled values. 

The modifications to the RM3 and PTO-Sim are to reflect a wave-powered water pump device.

**Describe the type of conclusion or resolution you are looking for**

I'm looking for methods to speed up the simulation runtime, without compromising model accuracy. 

**Additional context**
WEC-Sim Version 5.0.1
Windows 11
Computer Specs: RAM 16GB, intel-i7 processor 2.80 GHz, 64-bit op sys

dforbush2, 2023-05-31:
Hi @ccz359 
Any update on this model? In group discussions, we had an additional suggestion.

1). Look at simu.cicDt, part of simulation class. This will allow you to resolve the excitation CIC at a different, hopefully longer, timestep than the rest of your simulation. You may need 0.0001s to resolve PTO motion, but you almost certainly don't need this resolution on wave excitation. (thanks @akeeste and @salhus)

I've added a post to our project board based on this issue that will explore ways to improve solver time for cases like this. One suggestion is to provide an option to entirely pre-calculate the excitation force time series before simulation (for cases where this won't degrade accuracy). This will remove one convolution calculation from running every time step. 

Best,
Dom

ccz359, 2023-06-08:
Hi Dom,

Thank you very much for providing some troubleshooting techniques for this issue. I have been able to try the following: 

From your first post:
0. Using the 'regular' wave class does significantly improve runtime.
1. ode23tb and ode15s do not change the runtime, it is still very long.
2. I have not yet tried spectrum import, but will try this next week.
3. I have not yet tried altering the library blocks as you've suggested.

From your most recent post:
1. I added simu.cicDt = 0.01 to the simulation class, and runtime has not improved.

I am currently pursuing running the simulation on a high performance computing cluster to see whether this will solve the issue. One other item to note is I am using Capytaine v.1.5. for the BEM.

Thanks again!
Chelsea

dforbush2, 2023-06-08:
0.) what about the 'irregular' wave class or 'regularCIC'? These are convolution-integral based calculations similar to the user defined free-surface and can help clarify the problem.

1) simu.cicDt not affecting the run time substantially suggests that it is not what's causing your run time slow down. It's still possible that spectrumImport and modifying the body block as suggested will help, but less likely. Please try them in any case, hopefully they will provide at least small speed-ups.

If you can please provide a *zip of your model w/ all necessary files I can provide more specific help. If suspect that the issue you are having is that the combination of parameters in your PTO have created a very stiff system. I'm guessing if you look at a trace of your PTO-related results there are lots of jumps and sharp transitions.

nathanmtom, 2023-06-10:
@ccz359 Thank you for sharing a .zip file of your model. We'll take a look and follow-up if there is any other input to provide. 

In terms of the state space approximation for the convolution integral, how to update the `wecSimInputFile.m` can be found at the link below which also provide the link to the theory behind the implementation.

http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#state-space-representation

dforbush2, 2023-06-15:
@ccz359 
I have reviewed your model and have several suggestions. All cases were tested with a 4 s period (peak, for irregular waves) and a height (significant for irregular) of 0.3 m. I ran with dt = 0.001 and ode23t. I tested regular, regularCIC, irregular, eta import, and an equivalent spectrum import. I simulated 12 s with a 4 s ramp to keep the overall run time manageable.

* you have not calculated state space matrices in your BEM data. As suggested by @nathanmtom this is likely to give you significant speedup above that presented here. 
* increasing cicDt to 0.01 decreased simulation time by a factor of 5 to 10, depending on wave class
* Your BEM IRF decays very slowly. This is likely non-physical so I recommend revision. However, for the provided *h5, decreasing cicEndTime to 30 s from the default 60 s decreases run time further without degrading accuracy too much.
*  using the equivalent spectrum import from your provided eta import speeds is ~20% faster than using eta directly. Further, this spectrum can be substantially truncated, ignoring frequencies above about 5 Hz, without losing significant energy. Because this will use less components for each calculation, this will speed run time further. 

Employing all of the above, I simulated 100 s in on my laptop in 279 seconds. In my experience, this run time is about on par with simulations including detailed PTOs. If this is still inadequate:
* truncate spectrum of your provided time series to only energy containing low frequency bands (this will degrade phase agreement with the provided eta time series slightly). 
* re-compute your h5 file from *out to include state space coefficients. 
* see library modifications suggestions earlier in this thread. 
* if you are running a batch of simulations, consider using wecSim with the parallel computing toolbox (>>doc parsim, and see 
http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#parallel-computing-toolbox-pct

closed 2023-06-15 by dforbush2

ccz359, 2023-06-24:
Many thanks @dforbush2 and @nathanmtom 

Following your last round of suggestions I did significantly improve the runtime. Instead of 14 days to simulate 20 minutes, it is down to 3.5 days. I will attempt to utilize the supercomputing resources available to me for this project, however your suggestions were a significant (4X) improvement. Thanks very much!

(Note: I do need to run the time step at 0.0001s in order to get accurate results - not 0.001s which @dforbush2 was using to troubleshoot)

dforbush2, 2023-06-26:
One quick follow-up, if you have not implemented this already:  If you make the suggested library modification (change transport delay --> low order low pass filter with a <~50 Hz cutoff frequency) you may find that the larger time step will have a less deleterious effect on the accuracy of your output data.  This is a "smoother" way to break the feedback loop in the solution of the dynamic equation and will decrease the numerical stiffness. During my tests I had made this change and wasn't seeing a great deal of difference in results between dt = 1e-4 and dt = 1e-3, though my simulations were abbreviated and I wasn't looking at the data as thoroughly as I'm sure you are.