WEC-Sim/WEC-Sim #1060: [Theory or Implementation] Implementation of a Submerged Point Absorber with a Coupled PTO/Mooring system

razaali32, 2023-05-17:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
An update to this post: https://github.com/WEC-Sim/WEC-Sim/issues/1004

I am attempting to implement a submerged single tether point absorber with a floating 3dof constraint. I have been able to get a heave only implementation of this device using the cable block and two fixed constraints between the float and a dummy PTO body. 
(Shown below)

[image]

However, when attempting to expand this to a 3-dof (Surge, Heave, Pitch) set-up, I am unable to use the cable block as any translational degree of freedom renders the cable block useless. Thus, I switched my implementation to have the tether modeled with moordyn. This works when I have a 3dof constraint, the moordyn block, and the float body. However, when I try to add a translational PTO this breaks.

**Describe the theory or implementation approach that you have a question**
I read in a different issue post that a dummy body block between the 3dof constraint and translational PTO can help, this helped the simulation run for the noWavesCIC case, but once regular waves are introduced the simulation becomes unstable. The set up is shown below: 
[image]

**Personal project context**
N/A

**Describe the type of conclusion or resolution you are looking for**
I am looking for a way to implement a submerged point absorber with a coupled PTO/mooring system in line in Simulink. 

**Additional context**
I have attached the simulation folder with geometry, hydrodata, and input files. 

[Flume Scale-3dof.zip](https://github.com/WEC-Sim/WEC-Sim/files/11503079/Flume.Scale-3dof.zip)

dforbush2, 2023-05-18:
Assigning @jtgrasb since they helped #1004 

Looking at your models, I'm quite confused by the physical system you are trying to model. As I understand your desire you would like to model:

1) a submerged float
2) a connection from the float to a translational PTO.
3) a connection from the translational PTO to a cable.
4) a connection from the cable to the sea bed. 

Note that the cable can totally work with the 3DOF... you'll want to use a spherical constraint. Based on this description, I would build your model like it is shown in PA2022a.slx, but I would use spherical constraints on both ends of the cable. On the sea-bed side of the cable, this will fix the connection point translationally, but it will rotate freely. On the PTO side of the cable, its position will be fixed to the location of the PTO body, but it will rotate freely.  

There already are 2 drag bodies in the cable block, so I'm guessing you're using the body(2) block to model the end of the PTO that connects to the cable. 

I tested this suggestion and it ran for your regular wave so I think this should be good to go for you.

razaali32, 2023-05-18:
I appreciate the suggestion of the spherical constraint. That solved my problems with the cable block, Cheers!

closed 2023-05-31 by kmruehl