WEC-Sim/WEC-Sim #1067: Error due to multiple Goto on tutorial examples

antbog, 2023-06-01:
Hi, 
I'm trying to run the tutorials of WEC-Sim but the following error appears, for both the RM3 and the OSWEC examples : 

> List of Constraint(s): Number of Constraints = 1 
> 
> 	***** Constraint Name: Constraint1 *****
> 
> 
> Simulating the WEC device defined in the SimMechanics model $WECSIM\tutorials\RM3\RM3.slx...   
> Elapsed time is 8.429617 seconds.
>
> Error using wecSim
> Error due to multiple causes.
> Caused by:
>     Error using wecSim
>     Multiple Gotos found with tag named 'disp1'
>     Error using wecSim
>     First offending Goto 'RM3/Float/Hydrodynamic Body/Goto'
>     Error using wecSim
>     Second offending Goto 'RM3/Spar//Plate/Hydrodynamic Body/Goto'

I saw this error had already been reported two years ago for an older version of MatLab, but I don't know how to solve it right now with the latest version I use (9.14.0.2254940 R2023a). 

Do you have any solution to deal with this issue ?

Thanks

nathanmtom, 2023-06-02:
Thank you for your question. We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "Bug report" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title.

antbog, 2023-06-05:
Just reopened a new issue

closed 2023-06-05 by antbog