WEC-Sim/WEC-Sim #1068: [Theory or Implementation] CG location of the example sphere

yman95, 2023-06-03:
Hi sorry to ask about CG again

**Is your question request related to a problem? Please describe.**
I'm trying to compare the BEM results from mine to the sphere example in \example\BEMIO\NEMOH. The example geometry \Mesh\Nemoh.dat is the submerged part of a sphere with radius 5, but its CG is located at (0,0,-2). It seems like it is the CG of the submerged part only and not the entire sphere. (The black boxes in figures show CG)
Example geometry:
[image]

When I entered the example sphere's mass and use archimedes function, I get the exact same center of buoyancy but my CG is at the sphere's geometric center. 
[image]
Submerged part:
[image]

So should the CG be done about the entire geometry or just the submerged part? Thanks.

**Describe the theory or implementation approach that you have a question**
Since the BEM calculations are to be done about CG, so I figure it's important to understand where CG should be. 

**Describe the type of conclusion or resolution you are looking for**
Understanding the location of CG

Thanks for the support,
Yi

dforbush2, 2023-06-08:
Hi @yman95 

The origin of the BEM run should be at the CG of the whole body and the BEM mesh file should contain only the submerged geometry. Because the origin _must_ be at the CG of the body, we specify the offset between the body (0,0,0 at CG) and global coordinate system in the input file, in this case, (0,0,2). 

If you look at the WAMIT or CAPYTAINE versions of this example, you will see a mesh position offset by +0,0,2, which makes sense in this context. It seems that this is intended to model either a case where a hemisphere is the total geometry (and thus has a CG at 3*radius/8...perhaps approximated as 2 here?) or a sphere with a non-uniform internal mass distribution resulting in a CG below that which would be predicted purely from geometry. 

As you point out, however, the NEMOH example is different. 

To be honest I have not run NEMOH in so long I forget if its coordinate system definition differs from WAMIT, but it is possible you have identified a bug here. I will follow up soon.

yman95, 2023-06-13:
Understood, thanks. So, it was because the mass of the geometry was not uniformly distributed (bottom heavy), such that the CG is located at a lower place. Also, NEMOH has different coordinates convention than other software. It was confusing to me but now I get it.

I was able to get similar results using my mesh with the same parameters with minor spike differences, for example the normalized excitation force magnitude for the cylinder example.
WEC-Sim example result:
[image]

my result:
[image]

WEC-Sim example result:
[image]

my result:
[image]

Am I correct to assume that those are because mesh differences?
WEC-Sim mesh (regular rectangles): 
[image]

my mesh (from FreeCAD, triangular):
[image]

dforbush2, 2023-06-14:
Hi @yman95 

Your results look great. It's hard to say exactly where the minor differences come from, but its probable, as you suggest, that this is a mesh difference. 

This sphere example was from an IEA OES testing campaign and represents an actual physical device that was tested, so the CG that is not at the geometric center is intentional and is representative of the tested device.

closed 2023-06-14 by dforbush2