WEC-Sim/WEC-Sim #1069: [BUG] Error due to multiple Goto on tutorial examples

antbog, 2023-06-05:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_A clear and concise description of what the perceived bug is._

When running the tutorials of WEC-Sim, an error appears, for both the RM3 and the OSWEC examples, caused by multiple Gotos according to the error message :

> {Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
> Error due to multiple causes.
> Caused by:
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     Multiple Gotos found with tag named 'disp1'
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     First offending Goto '<a href="matlab:open_and_hilite_hyperlink ('RM3/Float/Hydrodynamic Body/Goto','error')">RM3/Float/Hydrodynamic Body/Goto</a>'
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     Second offending Goto '<a href="matlab:open_and_hilite_hyperlink ('RM3/Spar//Plate/Hydrodynamic Body/Goto','error')">RM3/Spar//Plate/Hydrodynamic Body/Goto</a>'
> } 

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._

WEC-Sim/tutorials/RM3 and WEC-Sim/tutorials/OSWEC (both the same issue, after running the simulation using >wecSim in MatLab terminal).

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

Just following the steps given in the tutorial section of the User Manual :
1. Run bemio
2. Build SimuLink model (RM3 or OSWEC)
3. write wecSimInputFile.m
4. run WEC-Sim
5. see the initialization, preprocessing, wave environment, list of Body, list of PTO and list of constraints running normally
6. just after simulation of the WEC device defined in SimMechanics model : the error appears.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

Command line "wecSim"

**Expected behavior**
_A clear and concise description of what you expected to happen._

Visualize and the simulation as it should be and have the output written.

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

> WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory... 
> WEC-Sim: An open-source code for simulating wave energy converters
> Version: 5.0.1
> 
> Initializing the Simulation Class...
> 	Case Dir: C:\Users\antoinebo\Documents\GitHub\WEC-Sim\tutorials\RM3 
> Elapsed time is 0.225738 seconds.
> 
> WEC-Sim Pre-processing ...   
> [Warning: Removed 'C:\Users\antoinebo\Documents\GitHub\WEC-Sim\tutorials\RM3\output'
> from the MATLAB path for this MATLAB session.
> 	See 'doc path' for more information.] 
> [> In <a href="matlab:matlab.internal.language.introspective.errorDocCallback('simulationClass/setup', 'C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\objects\simulationClass.m', 162)" style="font-weight:bold">simulationClass/setup</a> (<a href="matlab: opentoline('C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\objects\simulationClass.m',162,0)">line 162</a>)
> In <a href="matlab:matlab.internal.language.introspective.errorDocCallback('initializeWecSim', 'C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\functions\initializeWecSim.m', 215)" style="font-weight:bold">initializeWecSim</a> (<a href="matlab: opentoline('C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\functions\initializeWecSim.m',215,0)">line 215</a>)
> In <a href="matlab:matlab.internal.language.introspective.errorDocCallback('run', 'C:\Program Files\MATLAB\R2023a\toolbox\matlab\lang\run.m', 91)" style="font-weight:bold">run</a> (<a href="matlab: opentoline('C:\Program Files\MATLAB\R2023a\toolbox\matlab\lang\run.m',91,0)">line 91</a>)
> In <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim', 'C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\wecSim.m', 43)" style="font-weight:bold">wecSim</a> (<a href="matlab: opentoline('C:\Users\antoinebo\Documents\GitHub\WEC-Sim\source\wecSim.m',43,0)">line 43</a>)] 
> 	Infinite water depth specified in BEM and "waves.waterDepth" not specified in input file.
> Set water depth to 200m for visualization.
> Elapsed time is 0.155906 seconds.
> 
> WEC-Sim Simulation Settings:
> 	Start Time                     (sec) = 0
> 	End Time                       (sec) = 400
> 	Time Step Size                 (sec) = 0.1
> 	Ramp Function Time             (sec) = 100
> 	 Number of Time Steps     = 4000 
> 
> Wave Environment: 
> 	Wave Type                            = Regular Waves (Constant Hydrodynamic Coefficients)
> 	Wave Height, height (m)                   = 2.5
> 	Wave Period, period (sec)                 = 8
> 
> List of Body: Number of Bodies = 2 
> 
> 	***** Body Number 1, Name: float *****
> 	Body CG                          (m) = [0,0,-0.72]
> 	Body Mass                       (kg) = 725833 
> 	Body Diagonal MOI              (kgm2)= [2.09073E+07,2.13061E+07,3.70855E+07]
> 
> 	***** Body Number 2, Name: spar *****
> 	Body CG                          (m) = [0,0,-21.29]
> 	Body Mass                       (kg) = 886687 
> 	Body Diagonal MOI              (kgm2)= [9.44196E+07,9.44071E+07,2.85422E+07]
> 
> List of PTO(s): Number of PTOs = 1 
> 
> 	***** PTO Name: PTO1 *****
> 	PTO Stiffness           (N/m;Nm/rad) = 0
> 	PTO Damping           (Ns/m;Nsm/rad) = 1.2E+06
> 
> List of Constraint(s): Number of Constraints = 1 
> 
> 	***** Constraint Name: Constraint1 *****
> 
> 
> Simulating the WEC device defined in the SimMechanics model C:\Users\antoinebo\Documents\GitHub\WEC-Sim\tutorials\RM3\RM3.slx...   
> Elapsed time is 8.429617 seconds.
> {Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
> Error due to multiple causes.
> Caused by:
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     Multiple Gotos found with tag named 'disp1'
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     First offending Goto '<a href="matlab:open_and_hilite_hyperlink ('RM3/Float/Hydrodynamic Body/Goto','error')">RM3/Float/Hydrodynamic Body/Goto</a>'
>     Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wecSim')" style="font-weight:bold">wecSim</a>
>     Second offending Goto '<a href="matlab:open_and_hilite_hyperlink ('RM3/Spar//Plate/Hydrodynamic Body/Goto','error')">RM3/Spar//Plate/Hydrodynamic Body/Goto</a>'
> } 

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10 Enterprise, Version 22H2
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 9.14.0.2254940 R2023a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5.0.1

**Additional context**
_Add any other context about the problem here._
N/A

jtgrasb, 2023-06-05:
@antbog 

Thank you for opening a new issue using the template. It seems your bug is likely related to the update to MATLAB R2023a as WEC-Sim has not yet been tested for this version (similar to Issue #549 when MATLAB 2021a was released). We will work on addressing this bug

akeeste, 2023-06-05:
Hi @antbog 

I am able to run both the WEC-Sim RM3 and OSWEC examples on R2023a without issue. Once built, the tutorials should essentially be the same case, so they should also run without error. Can you test the prebuilt RM3 or OSWEC examples in R2023a?

Otherwise, this error may come up if there's a set-up problem with the Simulink model. Can you check that each body has a different number in the Simulink model?
[image]

Adam

antbog, 2023-06-06:
Hi @akeeste,

The error came indeed from this Simulink set-up problem, now it's working.

Thanks for the support!

closed 2023-06-06 by antbog

akeeste, 2023-06-06:
@antbog Glad to hear your simulations are working!