WEC-Sim/WEC-Sim #1072: NEMOH CODE FOR MULTIPLE BODIES

ssarant, 2023-06-18:
Hello

I would like to ask if there is a specific code (or process) to calculate the hydrodynamic loads acting on multiple interacting bodies, with the aim of NEMOH.

Thank you in advance

dav-og, 2023-06-18:
I'm not sure exactly what you are asking - have you looked over the WEC-Sim documentation? 

- Theory: https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#boundary-element-method-bem
- Workflow: https://wec-sim.github.io/WEC-Sim/master/user/workflow.html#workflow

In short, WEC-Sim requires frequency-domain hydrodynamic coefficients to provided as an input, which it uses to compute time-domain hydrodynamic forces.

You can use Nemoh to compute these hydrodynamic forces, but I would recommend looking into [Capytaine](https://github.com/capytaine/capytaine), which is actively being supported and is a Python-based, open-source re-write of Nemoh, with many improvements.

On the commercial software side, WAMIT and AQWA are 2 popular BEM codes for computing hydrodynamic coefficients.

I hope that's helpful! :)

Dave

ssarant, 2023-06-18:
Dear David

Thank you for your response. As a matter of fact, I am trying to calculate the hydrodynamic load (exciting forces and moments) acting on a multiple-body structure. I know that NEMOH can calculate these loads for one body but I am not sure if it calculate these loads in case of a multiple-body structure.

Best regards

Sarantos
________________________________
Από: David Ogden ***@***.***>
Στάλθηκε: Κυριακή, 18 Ιουνίου 2023 6:09 μμ
Προς: WEC-Sim/WEC-Sim ***@***.***>
Κοιν.: ΣΑΡΑΝΤΟΣ ΣΑΡΑΝΤΙΔΗΣ ***@***.***>; Author ***@***.***>
Θέμα: Re: [WEC-Sim/WEC-Sim] NEMOH CODE FOR MULTIPLE BODIES (Issue #1072)

I'm not sure exactly what you are asking - have you looked over the WEC-Sim documentation?

  *   Theory: https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#boundary-element-method-bem
  *   Workflow: https://wec-sim.github.io/WEC-Sim/master/user/workflow.html#workflow

In short, WEC-Sim requires frequency-domain hydrodynamic coefficients to provided as an input, which it uses to compute time-domain hydrodynamic forces.

You can use Nemoh to compute these hydrodynamic forces, but I would recommend looking into Capytaine<https://github.com/capytaine/capytaine>, which is actively being supported and is a Python-based, open-source re-write of Nemoh, with many improvements.

On the commercial software side, WAMIT and AQWA are 2 popular BEM codes for computing hydrodynamic coefficients.

I hope that's helpful! :)

Dave

dav-og, 2023-06-28:
Hi @ssarant, yes Nemoh can compute the hydrodynamic coefficients for multiple bodies (and their hydrodynamic interactions). e.g. for a 2-body system you would obtain a 12x12 added mass matrix, a 12x1 excitation force vector etc (for each frequency).

WAMIT, AQWA and Capytaine also support multiple bodies. As far as I know, the only BEM code that does not is HAMS.

Let me know if you have any more questions or if I can close out this issue 👍

Cheers!
Dave

ssarant, 2023-06-30:
Hy David

Is there a specific code (written in MATLAB code) that I can use in order to calculate the hydrodynamic forces acting on multiple cylinder arrangements, in NEMOH computational environment?

Best regards

Sarantos
________________________________
Από: David Ogden ***@***.***>
Στάλθηκε: Τετάρτη, 28 Ιουνίου 2023 5:31 μμ
Προς: WEC-Sim/WEC-Sim ***@***.***>
Κοιν.: ΣΑΡΑΝΤΟΣ ΣΑΡΑΝΤΙΔΗΣ ***@***.***>; Mention ***@***.***>
Θέμα: Re: [WEC-Sim/WEC-Sim] NEMOH CODE FOR MULTIPLE BODIES (Issue #1072)

Hi @ssarant<https://github.com/ssarant>, yes Nemoh can compute the hydrodynamic coefficients for multiple bodies (and their hydrodynamic interactions). e.g. for a 2-body system you would obtain a 12x12 added mass matrix, a 12x1 excitation force vector etc (for each frequency).

WAMIT, AQWA and Capytaine also support multiple bodies. As far as I know, the only BEM code that does not is HAMS.

Let me know if you have any more questions or if I can close out this issue 👍

Cheers!
Dave

dav-og, 2023-07-26:
Hi @ssarant there are MATLAB wrappers available for Nemoh...but Nemoh is not being actively maintained (and hasn't been for several years) - generally we recommend people to use the Python-based BEM code, [Capytaine](https://github.com/capytaine/capytaine), which we're currently supporting - check it out and if you have further questions you can use the issues board there 👍
Cheers,
Dave

closed 2023-07-26 by dav-og