WEC-Sim/WEC-Sim #1075: [Theory or Implementation]

Omirzadeh, 2023-06-27:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I have a problem with the state space model of radiation forces in Wec Sim.  I do not understand why Wec Sim returns a non-zero D_r matrix.

**Describe the theory or implementation approach that you have a question**
I have a problem with the state space model of radiation forces in Wec Sim. Why Wec Sim returns a non-zero D_r matrix?

**Personal project context**
I've modeled a cylindrical buoy in Ansys Aqua and I want to use state space matrices in my project.

**Describe the type of conclusion or resolution you are looking for**
N/A

**Additional context**
N/A

nathanmtom, 2023-06-27:
@Omirzadeh Thank you for reopening the issue following our template. 

For clarity, the theory and implementation of the state space model of the radiation forces can be found here:

http://wec-sim.github.io/WEC-Sim/master/theory/theory.html#state-space

As briefly described in our documentation, with greater detail provided in the references, the realization theory used may result in a non-zero D_{r} matrix but upon implementation we ignore this term per the reasons in our theory. 

If you have other questions or comments about our implementation please let us know.

Cheers, 
Nathan

salhus, 2023-06-28:
Hi @Omirzadeh ,

In the papers @nathanmtom mentioned, https://www.sciencedirect.com/science/article/pii/S0029801805000946 and https://www.sciencedirect.com/science/article/pii/S0029801807002363
the work uses MATLAB's *imp2ss* function which comes as part of additional toolboxes for Control Systems.

The SS estimation is an iterative process that checks the error between the radiation IRF from BEM data and the radiation IRF of the estimated SS. The calculation of the IRF $K_{SS}$ from the estimated SS system at a $k^{th}$ iteration is given by,  
$K_{SS}(k) = C_d e^{A_d dt(k-1)} B_d$

Notice, that it doesn't have any $D_d$, and rightly so, as we are estimating a causal system as the radiation IRF is $0$ for all $t<0$.  Now, since we know the IRF is causal and the IRF calculation doesn't use the $D_d$ matrix, we can safely set the $D_d$ to $0$.

A general guideline for optimization/system identification is to know the features of the system being solved, which in this case is the causality of the radiation IRF. 

Cheers,
Sal

nathanmtom, 2023-07-05:
@Omirzadeh Please let us know by the end of this week if you have any further questions regarding our responses. If you do not provide a response we will close the issue and assume it has been resolved.

Omirzadeh, 2023-07-05:
Thanks for your prompt response.

closed 2023-07-05 by Omirzadeh